mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #701: Need help with my P10 Matrix Panel doesn't working

dy-f0rmidable, 2024-11-19:
Hello
I'm working on controlling P10 Matrix LED Panel using ESP32
I have successfully upload my code to my ESP32 and make sure the P10 panel is using 5V voltage but my LED doesn't turn on
pls help

here is my code:
[image]

[image]

My ESP32 is WROOM-1-N16R8 ESP32-S3-DevKitC-1
[image]

My P10 Matrix LED Panel is Green Only Color
[image]

board707, 2024-11-19:
> My P10 Matrix LED Panel is Green Only Color

The library supports a RGB panels only.

dy-f0rmidable, 2024-11-21:
> > My P10 Matrix LED Panel is Green Only Color
> 
> The library supports a RGB panels only.

Do you know any library that support for single color panels ?

board707, 2024-11-21:
With which controller? ESP32 ? Or any other?

dy-f0rmidable, 2024-11-21:
> With which controller? ESP32 ? Or any other?

yes, with ESP32

board707, 2024-11-21:
the only I know is
https://github.com/Qudor-Engineer/DMD32

dy-f0rmidable, 2024-11-21:
> the only I know is https://github.com/Qudor-Engineer/DMD32
> 
> Disclaimer: I haven't used the library and don't know the author

thanks you, i will check it

board707, 2025-02-06:
> As you know, single-color panels have only one data line, and theoretically everything else is the same.

Some things are different:
- the entire HUB12 panel is a single data channel, while a HUB75 panels has a two parallel channels for upper and lower parts
- every HUB12 panel is a "quarter-scan panel" with non-straight pixel mapping 
- the principle of using LAT and OE signals in monochrome panels differs from HUB75

The last factor, I think, is what causes the ghostings in your case.

I wrote a library for STM32 controllers that works with both monochrome and color panels. But I use different code to output signals in each case.

Endorfin35, 2025-02-06:
1. and 2. You are right, but I can optimize the situation to overcome this issue. You must have understood this from the image.

3. I didn't fully understand this point. Are you saying that the latch and OE operations are different? Could you explain a bit?

Would I have a chance to get results by editing the library a little?

board707, 2025-02-06:
> Are you saying that the latch and OE operations are different?

On some (many) HUB75 panels the driver has a dual buffer, where LAT is used as latch for a first buffer and OE for a second.  On monochrome panels, in contrast, the OE always used for switching output ON and OFF.
This determines the different sequence of switching on signals on monochrome and RGB matrices to prevent ghosting.

Endorfin35, 2025-02-07:
I set the value of uint8_t colouridx = fb->rowBits[row_idx]->colour_depth; to 1 in ESP32-HUB75-MatrixPanel-I2S-DMA.cpp and achieved a better result. Currently, there is a slight shadowing above and below the rows. I believe it is related to row scanning. I guess I'll give up on it.

[image]

[image]