mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #702: sm16208s and sm5166 panel

j-g00da, 2024-11-21:
I need to buy an outdoor panel and want it to work with the library.
I asked sellers and sent them list of supported ICs asking for panels that use the same ones or similar and got recommended [this panel](https://pl.aliexpress.com/item/1005005549507676.html?pdp_npi=3%40dis%21USD%21%21US%20%2420.92%21%21%21%21%21%40212a70c117321740954017611e81d9%21%21im%21%21&gatewayAdapt=glo2pol), seller says it uses sm16208s and sm5166. They also say that "Equivalent to 138 lines decoding It is the most commonly used line decoding".

Will it work? If not, can I get a recommendation on some 160mm high outdoor screen that is tested and will work?

I also can report on some other panels I already tested:
[This 80x40 flex screen](https://pl.aliexpress.com/item/1005004377560075.html?spm=a2g0s.imconversation.0.0.38ca3e5fYcen8u&pdp_npi=4%40dis%21PLN%2177%2C86%20zł%2177%2C79%20zł%21%21%2118.46%2118.44%21%402103891017321756267702431ecc3c%2112000028987536420%21sh%21PL%211719956689%21X&gatewayAdapt=glo2pol&_gl=1*11k91kq*_gcl_aw*R0NMLjE3MzE0OTI5MzMuQ2p3S0NBaUF1ZEc1QmhBUkVpd0FXTWxTaktpdjZndVdCZG9NMDJwMW1NZ2VTYk9NUzJoRTZwVHVuSEM2ZjVYNGRxT3h4R0NXT3h3azVob0M1UG9RQXZEX0J3RQ..*_gcl_dc*R0NMLjE3MzE0OTI5MzMuQ2p3S0NBaUF1ZEc1QmhBUkVpd0FXTWxTaktpdjZndVdCZG9NMDJwMW1NZ2VTYk9NUzJoRTZwVHVuSEM2ZjVYNGRxT3h4R0NXT3h3azVob0M1UG9RQXZEX0J3RQ..*_gcl_au*ODc0ODI5MDYwLjE3Mjk1ODg5MzQ.*_ga*Mjk4MjIzMDE5LjE3Mjk1ODg5MDI.*_ga_VED1YSGNC7*MTczMjE3MzA4Ny4xMy4xLjE3MzIxNzY5OTAuNjAuMC4w) does work well (I have a chain of two) with `mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;` setting. I would use it but I need outdoor version...
[This 80x40 outdoor screen](https://pl.aliexpress.com/item/1005005996377086.html?spm=a2g0s.imconversation.0.0.38ca3e5fYcen8u&algo_pvid=60554944-12c3-4e2f-992d-df63432076e8&algo_exp_id=60554944-12c3-4e2f-992d-df63432076e8-0&pdp_npi=4%40dis%21PLN%21154.43%2183.39%21%21%2136.62%2119.78%21%40211b80e117321753143412569ed10a%2112000035228997415%21sea%21PL%211719956689%21X&curPageLogUid=SlkQZ6xMubNe&utparam-url=scene%3Asearch%7Cquery_from%3A&_gl=1*1l3cju6*_gcl_aw*R0NMLjE3MzE0OTI5MzMuQ2p3S0NBaUF1ZEc1QmhBUkVpd0FXTWxTaktpdjZndVdCZG9NMDJwMW1NZ2VTYk9NUzJoRTZwVHVuSEM2ZjVYNGRxT3h4R0NXT3h3azVob0M1UG9RQXZEX0J3RQ..*_gcl_dc*R0NMLjE3MzE0OTI5MzMuQ2p3S0NBaUF1ZEc1QmhBUkVpd0FXTWxTaktpdjZndVdCZG9NMDJwMW1NZ2VTYk9NUzJoRTZwVHVuSEM2ZjVYNGRxT3h4R0NXT3h3azVob0M1UG9RQXZEX0J3RQ..*_gcl_au*ODc0ODI5MDYwLjE3Mjk1ODg5MzQ.*_ga*Mjk4MjIzMDE5LjE3Mjk1ODg5MDI.*_ga_VED1YSGNC7*MTczMjE3MzA4Ny4xMy4xLjE3MzIxNzcwOTguNjAuMC4w) doesn't work at all, it's all black no matter what settings I use. Can't tell what ICs it uses as the board is waterproofed.

board707, 2024-11-21:
> sm5166. They also say that "Equivalent to 138 lines decoding It is the most commonly used line decoding".

true, sm5166 chip is a "type138" line decoder

> Will it work?

Any of outdoor panels, even with supported chips - need to configure the order of the pixel output, which is not so easy to do in this library. I configured such panels for several people and described the scheme of the geometric function in the discussion #622.
Most panels are quite easy to configure, but sometimes there are very complex cases; it is impossible to predict the result without testing on hardware.

j-g00da, 2024-11-21:
I will probably buy few different models and test which one works best.
I also got response from another seller, they have solutions with ICN2037BP+RUC7258D and LS9930CS+RUC7258D. Are these combinations worth checking?

board707, 2024-11-21:
> they have solutions with ICN2037BP+RUC7258D and LS9930CS+RUC7258D. Are these combinations worth checking?

The first option should work, the second definitely won't, LS9930CS - unsupported driver

j-g00da, 2024-11-22:
Ok so seller informed me that they have two versions of [the 80x40 outdoor module I talked about in the first comment](https://pl.aliexpress.com/item/1005005996377086.html?spm=a2g0s.imconversation.0.0.38ca3e5fYcen8u&algo_pvid=60554944-12c3-4e2f-992d-df63432076e8&algo_exp_id=60554944-12c3-4e2f-992d-df63432076e8-0&pdp_npi=4%40dis%21PLN%21154.43%2183.39%21%21%2136.62%2119.78%21%40211b80e117321753143412569ed10a%2112000035228997415%21sea%21PL%211719956689%21X&curPageLogUid=SlkQZ6xMubNe&utparam-url=scene%3Asearch%7Cquery_from%3A&_gl=1*1l3cju6*_gcl_aw*R0NMLjE3MzE0OTI5MzMuQ2p3S0NBaUF1ZEc1QmhBUkVpd0FXTWxTaktpdjZndVdCZG9NMDJwMW1NZ2VTYk9NUzJoRTZwVHVuSEM2ZjVYNGRxT3h4R0NXT3h3azVob0M1UG9RQXZEX0J3RQ..*_gcl_dc*R0NMLjE3MzE0OTI5MzMuQ2p3S0NBaUF1ZEc1QmhBUkVpd0FXTWxTaktpdjZndVdCZG9NMDJwMW1NZ2VTYk9NUzJoRTZwVHVuSEM2ZjVYNGRxT3h4R0NXT3h3azVob0M1UG9RQXZEX0J3RQ..*_gcl_au*ODc0ODI5MDYwLjE3Mjk1ODg5MzQ.*_ga*Mjk4MjIzMDE5LjE3Mjk1ODg5MDI.*_ga_VED1YSGNC7*MTczMjE3MzA4Ny4xMy4xLjE3MzIxNzcwOTguNjAuMC4w). One is ICN2037BP+RUC7258D and the second one is LS9930CS+RUC7258D, and you get a random one... They sent me pictures of ICN2037BP module and it has slightly different pcb layout so it makes sense that mine must be LS9930CS, and that's why it doesn't display anything...

j-g00da, 2024-12-10:
Got the ICN2037BP+RUC7258D panel - it works but as expected, the order of pixels is messed up. Was such order observed before (might save me some time)? I will try to configure it.

The module is 80x40, it goes by 16-wide lines, two at the time, renders half of the screen, then stops (probably renders off-screen), then proceeds to re-render the top side (Idk if you can see it on video, but pixels get a bit brighter), then it proceeds to do the same to lower half.

https://github.com/user-attachments/assets/40c36d18-87cb-4673-bc6f-3c47ef6893a9

board707, 2024-12-10:
please show the code that produced this video

j-g00da, 2024-12-10:
I drew it with such function:
```
void testPixels() {
  virtualDisp->clearScreen();
  int h = virtualDisp->height();
  int w = virtualDisp->width();
  int col = 0;
  int cnt = 0;
  for (int i = 0; i < h; i++) {
    for (int j = 0; j < w; j++) {
      virtualDisp->drawPixel(j, i, RED);
      delay(100);
    }
  }
}
```

Screen init part:
```
HUB75_I2S_CFG mxconfig(
  panel_x,
  panel_y,
  rows * cols);
dmaDisplay = new MatrixPanel_I2S_DMA(mxconfig);
dmaDisplay->setLatBlanking(4);
dmaDisplay->begin();
dmaDisplay->setBrightness8(200);  // 0-255, values higher than 200 may introduce ghosting
virtualDisp = new VirtualMatrixPanel((*dmaDisplay), rows, cols, panel_x, panel_y, chain_type);
virtualDisp->clearScreen();
virtualDisp->setCursor(0, 0);
```

panel_x = 80
panel_y = 40
These don't matter, I use VirtualMatrixPanel only because I used a chain before testing the new screen:
num_rows = 1
num_cols = 1
chain_type = CHAIN_BOTTOM_RIGHT_UP

board707, 2024-12-10:
Please show a full code

board707, 2024-12-10:
> I use VirtualMatrixPanel only because I used a chain

No, it is not correct. You must use a VirtualMatrixPanel even for a single panel if you have a i.e. QuarterScanPanel type. What is the scan factor of your 80x40 matrix?

j-g00da, 2024-12-10:
> > I use VirtualMatrixPanel only because I used a chain
> 
> No, it is not correct. You must use a VirtualMatrixPanel even for a single panel if you have a i.e. QuarterScanPanel type. What is the scan factor of your 80x40 matrix?

I know, what I meant is _for now_ it doesn't do anything.

Panel is 1/10 scan, so a "four-scan". It's part of larger code so I think I will make a minimal example instead of sharing that, tested and produces the same output:

```
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

// Display settings
#define PANEL_RES_X 80
#define PANEL_RES_Y 40
#define NUM_COLS 1
#define NUM_ROWS 1
#define PANEL_CHAIN NUM_COLS * NUM_ROWS
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_RIGHT_UP

// Colors
#define WHITE virtualDisp->color565(255, 255, 255)
#define BLACK virtualDisp->color565(0, 0, 0)
#define RED virtualDisp->color565(255, 0, 0)
#define GREEN virtualDisp->color565(0, 255, 0)
#define BLUE virtualDisp->color565(0, 0, 255)

MatrixPanel_I2S_DMA *dmaDisplay = nullptr;
VirtualMatrixPanel *virtualDisp = nullptr;

void testPixels() {
  virtualDisp->clearScreen();
  int h = virtualDisp->height();
  int w = virtualDisp->width();
  for (int i = 0; i < h; i++) {
    for (int j = 0; j < w; j++) {
      virtualDisp->drawPixel(j, i, RED);
      delay(100);
    }
  }
}

void setupScreen(int panel_x, int panel_y, int rows, int cols, PANEL_CHAIN_TYPE chain_type) {
  HUB75_I2S_CFG mxconfig(
    panel_x,
    panel_y,
    rows * cols);

  dmaDisplay = new MatrixPanel_I2S_DMA(mxconfig);
  dmaDisplay->setLatBlanking(4);
  dmaDisplay->begin();
  dmaDisplay->setBrightness8(200);  // 0-255, values higher than 200 may introduce ghosting
  virtualDisp = new VirtualMatrixPanel((*dmaDisplay), rows, cols, panel_x, panel_y, chain_type);
  virtualDisp->clearScreen();
  virtualDisp->setCursor(0, 0);
}

void setup() {
  setupScreen(PANEL_RES_X, PANEL_RES_Y, NUM_ROWS, NUM_COLS, VIRTUAL_MATRIX_CHAIN_TYPE);
}

void loop() {
  testPixels();
  delay(2000);
}
```

j-g00da, 2024-12-10:
In mxconfig I changed panel_y to panel_y/2 - now it render continously, doesn't go "off-screen" half way through as before. Still renders in two places at the same time.

board707, 2024-12-10:
> now it render continously,

Could you show the video after the change?

j-g00da, 2024-12-10:
Here: mxconfig set to panel_x, panel_y/2. Only other change is delay 100->10.

https://github.com/user-attachments/assets/77e2f649-4d06-4947-868b-77b5d10a9b13

I also tried to set set mxconfig to panel_x\*2, panel_y/2, now it renders only one pixel at a time as it should, but only left side (with masonry shape in the middle) was rendered.
Then I tried following #622 - set mxconfig to panel_x\*2, panel_y/2 and used setPhysicalPanelScanRate to use my own code:
```
        if ((coords.y & 10) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX;
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX;
        }
        coords.y = (virt_y / 20) * 10 + (virt_y & 9);
```
It left underndered horizontal strips.

j-g00da, 2024-12-10:
OK! I'm close now, had to change math a bit, since the resolution is not in powers of 2.

```
    else if (panel_scan_rate == FOUR_SCAN_40PX_HIGH)
    {
        if ((coords.y / 10) % 2 == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX;
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX;
        }
        coords.y = (coords.y / 20) * 10 + (coords.y % 10);
    }
```

j-g00da, 2024-12-10:
I've set the pixel base to 16.
```
uint8_t pxbase = 16;
if ((coords.y / 10) % 2 == 0)
{
    coords.x += ((coords.x / pxbase) + 1) * pxbase;
}
else
{
    coords.x += (coords.x / pxbase) * pxbase;
}
coords.y = (coords.y / 20) * 10 + (coords.y % 10);
```
Almost there:

https://github.com/user-attachments/assets/a96f0d2d-31e1-4e01-9da7-de5a10944e06

board707, 2024-12-10:
Amazing results :)
Just a little bit left to success.

j-g00da, 2024-12-10:
Yeah, I fixed that and then tried to display some multicolor image and realised that the shift is different on different color channels, which is very weird.

board707, 2024-12-10:
> I fixed that

Could you explain how did you fixed it? Show the code?

board707, 2024-12-10:
According to the video, it's seems that the first CLK edge or first data pulse is lost.
Probably it could be fixed by changing a data output parameters, say `clkphase` or tuning the line 
```
dmaDisplay->setLatBlanking(4);
```

May be @mrcodetastic could help

mrcodetastic, 2024-12-10:
clkphase setting?

board707, 2024-12-10:
sorry, I don't know.
this is just a guess

j-g00da, 2024-12-11:
I'll be working on it later today. Will give update for that. I'm using cystom board that has matched length for RGB lanes and works perfectly on other, indoor display.
My guesses are:
1. Power - I'm using 5V3A psu, on other board I observed some ghosting on parts of the screen, after hours of tinkering I decided to turn the voltage adjustment knob on the psu - it started working perfectly on about 6V. I tried doing the same on this board, some pixels aligned but other stayed shifted. Maybe I will try adding a capacitor?
2. Damage - I think it's rather unlikely but to mount the custom board (it plugs directly to hub75) I had to remove some of the black plastic on the back, maybe it damaged the panel's pcb?
3. Quirky panel, will try changing clkphase and if it doesn't work I will try to modify code to only shift red...

j-g00da, 2024-12-11:
Update on power - for some reason my PSU outputs 5.7V on lowest V adj. setting. Maybe this is too far from the 5V?

mrcodetastic, 2024-12-11:
Do you have a 3.3->5v logic level shifter between the ESP32 and the panel? Sorry if this has been asked before.

j-g00da, 2024-12-11:
No, I asked if it would have any benefit in #703

board707, 2024-12-11:
In your case we need to test everything, so it could be worth to test a using of level shifter

Ogami-B, 2025-01-05:
> Yeah, I fixed that and then tried to display some multicolor image and realised that the shift is different on different color channels, which is very weird.

Hey, how exactly did you fix it? like what code did you use?