mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #705: ESP32 driver LED panel no display P1.92 resolution 104X52

1359450511, 2024-12-09:
Hello, I have a panel model P1.92/104X52/26S-SMD1515

The chips that can be seen by the eyes are D7258 ICN2053BP 74HC04D 74HC245. I am currently unable to light them up using your library and there is no response. Does your library support these chips?

Hello, I have a panel model P1.92/104X52/26S-SMD1515

The chips that can be seen by the eyes are D7258 ICN2053BP 74HC04D 74HC245. I am currently unable to light them up using your library and there is no response. Does your library support these chips?

[image]

board707, 2024-12-09:
To support a panels with ICN2053 driver see the discussion #324

mrcodetastic, 2024-12-10:
> Does your library support these chips?

No