mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #706: Need help with WaveShare P2.5 64x32 LED Matrix on ESP32 (ESP-IDF)

mibar-me, 2024-12-11:
Hej!

I have to setup a small project displaying something on a LED Matrix for university project. However after booting up my ESP it always displays random pixel artifacts on the matrix. This "random" pattern doesn't change with every flash of the esp. It only changes when I unplug and replug the ESP32. The matrix does however react on brightness changes. I can't however get it to display the stuff I want it to display.

Here is a selection of different boot up patterns:

[image]
[image]
[image]

**Bonus:** when I slightly touch the C, D, CLK and LAT pins while they are connected the pattern changes:
[image]

I wired the ESP32-D0WD-V3 with a [WaveShare P2.5 64x32 LED Matrix](https://www.waveshare.com/wiki/RGB-Matrix-P2.5-64x32)
- The matrix is connected via the terminal power pin to get external power
- The two unconnected pins are_
    - the E pin (I suppose that I don't need it, because I don't chain multiple matrixes togeteher)
    - the last GRND pin (I also connected it to the esp's grnd pin during my countless tries - it doesn't seem to change anything so I unconnected it. A second GND link also isn't connected to the ESP in [the manufacturer's pin map](https://www.waveshare.com/w/upload/4/45/RGB-Matrix-P2-64x64_ESP3209.jpg))

[image]
 

My example code is based on examples/esp-idf/with-gfx while ""debugging"", reading FAQs and multiple other issues in this repo, this is the latest code I tried out:

```C
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
#include "esp_log.h"

MatrixPanel_I2S_DMA *dma_display = nullptr;
uint16_t myColor  = dma_display->color565(0, 255, 255);
uint16_t myColor2 = dma_display->color565(255, 255, 0);

#define my_R1_PIN 25
#define my_G1_PIN 26
#define my_B1_PIN 27
#define my_R2_PIN 14
#define my_G2_PIN 12
#define my_B2_PIN 13
#define my_A_PIN  23
#define my_B_PIN  22
#define my_C_PIN  5
#define my_D_PIN  17
#define my_E_PIN  -1
#define my_LAT_PIN 4
#define my_OE_PIN  15
#define my_CLK_PIN 16

HUB75_I2S_CFG::i2s_pins _pins={
    my_R1_PIN,
    my_G1_PIN,
    my_B1_PIN,
    my_R2_PIN,
    my_G2_PIN,
    my_B2_PIN,
    my_A_PIN,
    my_B_PIN,
    my_C_PIN,
    my_D_PIN,
    my_E_PIN,
    my_LAT_PIN,
    my_OE_PIN,
    my_CLK_PIN
};

extern "C" void app_main() {

  ESP_LOGI("WAITING", "for panel");
  delay(2000);

  HUB75_I2S_CFG mxconfig(/* width = */ 64, /* height = */ 32, /* chain = */ 1, _pins);
  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_8M;
  mxconfig.driver = HUB75_I2S_CFG::FM6047; // FM6047

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  ESP_LOGI("WAITING", "for panel");
  delay(2000);

  dma_display->begin();
  dma_display->setPanelBrightness(10);
  dma_display->clearScreen();
  dma_display->fillScreen(myColor2);

  while(1) {
    dma_display->fillScreen(myColor);
    ESP_LOGI("Fillscreen", "myColor");
    delay(500);
    dma_display->fillScreen(myColor2);
    ESP_LOGI("Fillscreen", "myColor2");
    delay(500);
    dma_display->clearScreen();

  }

}
```

And this is the output of `idf.py flash monitor`:
```
I (31) boot: ESP-IDF v5.1.5 2nd stage bootloader
I (31) boot: compile time Dec 11 2024 16:22:19
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 40MHz
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 8MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (82) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00100000
I (97) boot: End of partition table
I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0e3c0h ( 58304) map
I (119) esp_image: segment 1: paddr=0001e3e8 vaddr=3ffb0000 size=01c30h (  7216) load
I (120) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=1dd14h (122132) map
I (145) esp_image: segment 3: paddr=0003dd3c vaddr=3ffb1c30 size=00750h (  1872) load
I (146) esp_image: segment 4: paddr=0003e494 vaddr=40080000 size=0c154h ( 49492) load
I (167) boot: Loaded app from partition at offset 0x10000
I (167) boot: Disabling RNG early entropy source...
I (179) cpu_start: Multicore app
I (180) cpu_start: Pro cpu up.
I (180) cpu_start: Starting app cpu, entry point is 0x40081204
--- 0x40081204: call_start_cpu1 at /home/user/Programme/esp/esp-idf-v5.1.5/components/esp_system/port/cpu_start.c:159

I (0) cpu_start: App cpu up.
I (197) cpu_start: Pro cpu start user code
I (197) cpu_start: cpu freq: 160000000 Hz
I (197) cpu_start: Application information:
I (202) cpu_start: Project name:     with-gfx
I (207) cpu_start: App version:      1
I (211) cpu_start: Compile time:     Dec 11 2024 16:22:12
I (217) cpu_start: ELF file SHA256:  3cf34dd509969150...
I (223) cpu_start: ESP-IDF:          v5.1.5
I (228) cpu_start: Min chip rev:     v0.0
I (233) cpu_start: Max chip rev:     v3.99 
I (238) cpu_start: Chip rev:         v3.0
I (243) heap_init: Initializing. RAM available for dynamic allocation:
I (250) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (256) heap_init: At 3FFB3000 len 0002D000 (180 KiB): DRAM
I (262) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (268) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (275) heap_init: At 4008C154 len 00013EAC (79 KiB): IRAM
I (282) spi_flash: detected chip: gd
I (285) spi_flash: flash io: qio
I (290) app_start: Starting scheduler on CPU0
I (294) app_start: Starting scheduler on CPU1
I (294) main_task: Started on CPU0
I (304) main_task: Calling app_main()
I (307) WAITING: for panel
I (2310) WAITING: for panel
I (4310) begin(): Using GPIO 25 for R1_PIN
I (4310) begin(): Using GPIO 26 for G1_PIN
I (4310) begin(): Using GPIO 27 for B1_PIN
I (4313) begin(): Using GPIO 14 for R2_PIN
I (4318) begin(): Using GPIO 12 for G2_PIN
I (4322) begin(): Using GPIO 13 for B2_PIN
I (4327) begin(): Using GPIO 23 for A_PIN
I (4332) begin(): Using GPIO 22 for B_PIN
I (4337) begin(): Using GPIO 5 for C_PIN
I (4341) begin(): Using GPIO 17 for D_PIN
I (4346) begin(): Using GPIO -1 for E_PIN
I (4350) begin(): Using GPIO 4 for LAT_PIN
I (4355) begin(): Using GPIO 15 for OE_PIN
I (4360) begin(): Using GPIO 16 for CLK_PIN
I (4365) gpio: GPIO[25]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4374) gpio: GPIO[14]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4384) gpio: GPIO[26]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4393) gpio: GPIO[12]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4402) gpio: GPIO[27]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4412) gpio: GPIO[13]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4421) gpio: GPIO[16]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4431) gpio: GPIO[4]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4440) gpio: GPIO[15]| InputEn: 1| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (4450) I2S-DMA: Free heap: 377864
I (4453) I2S-DMA: Free SPIRAM: 0
I (4457) I2S-DMA: allocating rowBitStructs with pixel_color_depth_bits of 8
I (4465) I2S-DMA: Allocating 16384 bytes memory for DMA BCM framebuffer(s).
I (4473) I2S-DMA: Minimum visual refresh rate (scan rate from panel top to bottom) requested: 60 Hz
W (4482) I2S-DMA: lsbMsbTransitionBit of 0 gives 30 Hz refresh rate.
W (4489) I2S-DMA: lsbMsbTransitionBit of 1 gives 61 Hz refresh rate.
W (4496) I2S-DMA: lsbMsbTransitionBit of 1 used to achieve refresh rate of 60 Hz. Percieved colour depth to the eye may be reduced.
I (4509) I2S-DMA: DMA has pixel_color_depth_bits of 7
I (4515) I2S-DMA: Recalculated number of DMA descriptors per row: 64
I (4522) ESP32/S2: Allocating memory for 1024 DMA descriptors.
W (4534) ESP32/S2: Creating final DMA descriptor and linking back to 0.
I (4535) I2S-DMA: 1024 DMA descriptors linked to buffer data.
I (4542) ESP32/S2: Performing config for ESP32 or ESP32-S2
I (4548) ESP32/S2: Performing DMA bus init() for ESP32 or ESP32-S2
I (4555) I2S-DMA: DMA setup completed
I (4563) Fillscreen: myColor
I (5063) Fillscreen: myColor2
((it will loop from here onwards))
```

Any help, tips or hints are deeply appreciated! Otherwise I guess I'll get an well-deserved F for this project from my professor.

mrcodetastic, 2024-12-11:
I would do the following:

* Use shorter cables to start with. No more than 15cm
* Start with just one half of the panel and try get that working,  so connect the minimum:
    my_R1_PIN,
    my_G1_PIN,
    my_B1_PIN,
    my_A_PIN,
    my_B_PIN,
    my_C_PIN,
    my_D_PIN,
    my_LAT_PIN,
    my_OE_PIN,
    my_CLK_PIN

* Ensure that the ESP32 ground is connected to a proper ground pin on the panel. Use a multi-meter to confirm.
* Is ` mxconfig.driver = HUB75_I2S_CFG::FM6047; // FM6047` correct for these panels? I have no idea.

Happy to get on a discord or whatever with you. It's Christmas afterall.

mibar-me, 2024-12-11:
**Thank you so much for your reply!** :)

> * Use shorter cables to start with. No more than 15cm

Will try this tomorrow!

> * Start with just one half of the panel and try get that working, so re-connect the minimum: ...

Understood! So now only this area is visible. I haven't changed anything on my source code.
[image]

> * Ensure that the ESP32 ground is connected to a proper ground pin on the panel. Use a multi-meter to confirm.

I reconnected the led panel's [second grnd pin](https://www.waveshare.com/w/upload/4/45/RGB-Matrix-P2-64x64_ESP3209.jpg)(bottom right in the pin out map) to one of the esp's grnd pins (the one next to GPIO23). I can try to get a multi meter tomorrow as I don't have one right now.

> Is  mxconfig.driver = HUB75_I2S_CFG::FM6047; // FM6047 correct for these panels? I have no idea.

This is used in the [manufacturer's demo code](https://files.waveshare.com/upload/b/b6/RGB-Matrix-P2.5-64x32-Demo.zip). I didn't find this particular driver in this repository. It seems to me that they just copied the FM6124 drivers and then renamed it to FM6047 in their project. 
This is the init function they copied (at least to me it seems like they just copied it):
https://gist.github.com/mibar-me/c5aecd8d7c61a5a670fd88b114eafd6a

However during earlier tries I tried out any driver I have found in src/ESP32-HUB75-MatrixPanel-leddrivers.cpp
```C
// src/ESP32-HUB75-MatrixPanel-leddrivers.cpp
  /**
   * Enumeration of hardware-specific chips
   * used to drive matrix modules
   */
  enum shift_driver
  {
    SHIFTREG = 0,
    FM6124,
    FM6126A,
    ICN2038S,
    MBI5124,
    SM5266P,
    DP3246_SM5368
  };
```

I retried all the drivers again.

`SHIFTREG, FM6124, FM6126A, ICN2038S, MBI5124`
With those drivers I get the output I showed above.

`SM5266P, DP3246_SM5368`
With those only one led row is lit up white.

> Happy to get on a discord or whatever with you. It's Christmas afterall.
Its just "mibar" on discord. Feel free to reach out.

mrcodetastic, 2024-12-11:
I can't tell from the photo if that's the top or bottom of the panel etc. Can you confirm, and also show a photo of the back?

panki27, 2024-12-11:
Hi, can you show a picture of the panels backside? I have the same display I believe,
If we actually do have the same model, you will have to swap the green and blue channels.

I had it working fine with an ESP wired according to the [wiki picture](https://www.waveshare.com/wiki/RGB-Matrix-P2.5-64x32#Working_With_ESP32), but I also remember that I had to set `PIN_E` for it to work.

Here's what I ended up using:
```
#define R1 25
#define G1 27 //26
#define BL1 26 //27
#define R2 14
#define G2 13 //12
#define BL2 12 //13
#define CH_A 23
#define CH_B 22
#define CH_C 5
#define CH_D 17
#define CH_E 32
#define CLK 16
#define LAT 4
#define OE 15

#define PIN_E 32
#define PANEL_WIDTH 64
#define PANEL_HEIGHT 32

#define PANELS_NUMBER 1
```
Driver setup:
```
  mxconfig.gpio.e = PIN_E;
  mxconfig.driver = HUB75_I2S_CFG::FM6126A;   // for panels using FM6126A chips

  mxconfig.latch_blanking = 4;
  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;
  mxconfig.clkphase = true;
```

mibar-me, 2024-12-11:
**Thank you so much for your replies!** :)

> I can't tell from the photo if that's the top or bottom of the panel etc. Can you confirm, and also show a photo of the back?
Unfortunately I can't tell if the top or bottom was lit up. Now (see below) only half of the top is lit up. 

Here is a picture of the bare back:
[image]

These are the currently plugged in cables: (As I said I will try out shorter cables tomorrow)
red black power cords are connected to power via a terminal power adapter
[image]

Thanks for your configuration and tips @panki27 I tried it out. This is my current source code:
```C
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
#include "esp_log.h"

MatrixPanel_I2S_DMA *dma_display = nullptr;
uint16_t myColor  = dma_display->color565(0, 0, 255);
uint16_t myColor2 = dma_display->color565(255, 0, 0);

#define my_R1_PIN 25
#define my_G1_PIN 27 // (26) maybe falsely wired
#define my_B1_PIN 26 // (27) maybe falsely wired
#define my_R2_PIN 14
#define my_G2_PIN 13 // (12) maybe falsely wired
#define my_B2_PIN 12 // (13) maybe falsely wired
#define my_A_PIN  23
#define my_B_PIN  22
#define my_C_PIN  5
#define my_D_PIN  17
#define my_E_PIN  32
#define my_LAT_PIN 4
#define my_OE_PIN  15
#define my_CLK_PIN 16

HUB75_I2S_CFG::i2s_pins _pins={
    my_R1_PIN,
    my_G1_PIN,
    my_B1_PIN,
    my_R2_PIN,
    my_G2_PIN,
    my_B2_PIN,
    my_A_PIN,
    my_B_PIN,
    my_C_PIN,
    my_D_PIN,
    my_E_PIN,
    my_LAT_PIN,
    my_OE_PIN,
    my_CLK_PIN
};

extern "C" void app_main() {

  ESP_LOGI("WAITING", "for panel");
  delay(1000);

  HUB75_I2S_CFG mxconfig(/* width = */ 64, /* height = */ 32, /* chain = */ 1, _pins);
  mxconfig.gpio.e = my_E_PIN;
  mxconfig.driver = HUB75_I2S_CFG::FM6126A;   // for panels using FM6126A chips
  mxconfig.latch_blanking = 4;
  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;
  mxconfig.clkphase = true;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  ESP_LOGI("WAITING", "for panel");
  delay(1000);

  dma_display->begin();
  dma_display->setPanelBrightness(10);
  dma_display->clearScreen();
  dma_display->fillScreen(myColor2);

  while(1) {
    dma_display->fillScreen(myColor);
    ESP_LOGI("Fillscreen", "myColor");
    delay(500);
    dma_display->fillScreen(myColor2);
    ESP_LOGI("Fillscreen", "myColor2");
    delay(500);
    dma_display->clearScreen();
  }

}
```

Currently the matrix has this output:
[image]

board707, 2024-12-11:
@mibar-me 
According to the pictures, you have a problem with D pin. Test it connection again.

mibar-me, 2024-12-11:
Thank you for your reply @board707 ! :)

> According to the pictures, you have a problem with D pin. Test it connection again.
What do you mean with "test the connection"? When I slightly touch the D pin connection the top 8 lit up rows (0-7) switch and the next 8 rows (8-15) are lit up. Might there be something wrong with the pin on my ESP I wired to the D pin of my matrix?

> Could you read a partnumber on UG3 or UB3 chips?
Of course. They are both from [Chipone](https://www.chiponeic.com/en/). Their part number reads `ICN2037BP D18837NB`. This seems to apply to all UGx/UBx chips on the panel's back.

board707, 2024-12-11:
> When I slightly touch the D pin connection the top 8 lit up rows (0-7) switch and the next 8 rows (8-15) are lit up.

Wow!
This is a definite proof that you have a problem with D pin or with its signal.
Check the reliability of the connection of this line along the entire signal path - starting from the pin on the ESP32 board, then the connectors on the flat cable, and then the reliability of the cable installation on the HUB75 connector. If you have another similar panel - try it, maybe the first panel is faulty.

> Their part number reads `ICN2037BP `

It's means that you have a standard LED driver.  You don't need any special driver settings for this panel like 
```
mxconfig.driver = HUB75_I2S_CFG::FM6126A; 
```

mibar-me, 2024-12-11:
> Check the reliability of the connection of this line along the entire signal path - starting from the pin on the ESP32 board, then the connectors on the flat cable, and then the reliability of the [...]

Will do! Thank you. I wanted to try out shorter cables any. In doing so I can also test out the D pin connection. Maybe I can also try out another similar panel then.

This happend when I touched the connection now: (it was flickery however they all lit up (currently only the R1, G1, B1 pins are connected to my ESP))

[image]

> It's means that you have a standard LED driver. You don't need any special driver settings

Okay. Sounds logic to. However, it's weird to me that the manufacturer does this in its demo projects as well. They also include the wrong driver then:

[source](https://files.waveshare.com/upload/b/b6/RGB-Matrix-P2.5-64x32-Demo.zip)
```C
// from waveshare's ESP32/SimpleTestShapes demo
void setup() {

  // Module configuration
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // module width
    PANEL_RES_Y,   // module height
    PANEL_CHAIN    // Chain length
  );

  mxconfig.clkphase = false;
  mxconfig.driver = HUB75_I2S_CFG::FM6047;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  // dma_display->shiftDriver(mxconfig.driver);
  dma_display->begin();
  // dma_display->shiftDriver();
  dma_display->setBrightness8(255); //0-255
  dma_display->clearScreen();
  dma_display->fillScreen(myWHITE);

  // ....
}
```

mibar-me, 2024-12-12:
Today I tried out another set of cables, two other ESPs and I even had the opportunity to test another LED matrix (same model, same manufacturer) with my code. It worked. So this definitely has to be some kind of faulty hard ware on my end. Since it only worked when I used the included 16 pin cable from a friend of mine (who uses the same LED matrix for his uni project) there has to be something going on with my cable. However it also didn't work with another set of 16 single jumper cables.

For anyone who stumbles accross this issue in the future and uses the same LED matrix this code worked in the end (just not with my cable/matrix/whatever due to faulty hard ware):
```C
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
#include "esp_log.h"

MatrixPanel_I2S_DMA *dma_display = nullptr;
uint16_t myColor  = dma_display->color565(0, 0, 255);
uint16_t myColor2 = dma_display->color565(255, 0, 0);

#define my_R1_PIN 25
#define my_G1_PIN 27 // (26) falsely wired
#define my_B1_PIN 26 // (27) falsely wired
#define my_R2_PIN 14
#define my_G2_PIN 13 // (12) falsely wired
#define my_B2_PIN 12 // (13) falsely wired
#define my_A_PIN  23
#define my_B_PIN  22
#define my_C_PIN  5
#define my_D_PIN  17
#define my_E_PIN  32
#define my_LAT_PIN 4
#define my_OE_PIN  15
#define my_CLK_PIN 16

HUB75_I2S_CFG::i2s_pins _pins={
    my_R1_PIN,
    my_G1_PIN,
    my_B1_PIN,
    my_R2_PIN,
    my_G2_PIN,
    my_B2_PIN,
    my_A_PIN,
    my_B_PIN,
    my_C_PIN,
    my_D_PIN,
    my_E_PIN,
    my_LAT_PIN,
    my_OE_PIN,
    my_CLK_PIN
};

extern "C" void app_main() {

  ESP_LOGI("WAITING", "for panel");
  delay(1000);

  HUB75_I2S_CFG mxconfig(/* width = */ 64, /* height = */ 32, /* chain = */ 1, _pins);
  mxconfig.clkphase = false;
  /* Other config parameters which were helpful to be set in other issues
  mxconfig.gpio.e = my_E_PIN;
  mxconfig.latch_blanking = 4;
  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;
  mxconfig.clkphase = true;*/

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  dma_display->begin();
  dma_display->setPanelBrightness(10);
  dma_display->clearScreen();
  dma_display->fillScreen(myColor2);

  while(1) {
    dma_display->fillScreen(myColor);
    ESP_LOGI("Fillscreen", "myColor");
    delay(500);
    dma_display->fillScreen(myColor2);
    ESP_LOGI("Fillscreen", "myColor2");
    delay(500);
    dma_display->clearScreen();
  }

}
```

Thanks to everyone who helped me debugging yesterday. :) I will mark this as closed now since my problem has nothing to do with this library.

closed 2024-12-12 by mibar-me