mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #707: SM16395 + SM5368PS 1/16 OUTDOOR & DP3364 + DP32020A-B 1/32 INDOOR Panels Help

nur476, 2024-12-16:
Hello,

I have these two P2.5 panels, and I need help about this chips is good for this library? I couldn't make outdoor panel work and not tested indoor panel yet. I will return these panels if they are not supported by this library. If supported please let me know.

Thanks in advance.

board707, 2024-12-16:
Unfortunately, DP3364 is unsupported driver type; as about a SM16395 driver, I can say nothing about it

nur476, 2024-12-23:
Not working. Closed.

closed 2024-12-23 by nur476

mrcodetastic, 2026-02-05:
Not supported. These are PWM chip based panels.

changathi123, 2026-02-07:
@board707 @mrcodetastic #include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

/* =========================================================
   SIMPLE Virtual Panel (no remap for full-screen test)
   ========================================================= */
class SimplePanel : public VirtualMatrixPanel {
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel;
};

/* =========================================================
   PANEL CONFIG
   ========================================================= */
#define PANEL_RES_X 80
#define PANEL_RES_Y 40

#define NUM_ROWS 1
#define NUM_COLS 1
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_NONE

/* =========================================================
   PIN DEFINITIONS
   ========================================================= */
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13

#define A_PIN  23
#define B_PIN  19
#define C_PIN  5
#define D_PIN  18

#define CLK_PIN 21
#define LAT_PIN 4
#define OE_PIN  15

MatrixPanel_I2S_DMA *dma_display = nullptr;
SimplePanel         *vpanel      = nullptr;

/* =========================================================
   COLOR TABLE (7 COLORS)
   ========================================================= */
uint16_t colors[7];

int colorIndex = 0;
unsigned long lastChange = 0;

/* =========================================================
   SETUP
   ========================================================= */
void setup() {
  HUB75_I2S_CFG cfg(
    PANEL_RES_X * 2,
    PANEL_RES_Y / 2,
    NUM_ROWS * NUM_COLS
  );

  // RGB
  cfg.gpio.r1 = R1_PIN;
  cfg.gpio.g1 = G1_PIN;
  cfg.gpio.b1 = B1_PIN;
  cfg.gpio.r2 = R2_PIN;
  cfg.gpio.g2 = G2_PIN;
  cfg.gpio.b2 = B2_PIN;

  // Address
  cfg.gpio.a = A_PIN;
  cfg.gpio.b = B_PIN;
  cfg.gpio.c = C_PIN;
  cfg.gpio.d = D_PIN;

  // Control
  cfg.gpio.clk = CLK_PIN;
  cfg.gpio.lat = LAT_PIN;
  cfg.gpio.oe  = OE_PIN;

  cfg.driver   = HUB75_I2S_CFG::ICN2038S;
  cfg.clkphase = true;

  dma_display = new MatrixPanel_I2S_DMA(cfg);
  dma_display->begin();

  // ⚠️ All LEDs ON → keep brightness safe
  dma_display->setBrightness8(30);
  dma_display->clearScreen();

  vpanel = new SimplePanel(
    *dma_display,
    NUM_ROWS,
    NUM_COLS,
    PANEL_RES_X,
    PANEL_RES_Y,
    VIRTUAL_MATRIX_CHAIN_TYPE
  );

  // Define 7 colors
  colors[0] = vpanel->color565(255,   0,   0);   // Red
  colors[1] = vpanel->color565(  0, 255,   0);   // Green
  colors[2] = vpanel->color565(  0,   0, 255);   // Blue
  colors[3] = vpanel->color565(255, 255,   0);   // Yellow
  colors[4] = vpanel->color565(  0, 255, 255);   // Cyan
  colors[5] = vpanel->color565(255,   0, 255);   // Magenta
  colors[6] = vpanel->color565(255, 255, 255);   // White

  lastChange = millis();
}

/* =========================================================
   LOOP — COLOR CYCLE
   ========================================================= */
void loop() {
  unsigned long now = millis();

  if (now - lastChange >= 2000) {   // 2 seconds
    lastChange = now;

    vpanel->fillScreen(colors[colorIndex]);
    dma_display->flipDMABuffer();

    colorIndex++;
    if (colorIndex >= 7) colorIndex = 0;
  }
}
this code works fine, but in case of any text ,text is broken and scrampled, i don't know exactly how iam missing, i tried so many ,still same