mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #708: Help me, i use the same the library but it can't display led matrix outdoor scan 1/8

Tin12185, 2024-12-18:
This is my code for p5 outdoor, but it can't display.
I checked pin. Indoor is Ok, but outdoor not Ok
#pragma once

#define R1_PIN_DEFAULT  25
#define G1_PIN_DEFAULT  26
#define B1_PIN_DEFAULT  27
#define R2_PIN_DEFAULT  14
#define G2_PIN_DEFAULT  12
#define B2_PIN_DEFAULT  13

#define A_PIN_DEFAULT   32
#define B_PIN_DEFAULT   33
#define C_PIN_DEFAULT   21
#define D_PIN_DEFAULT   17
#define E_PIN_DEFAULT   -1 // IMPORTANT: Change to a valid pin if using a 64x64px panel.
            
#define LAT_PIN_DEFAULT 4
#define OE_PIN_DEFAULT  22
#define CLK_PIN_DEFAULT 16

void setup()
{
  //Serial.begin(9600);
  Serial.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);
  //=========================================================================================================
  // EEProm_conver();
  //=========================================================================================================
  HUB75_I2S_CFG mxconfig(
      PANEL_RES_X * 2,    // DO NOT CHANGE THIS
      PANEL_RES_Y / 2,    // DO NOT CHANGE THIS
      NUM_ROWS * NUM_COLS // DO NOT CHANGE THIS
                          //,_pins            // Uncomment to enable custom pins
  );
  mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  mxconfig.clkphase = false;                    // Change this if you see pixels showing up shifted wrongly by one column the left or right.
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(192);    // range is 0-255, 0 - 0%, 255 - 100%
  dma_display->begin();
  // create VirtualDisplay object based on our newly created dma_display object
  virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, CHAIN_TOP_RIGHT_DOWN);
  // THE IMPORTANT BIT BELOW!
  virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);
  // Set Font and color 
  virtualDisp->setFont(&FreeSerif24pt7b);
  virtualDisp->setTextColor(virtualDisp->color565(255, 0, 0));
  virtualDisp->setTextSize(2);

  virtualDisp->setCursor(29, 59);           
  virtualDisp->print("123");
  //=========================================================================================================
  mb.begin(&Serial, REDE_PIN);
  // mb.begin(&Serial);
  mb.slave(SLAVE_ID);
  mb.addHreg(0, 0, 20);
  // id
  mb.Hreg(10, 1);
}

board707, 2024-12-18:
You provided a very little information about your setup.
You did not say what matrix, what drivers are located on it.
Also your code is incomplete, for example, the matrix size is not specified.

Tin12185, 2024-12-18:
#pragma region thư viện 
//library Matrix
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
#include <Fonts/FreeSerif24pt7b.h>
#include <Fonts/FreeMono9pt7b.h>
//library Modbus RS485
#include <ModbusRTU.h>

#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
//#define PANEL_CHAIN 1  // Total number of panels chained one to another
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 6 // Number of INDIVIDUAL PANELS per ROW

#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_RIGHT_UP

// MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;

VirtualMatrixPanel *virtualDisp = nullptr;

ModbusRTU mb;

const int resetW5500 = 15;

// EEProm
#include <EEPROM.h>         // libraries EEPROM
#define FLASH_MEMORY_SIZE 8

#pragma endregion

#pragma region Khai báo
int Data = 0;
int DataModbus[10] = {0,0,0,0,0,0,0,0,0,0};
uint8_t octet = 1;
IPAddress IP_Address;
const int resetip = 34;
unsigned long buttonPressStartTime = 0;
bool isButtonPressed = false;
bool buttonState = false;
bool eth_status = true;
#pragma endregion

#define SLAVE_ID 1
#define RX_PIN 1   // 17
#define TX_PIN 3   // 16
#define REDE_PIN 2

/**
 * các hàm xử lý
 */
static void EEProm_conver();
static void Display3_Up(int messageIndex);
static void Display3_Down(int messageIndex);
static void Display1_2_Up(int messageIndex);
static void Display1_2_Down(int messageIndex);
static void Mapping();
// static void HardwareStatus();
static void Delay(float delayTime, void(func)());
static void Modbusconnect();
// setting

void setup()
{
  //Serial.begin(9600);
  Serial.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);
  //=========================================================================================================
  // EEProm_conver();
  //=========================================================================================================
  HUB75_I2S_CFG mxconfig(
      PANEL_RES_X * 2,    // DO NOT CHANGE THIS
      PANEL_RES_Y / 2,    // DO NOT CHANGE THIS
      NUM_ROWS * NUM_COLS // DO NOT CHANGE THIS
                          //,_pins            // Uncomment to enable custom pins
  );
  mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  mxconfig.clkphase = false;                    // Change this if you see pixels showing up shifted wrongly by one column the left or right.
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(192);    // range is 0-255, 0 - 0%, 255 - 100%
  dma_display->begin();
  // create VirtualDisplay object based on our newly created dma_display object
  virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, CHAIN_TOP_RIGHT_DOWN);
  // THE IMPORTANT BIT BELOW!
  virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);
  // Set Font and color 
  virtualDisp->setFont(&FreeSerif24pt7b);
  virtualDisp->setTextColor(virtualDisp->color565(255, 0, 0));
  virtualDisp->setTextSize(2);

  virtualDisp->setCursor(29, 59);           
  virtualDisp->print("123");
  //Display3_Up(123);
  //=========================================================================================================
  //Khởi tạo Modbus Rs485
  mb.begin(&Serial, REDE_PIN);
  // mb.begin(&Serial);
  mb.slave(SLAVE_ID);
  mb.addHreg(0, 0, 20);
  // id
  mb.Hreg(10, 1);
}

Hi, @board707. I want to test to display 123 (setup), but it is not work. I don't know where errors.
[image]

board707, 2024-12-18:
Please try to read an ID on every chip on the panel rear side.

If you need help, please provide a more details about the issue, the phrase "it is not work" is not very infornative.

closed 2025-01-06 by mrcodetastic