mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #710: Displaying wrong columns with board P2 128*64 (ICND2045)

haibluehx95, 2024-12-20:
I try to config with HUB75_I2S_CFG::DP3246_SM5368, when I draw position column 0, it will display in the 63rd column, when drawing in column 1 it will display column 2. 
is there any way to fix it? Thank you!!!
[image]

board707, 2024-12-20:
Please show a photo with full view of panel rear side

haibluehx95, 2024-12-20:
[image]
This is the back of the panel, utilizing the ICND2045 and TC7558 chips.

board707, 2024-12-20:
Your panel has a shift register type line driver TC7558, that is not fully supported yet.
To use it, the library code should be edited.

haibluehx95, 2024-12-23:
I based this on the function dp3246init with the following modifications:
Set m_cfg.clkphase to false.
Defined the REG1 array as:
bool REG1[16] = {1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 0, 0, 0, 0}; // MSB first
Defined the REG2 array as:
bool REG2[16] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0}; // MSB first
It is work, thanks.

board707, 2024-12-23:
> I based this on the function dp3246init... It is work

Very surprising.
This function could not help you to work correctly with the TC7558 line driver. 
Maybe you mixed up the labels on the IC and there are other chips there?

closed 2025-01-06 by mrcodetastic