mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #713: Waveshare ESP32 S3 R2

B-Zimmer, 2024-12-29:
Hello!

I am attempting to use this library with the "Waveshare ESP32-S3 SIM7670G" board and a FM6124A driven panel.

[Waveshare Link](https://www.waveshare.com/product/iot-communication/long-range-wireless/4g-gsm-gprs/esp32-s3-sim7670g-4g.htm )

I have had success using the sample code provided in this library with a ESP32-WROOM2D using the same panel. (mostly the sample patterns and the pattern plasma examples) however I can't seem to get any output on the panel at all when using the waveshare board.

I believe the waveshare is an ESP32-S3 R2 chip with a xtensa 32 bit LX7 dual core processor, I dont think this board is RISK-V (saw the warning in the code re RISK-V and LCD mode requirements). I am using the arduino ide with the correct board, port and libraries selected. The only changes I have made to the sample code between the two ESP32 boards are pin definitions, and driver selection. I have confirmed the waveshare is functional, using sample code and flashing external LEDs on some of the same GPIO ports I am using to drive the HUB75 output. I am using GPIO pins on the waveshare, I've avoided SPIO and pins used by other on board hardware. I've verified the code is running through use of the serial monitor. The code is not using the PSRAM/SPIRAM DMA buffer.

Note: Camera hardware is physically removed, I confimred with Waveshare support it has to be this way in order to use the camera's GPIO ports for other devices.

Has anyone used this board (or similar) with this library? if anyone has any advice, suggestions, potential problems or solutions that would be amazing. I am happy to send more information if there is something I have missed.

Thank you for your time, I apologise if the link is formatted incorrectly.

B-Zimmer, 2025-01-05:
Github dosent seem to have a direct messaging system. I would be happy to donate a board to "mrcodetastic" if you have any interest. 

Thank you. No pressure. All the best.

mrcodetastic, 2025-01-05:
Hi B-Zimmer. 

I would be absolutely most interested! Depending on where you are I'm happy to just 'borrow' one of these boards to try get it working. Please feel free to send me an email to codetastic@tty.us.to

closed 2025-01-06 by mrcodetastic