mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #715: Information Color Calibration

Hypogluco, 2024-12-30:
Thanks for your work, I use this library on several screens and overall I am satisfied with it. I even made my PCBs for this library.

For the test I used 5 different screens

1 screen 64x64 P3, everything is ok
2 screens: 32X64 P5, size 320X160, everything is ok
1 screen: 32X64 P3, size 192x96, everything is ok

**The only screen that gives me problems is the: 32x64 P2.5, size 160x80. Maybe it's not really compatible with the library.**

I made a code to visualize the colors from the terminal and it's wrong.

myRED = OK
myGREEN = BLUE
myBLUE = GREEN
myYELLOW = PINK
myORANGE = PURPLE
myPURPLE = YELLOW / KAKI
myCYAN = OK
myWHITE = OK

If you have an idea, that would be great

Thanks 

The code to test colors
**************************

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_RES_X 64
#define PANEL_RES_Y 32
#define PANEL_CHAIN 1

MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t myBLACK, myWHITE, myRED, myGREEN, myBLUE, myYELLOW, myORANGE, myPURPLE, myCYAN;

void setup() {
  Serial.begin(115200);

  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // width
    PANEL_RES_Y,   // height
    PANEL_CHAIN    // chain length
  );

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(128); // 0-255

  myBLACK = dma_display->color565(0, 0, 0);
  myWHITE = dma_display->color565(255, 255, 255);
  myRED = dma_display->color565(255, 0, 0);
  myGREEN = dma_display->color565(0, 255, 0);
  myBLUE = dma_display->color565(0, 0, 255);
  myYELLOW = dma_display->color565(255, 255, 0);
  myORANGE = dma_display->color565(255, 128, 0);
  myPURPLE = dma_display->color565(128, 0, 128);
  myCYAN = dma_display->color565(0, 255, 255);

  Serial.println("Début du test de couleurs");
}

void loop() {
  testSolidColors();
  testColorBars();
  testColorGradient();
  testColorMix();
}

void testSolidColors() {
  uint16_t colors[] = {myRED, myGREEN, myBLUE, myYELLOW, myORANGE, myPURPLE, myCYAN, myWHITE};
  const char* colorNames[] = {"RED", "GREEN", "BLUE", "YELLOW", "ORANGE", "PURPLE", "CYAN", "WHITE"};

  for (int i = 0; i < 8; i++) {
    dma_display->fillScreen(colors[i]);
    Serial.println(colorNames[i]);
    delay(2000);
  }
}

void testColorBars() {
  dma_display->fillScreen(myBLACK);
  int barWidth = PANEL_RES_X / 8;
  uint16_t colors[] = {myRED, myGREEN, myBLUE, myYELLOW, myORANGE, myPURPLE, myCYAN, myWHITE};

  for (int i = 0; i < 8; i++) {
    dma_display->fillRect(i * barWidth, 0, barWidth, PANEL_RES_Y, colors[i]);
  }
  Serial.println("Color Bars");
  delay(5000);
}

void testColorGradient() {
  for (int i = 0; i < PANEL_RES_X; i++) {
    uint8_t r = map(i, 0, PANEL_RES_X - 1, 0, 255);
    uint8_t g = map(i, 0, PANEL_RES_X - 1, 255, 0);
    uint8_t b = 128;
    uint16_t color = dma_display->color565(r, g, b);
    dma_display->drawFastVLine(i, 0, PANEL_RES_Y, color);
  }
  Serial.println("Gradient Color");
  delay(5000);
}

void testColorMix() {
  dma_display->fillScreen(myBLACK);
  int centerX = PANEL_RES_X / 2;
  int centerY = PANEL_RES_Y / 2;
  int radius = min(PANEL_RES_X, PANEL_RES_Y) / 4;

  dma_display->fillCircle(centerX - radius/2, centerY, radius, myRED);
  dma_display->fillCircle(centerX + radius/2, centerY, radius, myBLUE);
  dma_display->fillCircle(centerX, centerY - radius/2, radius, myGREEN);

  Serial.println("Color Mix");
  delay(5000);
}

********************************************************

https://github.com/user-attachments/assets/1eda97c7-3651-473f-8c37-64543391933f

mrcodetastic, 2025-01-02:
The panel is defective or the masking printed on the pc is wrong.  Flip the green and blue gpio pin mappings in your code for this panel.

mrcodetastic, 2025-01-02:
The HUB75 pins are different for this panel.  Simply flip the green and blue gpio pin mappings in your code for this panel.

Hypogluco, 2025-01-02:
> The HUB75 pins are different for this panel. Simply flip the green and blue gpio pin mappings in your code for this panel.

Thank you, I will try and come back.

Hypogluco, 2025-01-03:
Je reviens pour vous dire que le problème est résolu avec la configuration :

 #pragma once
// Pin pour les couleurs (R, G, B)
#define R1_PIN_DEFAULT  25
#define B1_PIN_DEFAULT  5   // Inversion Pin de G1 et B1
#define G1_PIN_DEFAULT  13  // Inversion Pin de G1 et B1
#define R2_PIN_DEFAULT  4
#define B2_PIN_DEFAULT  12  // Inversion Pin de G2 et B2
#define G2_PIN_DEFAULT  14  // Inversion Pin de G2 et B2

// Broches pour l'adressage des lignes (A, B, C, D, E)
#define A_PIN_DEFAULT   15
#define B_PIN_DEFAULT   16
#define C_PIN_DEFAULT   2
#define D_PIN_DEFAULT   18
#define E_PIN_DEFAULT   -1 // IMPORTANT: Change to a valid pin if using a 64x64px panel.

// Broches de contrôle (LAT, OE, CLK)
#define LAT_PIN_DEFAULT 17
#define OE_PIN_DEFAULT  23
#define CLK_PIN_DEFAULT 19

Merci beaucoup

closed 2025-01-06 by mrcodetastic