mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #717: 128x128 panel question

krulkip, 2025-01-02:
I have the above panel. It is driven by two separate HUB75 connectors. 
This means i can drive each half separately. It has HX6158SP and ICN2037BP drivers.
Using the program below i can light up some of the screen but it does not behave how i would like.
The drawPixel command does not show anything. The drawLine command does. However if i try to draw a short line of one Pixel it repeats the pixel 16 times and leaves a one pixel gap. If the line is 4 pixels long it is also repeated but no gap. See picture.
The panel had the A and B ports swapped so A goes to pin 19 and B to 23. 
[
[image]
](url)
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
const int panelResX = 128;      // Number of pixels wide of each INDIVIDUAL panel module. 
const int panelResY = 64;     // Number of pixels tall of each INDIVIDUAL panel module.
const int panel_chain = 1;      // Total number of panels chained one to another
MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup() {

  HUB75_I2S_CFG mxconfig(
    panelResX,   // module width
    panelResY,   // module height
    panel_chain    // Chain length
  );
  //mxconfig.double_buff = true;
  mxconfig.gpio.a = 19;
  mxconfig.gpio.b = 23;
  mxconfig.clkphase = false;
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  Serial.begin(115200);
  dma_display->begin();
  dma_display->clearScreen();
  for (int i=0;i<64;i++){
  dma_display->drawLine(i,0,i,0,dma_display->color565(0, 255, 0));
  dma_display->drawLine(i,32,i,32,dma_display->color565(255, 0, 0));
  }
  for (int i=64;i<128;i++){
  dma_display->drawLine(i,0,i,4,dma_display->color565(0, 255, 0));
  dma_display->drawLine(i,32,i,36,dma_display->color565(255, 0, 0));
  }
}

void loop() {
}

board707, 2025-01-06:
> quick google seems to indicate the HX chip means this panel is a PWM frame buffer style

It seems to me that your googling is not correct.
HX chip is a line driver (shift register type) and the 2037 is a standard led driver chip.
The panel has a good chance to be compatible with your library after completing the support of SR type line drivers.

krulkip, 2025-01-07:
Wrote a small driver program to understand how HX6158 works.
Turns out is uses a clock in solution for selecting the segment number.
Set a to e segment selection sequentially and clock on c pin.
See how it works.
I guess that means it is needed to change the driver in the segment selection rather than fiddle with virtual panel mapping.
Can anyone help.

// This is for a 128x64 pixel LED Matrix display for FM6124 and half of 128x128 for HX6158
// 1. Set all pins to LOW.
// 2. Set R1, G1, B1, R2, G2, and B2 to LOW or HIGH. This allows you to specify the color for two pixels separately. RGB1 and RGB2 refer to one half of the display.
// 3. Set CLK to HIGH.
// 4. Set CLK to LOW. The 2 pixels are determined to be the color specified by 2 by 3 and 4, and the next pixel is specified.
// 5. Repeat steps 2 to 4 a total of 128 times. This completes writing one section of data.
// 6. Use the 5 bits ABCDE to determine the section you want to send. For example, start with 00000. (There are 32 sections for R1, G1, B1 and 32 for R2, G2, B2.)
//    In HX6158 you clock in the ABCDE bits using a pin as clock and c pin as value. The a and b pins are labelled opposite.
// 7. Set OE to HIGH. It is the flag to enable screen display. High disables it.
// 8. Set LAT to HIGH. 
// 9. Set LAT to LOW. This clocks in the rgb settings into the segment
//10. Set OE to LOW. This lights up the screen with all LED's set as cloclked.
//11. Repeat steps 2 to 10 32 times, changing sections at a time. By repeating this at high speed, you can display the entire screen.

#define r1  25
#define g1  26
#define b1  27
#define r2  14
#define g2  12
#define b2  13
#define  a  23 // HX6158 panel has pin23 labelled b
#define  b  19 // HX6158 panel has pin19 labelled a
#define  c   5
#define  d  17
#define  e  18
#define clk 16
#define lat  4
#define oe  15

int block1[128],block2[128];
int pins[]={r1,g1,b1,r2,g2,b2,e,a,b,c,d,clk,lat,oe};
int leng=sizeof(pins)/sizeof(int);
int segment=0;
int rgb1,rgb2;
// select multiplexer type
//#define FM6124
#define HX6158

void setup() {
  // put your setup code here, to run once:
  for(int i=0;i<leng;i++)
  {
    pinMode(pins[i],OUTPUT);
    digitalWrite(pins[i],LOW);
  }
  for(int i=0;i<128;i++)
  {
    // Create color pattern on screen by row
    // block1 and block2 correspond to rgb1 and rgb2 on the display
    // Pattern Blue  White  Red   Green
    //         Green  Red  Green   Blue
    // 1 = Blue, 2 = Green, 4 = Red
    if (i<32) {block1[i]=1;block2[i]=2;}
    if ((i<64) && (i>=32)) {block1[i]=2;block2[i]=4;}
    if ((i<96) && (i>=64)) {block1[i]=4;block2[i]=7;}
    if (i>=96) {block1[i]=2;block2[i]=1;}
  }
}

void loop() {
      digitalWrite(oe,HIGH);//turn screen off
      digitalWrite(lat,LOW);// prepare latch to clock in colours
      digitalWrite(a,LOW);// for HX6158 start a LOW
      digitalWrite(c,LOW);// for HX6158 start c LOW
      for(int i=0;i<128;i++)
      {
        rgb1=block1[i];rgb2=block2[i];
        // r1, g1 and b1 are for bottom half of display
        digitalWrite(r1,rgb1&4);
        digitalWrite(g1,rgb1&2);
        digitalWrite(b1,rgb1&1);
        // r2, g2 and b2 are for top half of display
        digitalWrite(r2,rgb2&4);
        digitalWrite(g2,rgb2&2);
        digitalWrite(b2,rgb2&1);
        //Color clock
        digitalWrite(clk,HIGH);
        digitalWrite(clk,LOW);
      }
      #ifdef HX6158 
         // set up segment
         digitalWrite( c, ( segment & 1 ) );clocka();
         digitalWrite( c, ( segment & 2 ) >> 1);clocka();
         digitalWrite( c, ( segment & 4 ) >> 2);clocka();
         digitalWrite( c, ( segment & 8 ) >> 3);clocka();
         digitalWrite( c, ( segment & 16 ) >> 4);clocka();
      #endif
      #ifdef FM6124
         // set up segment
         digitalWrite( a, ( segment & 1 ) );
         digitalWrite( b, ( segment & 2 ) >> 1);
         digitalWrite( c, ( segment & 4 ) >> 2);
         digitalWrite( d, ( segment & 8 ) >> 3);
         digitalWrite( e, ( segment & 16 ) >> 4);
      #endif
      //segment latch and screen show
      digitalWrite(lat,HIGH);// latch segment 
      digitalWrite(oe,LOW);// light up screen      
      delayMicroseconds(100);// This controls brightness
      segment++;
      if(segment==32) // RGB1 and RGB2 each give half display so only have 32 segments
      {
        segment=0;
      }
}

void clocka (){
       digitalWrite( a, HIGH);
       digitalWrite( a, LOW);
}

[image]

jnthas, 2025-04-27:
Hello. I've been working with those 64x64 led matrices for a few years and I've never had any iproblem. This week I received one that didn't work with the brian's Trinity board, which is 100% compatible with these kind of HUB75 displays.

The display label is:
[image]

The chips that controls the panel are TC7262 (which it seems compatible) but the other one is, HX8863. I couldn't find much about it. Since if starts with HX, I'm asking in this thread. 

[image]
[image]

Besides no images at all, I noticed that when I connect the Trinity board in it, both the display and the trinity boards gets very hot.

Can you confirm the HX chip is not compatible?

board707, 2025-04-28:
See #694

closed 2025-05-04 by mrcodetastic