mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #719: Panel skipping columns on 104*52 matrix

alberplayer, 2025-01-03:
When trying to draw graphics it works fine in general but (0,0) draws in what should be (6,0), then it skips drawing the 15th column and skips every 16 columns until the end and then the first 6 columns to the left copy what is shown in the last 6 to the right. I try to show diagonal lines but they appear jagged. For testing I also try to draw green pixels in columns 8, 24, 40, 56, 72 and 88 but they don't show up at all.

I have tried the fixes in the Common Issues but none have worked for me.
I think the chip is ICN2037BP.
[This is the panel I bought.](https://www.aliexpress.com/item/1005008226958496.html)
[image]
[image]

Code I used for testing:
```
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Adafruit_GFX.h>

#define PANEL_RES_X 104      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 52    // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another

//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *matrix = nullptr;

uint16_t myBLACK, myWHITE, myRED, myGREEN, myBLUE;

void setup() {
  Serial.begin(9600); // open the serial port at 9600 bps:
// Module configuration
// Change these to whatever suits
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 19
#define C_PIN 5
#define D_PIN 21
#define E_PIN 32 // required for 1/32 scan panels, like 64x64px. Any available pin would do, i.e. IO32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16

HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};

  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // module width
    PANEL_RES_Y,   // module height
    PANEL_CHAIN,    // Chain length
	  _pins // pin mapping
  );
  mxconfig.clkphase = false;

  // Display Setup
  matrix = new MatrixPanel_I2S_DMA(mxconfig);
  matrix->begin();
  matrix->setBrightness8(20); //0-255
  matrix->clearScreen();

  myBLACK = matrix->color565(0, 0, 0);
  myWHITE = matrix->color565(255, 255, 255);
  myRED = matrix->color565(255, 0, 0);
  myGREEN = matrix->color565(0, 255, 0);
  myBLUE = matrix->color565(0, 0, 255);

  matrix->drawPixel(0, 0, myBLUE);
  delay(1000);
  Serial.println("W:" + String(matrix->width()));
  Serial.println("H:" + String(matrix->height()));
  matrix->drawLine(1, 1, matrix->width()-1, matrix->height()-1, myGREEN);
  matrix->drawLine(matrix->width()-1, 0, matrix->width()-1, matrix->height()-1, myRED);
  matrix->drawLine(1, 1, 50, 50, myWHITE);
  matrix->drawLine(51, 1, 100, 50, myWHITE);
  matrix->drawLine(101, 1, 150, 50, myWHITE);
  matrix->drawPixel(8, 8, myBLUE);
  matrix->drawPixel(24, 24, myBLUE);
  matrix->drawPixel(40, 40, myBLUE);
  
  matrix->drawPixel(4, 8, myRED);
  matrix->drawPixel(4, 24, myRED);
  matrix->drawPixel(4, 40, myRED);
  for (int i=8; i<matrix->width(); i = i + 16) {
   matrix->drawPixel(i, 2, myGREEN);
  }
}

void loop() {
}

```

alberplayer, 2025-01-03:
Is there a way to automatically skip the columns that are can't be rendered so they show on the next column in order to make it work? 
Like simulating the matrix being wider than it actually is and shifting the output every time it goes over a faulty column. 
Something like this (just to visualize it in case I haven't explained myself well enough):
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Alberti/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Alberti/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#467886" vlink="#96607D">

Supposed Output | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15 | 16 |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Actual Output | 1 | 2 |   | 3 | 4 |   | 5 | 6 |   | 7 | 8 |   | 10 | 11 |   | 12 | 13 | 14 | 15 | 16

</body>

</html>
Assuming the faulty columns are the ones supposed to show 3, 6, 9, 12 and 15.

For my project I don't need a high refresh rate so it could be a workaround because I don't think I have the knowledge to figure out the specific scanning pattern of this panel in order to modify the library for it to work correctly.
Any help is appreciated.

mrcodetastic, 2025-01-04:
You should not need to modify the library other than use the virtual matrix class Aug your own pixel mapping logic. 

What happens if you do a fillscreen? Does that light up all LEDs?

mrcodetastic, 2025-01-04:
You should not need to modify the library other than use the virtual matrix class with your own pixel mapping logic. 

What happens if you do a fillscreen? Does that light up all LEDs?

mrcodetastic, 2025-01-04:
Also, printing lines or dots can be misleading.  What happens if you print text?

Setting the panel resolution in the library to the pixel width and height of the physical panel may not be right.  It could be in reality,  a panel with half the real height and twice the length (for example).

alberplayer, 2025-01-04:
> Also, printing lines or dots can be misleading. What happens if you print text?
> 
> Setting the panel resolution in the library to the pixel width and height of the physical panel may not be right. It could be in reality, a panel with half the real height and twice the length (for example).

I ran some of the example codes
1_SimpleTestShapes:
[image]
[image]
With setTextSize(2);
[image]
/
2_PatternPlasma:
Fist with
```
#define PANEL_RES_X 104      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 52     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another
```
/
[image]

Then with
```
#define PANEL_RES_X 104      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 52     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 2      // Total number of panels chained one to another
```
[image]

I also tried to set it as a 128x64 panel just to try and I got this result:
[image]

alberplayer, 2025-01-04:
> You should not need to modify the library other than use the virtual matrix class with your own pixel mapping logic.
> 
> What happens if you do a fillscreen? Does that light up all LEDs?

How would I add the custom mapping logic? I can't find anything other than the Chained Panels tutorial.

mrcodetastic, 2025-01-06:
Looking at the screenshots where you've changed the resolution to the different to the physical number of LEDs... it actually looks promising. 

That is, when you set the width to 128 - it crops output to the left, but when it's 104, it's displaying stuff indended to the right (from the left).

So it seems like this weird panel has a DMA data 'width' <> physical LED width. Perhaps it's 111 pixels wide, 55 pixels high? Need to experiment, repeat.

mrcodetastic, 2025-01-06:
Looking at the screenshots where you've changed the resolution to be different to the physical number of LEDs... it actually looks promising. 

That is, when you set the width to 128 - it crops output to the left, but when it's 104, it's displaying stuff indented to the right (from the left).

So it seems like this weird panel has a DMA data 'resolution' <> physical LED resolution. Perhaps it's 111 pixels wide, 55 pixels high? Need to experiment with the PANEL_RES_X and Y values, repeat.

alberplayer, 2025-01-06:
I finally figured it out.
I set the width to 110 and that worked to get the origin point in the correct position but there were still mising columns, but using the solution from #622 I figured out how to set up a Virtual Matrix class with the logic to make the output look nice.

```

// Example sketch which shows how to display some patterns
// on a 104x52 LED matrix
//

#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class ColumnSkipPanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// define custom getCoords() method for specific pixel mapping
inline VirtualCoords ColumnSkipPanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

  uint8_t offset = 0;  // offset
    if (coords.x >= 14) {
     offset = (coords.x - 14) / 15 + 1;  // Calcula el offset basado en bloques de 15
     //if (offset > 5) offset = 5;         // Limita el offset máximo a 5 (no hace falta porque limitaría el tamaño)
     coords.x += offset ; // adds the offset to the Co-ordinates
    }
  return coords;
}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 110 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 52 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
ColumnSkipPanel   *real_Panel = nullptr;

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);

// Input a value 0 to 255 to get a color value.
// The colours are a transition r - g - b - back to r.
[... 134 lines cut ...]

}

uint8_t wheelval = 0;
void loop() {

    // animate by going through the colour wheel for the first two lines
    drawText(wheelval);
    wheelval +=1;

    delay(20);
  
}

```
Before: (I used the counter to know what resolution I had set up)
[image]

After:
[image]

The vertical lines are the ones that wouldn't show up before:
[image]

Thanks a lot for the help, I couldn't have figured it out by myself.

board707, 2025-01-06:
> set the width to 110

Most probably it should be 112 ( to be multiple of 16), since the number of outputs of the led driver chip on the panel.
I am successfully worked with similar (but not the same) panel already.

alberplayer, 2025-01-06:
> > set the width to 110
> 
> Most probably it should be 112 ( to be multiple of 16), since the number of outputs of the led driver chip on the panel.
> I am successfully worked with similar (but not the same) panel already.

I think I tested it with 112 and it started to crop the pixels on the left. I will try again tomorrow just to make sure.
At first the 6 pixels on the left were the ones that copied what appeared on the right so that was the amount I added.

mrcodetastic, 2025-01-06:
On a random note, these are nice  panels I must say. A good balance between resolution and size for personal projects.