mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #720: Outdoor 80x40 1/10scan custom panel scan rate

Ogami-B, 2025-01-04:
Hi
I am trying to get an outdoor 80x40 1/10scan panel to work with this library and have gotten really close but just need a little bit of help for the last bit. with this code 
`#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

// Display settings
#define R1_PIN  22
#define G1_PIN  23
#define B1_PIN  25
#define R2_PIN  26
#define G2_PIN  27
#define B2_PIN  14

#define A_PIN   4
#define B_PIN   5
#define C_PIN   18
#define D_PIN   19
#define E_PIN   -1

#define CLK_PIN 21
#define LAT_PIN 32
#define OE_PIN  33
#define PANEL_RES_X 80
#define PANEL_RES_Y 40
#define NUM_COLS 1
#define NUM_ROWS 1
#define PANEL_CHAIN NUM_COLS * NUM_ROWS
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_LEFT_DOWN

// Colors
#define WHITE virtualDisp->color565(255, 255, 255)
#define BLACK virtualDisp->color565(0, 0, 0)
#define RED virtualDisp->color565(255, 0, 0)
#define GREEN virtualDisp->color565(0, 255, 0)
#define BLUE virtualDisp->color565(0, 0, 255)

MatrixPanel_I2S_DMA *dmaDisplay = nullptr;
VirtualMatrixPanel *virtualDisp = nullptr;

void testPixels() {
  virtualDisp->clearScreen();
  int h = virtualDisp->height();
  int w = virtualDisp->width();
  for (int i = 0; i < h; i++) {
    for (int j = 0; j < w; j++) {
      virtualDisp->drawPixel(j, i, RED);
      delay(10);
    }
  }
}

void setupScreen(int panel_x, int panel_y, int rows, int cols, PANEL_CHAIN_TYPE chain_type) {
  HUB75_I2S_CFG::i2s_pins _pins = {
        R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
        A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
        LAT_PIN, OE_PIN, CLK_PIN
       };
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // DO NOT CHANGE THIS
    PANEL_RES_Y / 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    ,_pins            // Uncomment to enable custom pins
  );

  dmaDisplay = new MatrixPanel_I2S_DMA(mxconfig);
  dmaDisplay->setLatBlanking(1);
  dmaDisplay->begin();
  dmaDisplay->setBrightness8(200);  // 0-255, values higher than 200 may introduce ghosting
  virtualDisp = new VirtualMatrixPanel((*dmaDisplay), rows, cols, panel_x, panel_y, chain_type);
  virtualDisp->setPhysicalPanelScanRate(FOUR_SCAN_40PX_HIGH);
  virtualDisp->clearScreen();
  virtualDisp->setCursor(0, 0);
}

void setup() {
  setupScreen(PANEL_RES_X, PANEL_RES_Y, NUM_ROWS, NUM_COLS, VIRTUAL_MATRIX_CHAIN_TYPE);
}

void loop() {
  testPixels();
  delay(2000);
}`
and this done in ESP32-VirtualMatrixPanel-I2S-DMA.h
`else if (panel_scan_rate == FOUR_SCAN_40PX_HIGH)
    {
        uint8_t pxbase = 8;
        if ((coords.y / 10) % 2 == 0)
        {
            coords.x += (coords.x / pxbase) * pxbase; 
        }
        else
        {
            coords.x += ((coords.x / pxbase) + 1) * pxbase;                                 //odd
        }
        coords.y = (coords.y / 20) * 10 + (coords.y % 10);
    }`
i get this result.

https://github.com/user-attachments/assets/1449fdb1-22bb-4d1e-a97a-da7a45c79e8e

it works prefectly from y 1-10 and 21-30 but doesnt in between.
I've tried to use the same method used in #622 to reverse the blocks but I've had no luck.
anyone got any ideas?

Ogami-B, 2025-01-05:
so i managed to get it working with the code from #637 but now I've tried chaining 2 together to make an 80x80 display and i get another artifact and im not sure whats going on.

https://github.com/user-attachments/assets/c0e74405-a33a-4441-a843-3585b0f3fea7

mrcodetastic, 2025-01-05:
` uint8_t pxbase = 8;` should be ` uint8_t pxbase = 10;` ?

board707, 2025-01-07:
@Ogami-B 
The pattern must works equally  for a single panel and for a chain as well.

Test what it will be if you swap first and second panels?

Try to tune a clockphase settings; the video looks like you have a one extra pulse in clock line.

Check the connection between first and second panel.

Ogami-B, 2025-01-07:
turns out all i had to do was change
`bool _clockphase = true`
to
`bool _clockphase = false`
in ESP32-VirtualMatrixPanel-I2S-DMA.h

thanks heaps!

board707, 2025-01-07:
Is the problem solved?

Ogami-B, 2025-01-07:
Yes, thanks again.

closed 2025-01-07 by Ogami-B