mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #721: HUB75E P3 ICND2055CP + RUC7258D Question

so1der, 2025-01-05:
Hello there! So i have this LED-P3-2121-64X64-32S-V3.0 panel, I tried some examples (and all mxconfig.driver variations), and I got no result - panel didn't response at all.
Am I doing something wrong, or it isn't supported, or something else?
IC on board are following:
74HC245K
74HC138T
RUC7258D
ICND2055CP
Since it's  HUB75**E**, I defined E pin, but it didn't help.

Some photos:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mrcodetastic, 2025-01-05:
This is a PWM based panel which is a different technology not supported by this library. A google for 'ICND2055CP' gives me [this link](https://olympianled.com/wp-content/uploads/2021/05/ICND2055_datasheet_EN_2021_V1.8b.pdf).

so1der, 2025-01-05:
Oh, damn. I did read about S-PWM/PWM chips in README, but thought that since ICND2055CP and supported ICND2012 are quite similar in naming, this is will work :D
Well, that's a shame. Thank You for your time though!

closed 2025-01-05 by so1der

mrcodetastic, 2025-01-05:
Are you in Ukraine? Slava Ukraini!

If you intend to buy many more of these, and can find a datasheet, I can take a look. All PWM panels are totally different in operation (namely the registers that need to be set), but I have a reasonable understanding of them now. I was gifted a bunch of e-waste PWM panels and spent considerable time writing a totally different code-base to drive these.

i.e. https://github.com/mrcodetastic/MBI5153-DMG1075-MatrixPanel-DMA (of no use to your current panel however)

so1der, 2025-01-05:
Yup, Im in Ukraine, Heroyam Slava! :)

Wooah that would be awesome!
I already bought 2 of these panels, they were almost 4 times cheaper then other ones (well, now I know why :D)
So about datasheet, the one You provided above won't work, right? It should be more detailed, and etc?
I'll do my best to find the datasheet for this chip and when I do I'll let You know!
Thanks!

mrcodetastic, 2025-01-06:
Got a link to where you bought one? 

I found a related discussion at the Raspberry PI LED project: https://github.com/hzeller/rpi-rgb-led-matrix/issues/466

mrcodetastic, 2025-01-06:
Got a link to where you bought one? 

I found a related discussion at the [Raspberry PI LED project](https://github.com/hzeller/rpi-rgb-led-matrix/issues/466) with a link to some random chinese datasheet with the register information needed (I have put this through google translate): [icn2053-translated.pdf](https://github.com/user-attachments/files/18312400/icn2053-translated.pdf)

Whilst the ICN2053 < 2055, it's probably the same anyway from a configuration register perspective.

Feel free to send me a panel if you want ;-) and I can (don't promise) to take a look. codetastic@tty.us.to

I think I should be able to pretty much re-use the code I wrote for the MBI5153 chip based PWM panels.

board707, 2025-01-06:
> Whilst the ICN2053 \u003C 2055, it's probably the same anyway from a configuration register perspective

I have both 2053 and 2055 panels and I have my own code for 2153/2053 driver, but it not suitable for 2055 panel. It is not just a configuration registers. The pulse diagram is rather different too.

so1der, 2025-01-06:
> Got a link to where you bought one?

Yup, but that just a random Ukrainian shop, that sells various things for microcontrollers. [Here's the link](https://arduino.ua/prod4428-svitlodiodni-matrici-p3-64x64-192h192mm-komplekt). But I'll ask them anyway, maybe they have a datasheet.

> Whilst the ICN2053 < 2055, it's probably the same anyway from a configuration register perspective.

I thought the same way and tried [this project](https://github.com/SebiTimeWaster/ICN2053_ESP32_LedWall). But yeah, just like **board707** stated above, this didn't work well. Panel did output some things, but it's just random mess.

> Feel free to send me a panel if you want ;-)

I'd love to! But I guess the cost of international shipping will be way to much :(

mrcodetastic, 2025-01-06:
Sounds like as per what @board707 said, these panels have a different signal timing method.

It's probably easier to just buy a different supported panel unless you want to go down the rabbit hole of reverse engineering a signal to one of these board from a controller card.

so1der, 2025-01-06:
Yes, I'll probably just buy a supported one and that's it.

About reverse engineering - this is quite an interesting topic, so I guess I'll save these panels until I come across one of those controller cards :)

Thanks again!