mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #725: 74hc245kaj Chip

gxsins, 2025-01-07:
[image]
[image]

i ordered a new set from Aliexpress, this time with a new chip that i havent seen before.

old chips were 
d18807na

new ones are 
74hc245kaj

what i've noticed is the bottom right portion of a 64X32 panel is good, and all top right \ bottom left  have a shift, then top left has a bigger shift

gxsins, 2025-01-07:
Here is a better picture.

i wrote 3 lines, RGB over and over.
Good chip vs new one.

[image]
[image]

code:

```
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h> // Library for controlling the LED matrix

// Panel configuration
#define PANEL_RES_X 64                  // Panel width in pixels
#define PANEL_RES_Y 32                  // Panel height in pixels
#define PANEL_CHAIN 1                   // Number of panels chained together

MatrixPanel_I2S_DMA *dma_display;       // Pointer to the display object

void setup() {
    // Define the GPIO pins for the matrix
    HUB75_I2S_CFG::i2s_pins _pins = {
        // R1, G1, B1, R2, G2, B2, A, B, C, D, E, LAT, OE, CLK
           25, 26, 27, 14, 12, 13, 23, 19, 5, 17, -1, 4, 15, 16
    };

    // Initialize the display configuration
    HUB75_I2S_CFG mxConfig(PANEL_RES_X, PANEL_RES_Y, PANEL_CHAIN);
    mxConfig.gpio = _pins; // Assign the custom pin configuration

    dma_display = new MatrixPanel_I2S_DMA(mxConfig);
    dma_display->begin(); // Initialize the display
    dma_display->setBrightness8(64); // Set medium brightness (scale of 0–255)
    dma_display->clearScreen(); // Clear the screen at startup
}

void loop() {
    dma_display->clearScreen(); // Clear the display at the start of the loop

    // Draw RGB vertical lines across the entire width of the panel
    for (int x = 0; x < PANEL_RES_X; x++) {
        uint32_t color;

        // Repeat the RGB pattern
        if (x % 3 == 0) {
            color = 0xFF0000; // Red
        } else if (x % 3 == 1) {
            color = 0x00FF00; // Green
        } else {
            color = 0x0000FF; // Blue
        }

        drawVerticalLine(x, color); // Draw the line
    }

    delay(5000); // Delay to observe the pattern
}

// **Helper Function: Draw Vertical Line**
void drawVerticalLine(int x, uint32_t color) {
    for (int y = 0; y < PANEL_RES_Y; y++) { // Loop through all rows
        dma_display->drawPixelRGB888(x, y, (color >> 16) & 0xFF, // Red
                                     (color >> 8) & 0xFF,       // Green
                                     color & 0xFF);            // Blue
    }
}

```

board707, 2025-01-07:
> new ones are 74hc245kaj

No, it is neither an driver nor line switch. It's just a buffer. The same presents on most panels.

Look on the panel for another chips - you should able to found a two another ones.