mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #727: Matrix 64x32 1/32

Plehanov1971, 2025-01-13:
Hello people! I'm trying to connect the matrix 64x32 to the board esp32 wroom. I used different libraries and all without success. The matrix is ​​very strange, the resolution is 64x32 and scanning 1/32. Previously everything was fine but with matrices with scanning 1/8, 1/16 with this resolution. Matrix or shows half screen or 2 versions of the top of the screen at the top and bottom. Maybe someone can suggest something?

board707, 2025-01-13:
@Plehanov1971 
You can easily work with a single panel if you setup it as 64x64 1/32. But you will not able to chain more than one.
As far I know, there is no library at present moment that could work with such panels.

Plehanov1971, 2025-01-13:
> @Plehanov1971 You can easily work with a single panel if you setup it as 64x64 1/32. But you will not able to chain more than one. As far I know, there is no library at present moment that could work with such panels.

Tell me the settings, please!

board707, 2025-01-13:
try to set a height to 64 and use Y only in range 0-31

mjmokhtar, 2025-01-14:
same like i've ask in this issu 
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/540 

since desember 2023

board707, 2025-01-14:
@Plehanov1971 
Could you write me a message to bort707h@gmail.com ?