mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #729: Examples not compiling with latest FastLED

DamianPe, 2025-01-15:
Hello,

I'm trying to compile examples for ESP32 HUB75 MatrixPanel DMA but it fails.
First of all, Readme does not mention one of dependences, which is FastLED - a minor issue but still worth correcting. More serious is that library does not compile with following error `Compilation error: too few arguments to function 'void blur2d(CRGB*, uint8_t, uint8_t, fract8, const fl::XYMap&)'`
Looking into the code, function call is `blur2d(effects.leds, MATRIX_WIDTH > 255 ? 255 : MATRIX_WIDTH, MATRIX_HEIGHT > 255 ? 255 : MATRIX_HEIGHT, 192);` so in fact, last parameter is ommited... 
Any chance to fix this issue or any advice which FastLED library version is known to work? I'm working with 3.9.11 but tried down to 3.9.0 with no success...

mrcodetastic, 2025-01-15:
What example doesn't compile?

zackees, 2025-01-15:
FastLED used to do something evil and rely on a weak symbol being linked in, `uint16_t XY(uint16_t x, uint16_t y);` that the user would supply.

Now the user passes in a functor. This was the only part of the API that I changed.

What shall we do about it?

DamianPe, 2025-01-15:
> What example doesn't compile? Use GFX _Lite instead which is a library I made that takes a cut of FastLED and merges it with a cut of AdaFruit_GFX.

AuroraDemo, errors are found in PatternSpiro.h file.

Also in PatternInvaders.h file there is an error message: `no matching function for call to 'MatrixPanel_I2S_DMA::fillRect(int, int, int, int, CRGB&)' dma_display->fillRect(1 + x * 6, 1 + y * 6, 5 + x * 6, 5 + y * 6, color);`

zackees, 2025-01-16:
Anyone know how to compile this?

zackees, 2025-01-16:
Fixed in FastLED 3.9.12 (current version is 3.9.11)

closed 2025-01-19 by mrcodetastic