mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #730: P5 64x32 with custom pixel mapping

GysJef, 2025-01-16:
I have ordered multiple led module samples from different suppliers.
The modules from supplier 2 are not working as expected, the mapping of the pixel is not standard.

I'm running this demo-code on an ESP32-S3-WROOM-1U with DMA support with pixel color depth bits set to 2.
The final setup will be a virtual matrix with 192 x 64 pixels.

### Overwiew

|             | Supplier 1 | Supplier 2 | Supplier 3 |
| ----------- | ---------- | ---------- | ---------- |
| Led Driver  | ICN2037BP  | ICN2037BP  | ICN2037BP  |
| Multiplexer | \-         | RUC7258E   | SM5166PS   |
| Transceiver | \-         | 74HC245KAJ | \-         |
| Working     | Yes        | No         | Yes        |

### Datasheets

[SM5166AV.PDF](https://github.com/user-attachments/files/18436133/SM5166AV.PDF)
[74HC245KAJ.pdf](https://github.com/user-attachments/files/18436134/74HC245KAJ.pdf)
[ICN2037.pdf](https://github.com/user-attachments/files/18436135/ICN2037.pdf)
[74HC245KAJ_GoogleTransalte.pdf](https://github.com/user-attachments/files/18436138/74HC245KAJ_GoogleTransalte.pdf)
[RUC7258.pdf](https://github.com/user-attachments/files/18436139/RUC7258.pdf)
[RUC7258_GoogleTranslate.pdf](https://github.com/user-attachments/files/18436140/RUC7258_GoogleTranslate.pdf)

### Code

```
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

#include <stdio.h>
#include <stdarg.h>

#include <Fonts/FreeSans9pt7b.h>

static const char* TAG = "main.cpp";

#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

static uint8_t brightness = 25; // Default brightness

MatrixPanel_I2S_DMA *dma_display = nullptr;
VirtualMatrixPanel  *FourScanPanel = nullptr;

extern "C" void app_main()
{

    HUB75_I2S_CFG mxconfig(/* width = */ PANEL_RES_X * 2, /* height = */ PANEL_RES_Y / 2, /* chain = */ NUM_ROWS*NUM_COLS);

    //mxconfig.clkphase = false;
    //mxconfig.double_buff = true;
    //mxconfig.i2sspeed = 

    dma_display = new MatrixPanel_I2S_DMA(mxconfig);

    dma_display->begin();
    dma_display->setBrightness8(brightness);

    // create FourScanPanellay object based on our newly created dma_display object
    FourScanPanel = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);
    FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);

    int x_pixel_count = 0;  // Start with the first pixel
    int y_row_count = 0;  // Start with the first row

    while (1)
    {
        FourScanPanel->clearScreen();

        // Draw pixels row by row, extending the kast line one pixel each cycle
        for (int y = 0; y <= y_row_count; y++) {
            
            // Draw last line pixel by pixel
            if ( y == y_row_count) {
                for (int x = 0; x < x_pixel_count; x++) 
                {
                    uint16_t color = (x % 2 == 0) ? 0xF800 : 0x07E0;  // Red for even, green for odd
                    FourScanPanel->drawPixel(x, y, color);  // Draw pixel
                }
            }
            // Draw toher lines in one go
            else {
                for (int x = 0; x < PANEL_RES_X ; x++) 
                {
                    uint16_t color = (x % 2 == 0) ? 0xF800 : 0x07E0;  // Red for even, green for odd
                    FourScanPanel->drawPixel(x, y, color);  // Draw pixel
                }
            }
        }
        FourScanPanel->flipDMABuffer();  // flip the buffer to show changes
        vTaskDelay(1 / portTICK_PERIOD_MS);  // Delay for 5 milliseconds 

        if (x_pixel_count == 64){
                x_pixel_count = 0;
                y_row_count++;
        }
        else
        {
                x_pixel_count++;
        }

        // When all rows are filled, reset the process 
        if (y_row_count == PANEL_RES_Y) {
            
            vTaskDelay(1000 / portTICK_PERIOD_MS);  // Delay for 1 
            y_row_count = 0;  // Reset to the first row

            FourScanPanel->clearScreen();
            FourScanPanel->setFont(&FreeSans9pt7b);
            FourScanPanel->setCursor(10, 20);
            FourScanPanel->print("TEST");
            FourScanPanel->flipDMABuffer();  // Update the buffer to show changes
            vTaskDelay(5000 / portTICK_PERIOD_MS);  // Delay for 5 seconds (for stabilization)
        }
    }
}
```

### Video
https://github.com/user-attachments/assets/859c2277-fdf6-42b0-99e6-8488603add05

https://github.com/user-attachments/assets/76701b84-d372-44c9-8faf-bffee8d8e504

### Code

Custom Led mapping in 

`inline VirtualCoords VirtualMatrixPanel::getCoords(int16_t virt_x, int16_t virt_y)`

```
 if ((panel_scan_rate == FOUR_SCAN_32PX_HIGH)) {
     if ((coords.y & 8) != 0)
        {
            coords.x += ((coords.x / panelResX) ) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += ((coords.x / panelResX)  + 1) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        if(coords.y >= 16)
        {
            coords.y = coords.y%8 + 8;
        }
        else
        {
            coords.y = coords.y%8;
        }
}
```
Other mapping logic with same result

```
if ((panel_scan_rate == FOUR_SCAN_32PX_HIGH)) {
        if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        coords.y = (virt_y >> 4) * 8 + (virt_y & 0b00000111);
}
```
This "custom" mapping logic for supplier 2 looks better.
The rows are shifted but i'm missing half of the pixel, the seem to be in the: 64 - 128 range.

```
if ((panel_scan_rate == FOUR_SCAN_32PX_HIGH)) {
         coords.x *= 2; // Double the x-coordinate
        if (coords.y >= 8) {
            coords.y -= 8; // Subtract 8 from y-coordinate if y is 8 or greater
        }
}
```
### Video

https://github.com/user-attachments/assets/6f3ed93e-8807-4c12-aa14-3eca8aed6334

**How can i make the hidden pixel show up on the matrix and shift the rows up?**
Thanks for helping me out.

board707, 2025-01-16:
What is the scan factor for each panel?

GysJef, 2025-01-16:
Based on the datasheet all modules are using 1/8 scan factor.
Because of the 64x32 dimensions, i think that means they are Four/Quarter-Scan panels

board707, 2025-01-16:
Could you run this code and show the video?
```
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};
// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

 if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);

  return coords;
}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/8 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

  /*
       // 62x32 1/8 Scan Panels don't have a D and E pin!

       HUB75_I2S_CFG::i2s_pins _pins = {
        R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
        A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
        LAT_PIN, OE_PIN, CLK_PIN
       };
  */
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // DO NOT CHANGE THIS
    PANEL_RES_Y / 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,    VIRTUAL_MATRIX_CHAIN_TYPE);

  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
 // FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(20);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```

GysJef, 2025-01-16:
Hi @board707,

Thanks for helping me out.
With my test pattern and your mapping logic looks the same as the "Supplier.2.mp4" video which posted before.

With your test pattern and configuration the display looks like this.

https://github.com/user-attachments/assets/d41ce9c8-f274-490d-af6f-428cdd304ece

board707, 2025-01-16:
Thank you
Then replace the VirtualCoords function to this:
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

uint8_t pxbase =32;   // pixel base
if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);

  return coords;
}
```

GysJef, 2025-01-16:
Thanks board707, the progress looks promising.
These ar the result of the new mapping.

I created two videos, one with your test pattern and one with my test pattern for reference.

https://github.com/user-attachments/assets/8b522196-7897-4468-bb6f-ca972f9dd3b8
https://github.com/user-attachments/assets/b7093a04-1e2a-4c3e-9eb3-4104ea7881c0

board707, 2025-01-16:
> the progress looks promising.

Unfortunately, this is not quite true. The pattern I chose does not suit your panel.
I will try to find another one for you, but I will not have time today, so I will write to you tomorrow.

board707, 2025-01-17:
Hi, @GysJef 
Could you the result of this conversion?
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

uint8_t pxbase =32;   // pixel base
if ((coords.y & 1) == 0)
        {
            coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00001111) /2;

  return coords;
}
```

GysJef, 2025-01-18:
Hi @board707 

This is the result of the last mapping you suggested.

https://github.com/user-attachments/assets/7bfb5b5b-1168-4aea-804f-1d25c2c14500

board707, 2025-01-18:
@GysJef 
Did you use this code for a loop?
```
void loop() {
 for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(20);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```
this snippet doesn't contains a multicolor lines.
Please do not change my code or I will be unable to help you.

GysJef, 2025-01-18:
Hi Board,

Sorry for the confusion.

Your script is using red pixels only
In my test script even LEDs are printed red and Odd LEDs are being printed green.
At the end I'm printing "TEST", because that makes it easier for me to see how the pixels are mapped.

uint16_t color = (x % 2 == 0) ? 0xF800 : 0x07E0;  // Red for even, green for odd
FourScanPanel->drawPixel(x, y, color);  // Draw pixel

This video is with your test script.

https://github.com/user-attachments/assets/59518700-783d-42f5-a10a-a516a5a186c2

board707, 2025-01-18:
@GysJef 
Thank you.
Sorry for my non-kind reaction :(

It seems that I choose a wrong way. 
If you haven't lost the desire to continue, you'll probably need to do a few more tests.

board707, 2025-01-18:
I see that results are not much informative because the test is too faster.
Could you repeat the very first test again, I make it slower in 4 times.
It is not required to record it in a full - 3-4 lines of led will be enough. 
```
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};
// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

 if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);

  return coords;
}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/8 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

  /*
       // 62x32 1/8 Scan Panels don't have a D and E pin!

       HUB75_I2S_CFG::i2s_pins _pins = {
        R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
        A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
        LAT_PIN, OE_PIN, CLK_PIN
       };
  */
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // DO NOT CHANGE THIS
    PANEL_RES_Y / 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,    VIRTUAL_MATRIX_CHAIN_TYPE);

  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
 // FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(80);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```

GysJef, 2025-01-18:
Hi Board707,

An overview of all the different mappings and their result.

First & Fourth Mapping

https://github.com/user-attachments/assets/e1d694b6-6dd7-4ba3-bf41-0bb28a0b5500

```
if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
```

Second mapping

https://github.com/user-attachments/assets/8104cb33-5a89-4eec-b580-76ff5c8c28e2

```
uint8_t pxbase =32;   // pixel base
if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
```

Third

https://github.com/user-attachments/assets/ec41ac0f-585c-46a5-8e0a-3036cbf1ff2f

```
uint8_t pxbase =32;   // pixel base
if ((coords.y & 1) == 0)
        {
            coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00001111) /2;
```

board707, 2025-01-18:
Thank you very much!
Now try the second with pixbase =1
```
uint8_t pxbase =1;   // pixel base
if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
```

GysJef, 2025-01-18:
@board707 

Thanks for supporting me!
The mapping you suggested is the correct one for my Led module.

I hope that the documentation of our process can help other people in the future.

https://github.com/user-attachments/assets/630d8dab-77ba-49c4-aeb8-3da467ce32bc

If you want you can contact me by mail (gysjef@gmail.com), i would like to thank you for your effort.

closed 2025-01-18 by GysJef