mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #733: 32x64 1/8 Scan P5 Qiangli Panel Problem

Endorfin35, 2025-01-20:
The panel I am using is 32x64 pixels. It is selecting the rows with RUL5158C. It is also using SM16208 for pixels.

When I run the SimpleTestShapes example with DP3246_SM5368 (mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;), I get the following image. Although there is a problem with the image, I think the row scanning is correct with DP3246_SM5368.

[image]

When I wrote a simple code that scans the R1 channel for testing purposes, I saw that the panel scans 32 pixels from the top row, then scans 32 pixels from the bottom row, then finishes scanning with 32 from the top and 32 from the bottom.

I tried the example with ESP32-VirtualMatrixPanel-I2S-DMA.h, but I could only fill 16 rows of the panel. I am not sure if this is the solution I need to achieve. Any guidance would be very helpful.

mrcodetastic, 2025-01-20:
Can you post your code?

Endorfin35, 2025-01-20:
This is the code I made to fix the scan.

`
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h" 

#define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32     // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Zincirlenmiş BIREYSEL PANELLERIN satır sayısı
#define NUM_COLS 1 // Her SATIRDAKI BIREYSEL PANEL sayısı
 
//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;
// placeholder for the virtual display object
VirtualMatrixPanel  *FourScanPanel = nullptr;

void setup() {

    // Module configuration
    HUB75_I2S_CFG mxconfig(
                PANEL_RES_X*2,              // DO NOT CHANGE THIS
                PANEL_RES_Y/2,              // DO NOT CHANGE THIS
                NUM_ROWS*NUM_COLS           // DO NOT CHANGE THIS
                //,_pins            // Uncomment to enable custom pins
    );

    
    mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
    
    mxconfig.driver   = HUB75_I2S_CFG::DP3246_SM5368;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  
    // OK, now we can create our matrix object
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);

    dma_display->begin();

    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(96);    // range is 0-255, 0 - 0%, 255 - 100%
 
   
    dma_display->clearScreen();
    delay(500);
    
    // create FourScanPanellay object based on our newly created dma_display object
    FourScanPanel = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y);
    
	// THE IMPORTANT BIT BELOW!
    FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);    

}

void loop() 
{

 for (int i = 0; i < PANEL_RES_Y; i++)
  {
    for (int j = 0; j <PANEL_RES_X; j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 0, 0));
      ESP_LOGI("ELR","Scaning Pixel : x=%d (%d) y=%d (%d)",j,tmp_x,i,tmp_y);
      delay(10);
    }
  }    
  
}`

I also tried changes in the FOUR_SCAN_32PX_HIGH section in the ESP32-VirtualMatrixPanel-I2S-DMA.h file. tmp_x and tmp_y are ineffective. Now I remember that this panel is 1/8 scan... 

`   uint8_t pxbase =32;   // pixel base
        if ((virt_y & 4) == 0)
                {
                    //  coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
                    coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
                }
                else
                {
                    //coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
                    coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
                }

        //coords.y = (virt_y >> 4) * 4 + (virt_y & 0b00000111);
        coords.y = (virt_y >> 4) * 8 + (virt_y & 0b00000111);

        tmp_x = coords.x;
        tmp_y = coords.y;  `      

panel view: [](https://www.youtube.com/watch?v=9QAB1VIkdOQ)

board707, 2025-01-20:
Please show a clear photo of the panel rearside

Endorfin35, 2025-01-20:
[image]

[image]

Endorfin35, 2025-01-20:
I made a change. The first 16 lines of the panel work fine. Then it overwrites.

I think a correction needs to be made here.
coords.y = (virt_y >> 8) * 8 + (virt_y & 0b00000111);

Endorfin35, 2025-01-20:
I made another change and now I am getting successful results. Is this method suitable? I am planning to chain in the next stage.

`        uint8_t pxbase =32;   // pixel base
        if ((virt_y & 8) == 0)
                {
                    coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer                    

                }
                else
                {
                    coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
                }

        coords.y = (virt_y >> 8) * 8 + (virt_y & 0b00000111);
        if(virt_y>15) coords.y += 8;`

board707, 2025-01-20:
> Is this method suitable? I am planning to chain in the next stage.
```
    coords.y = (virt_y >> 8) * 8 + (virt_y & 0b00000111);
    if(virt_y>15) coords.y += 8;
```

I am afraid that the lines above won't works with more than one row of chained panels. But in any case, you need a test

Endorfin35, 2025-01-20:
> Bunu deneyebilir misin? (yukarıdaki iki tanesinin yerine)
> 
> ```
>  coords.y = (virt_y >> 4) * 8 + (virt_y & 0b00000111);
> ```

it's worked. Thanks.

Endorfin35, 2025-01-21:
When I connect the second panel, there is a ghost shadow problem. I took physical precautions against the shadowing problem (short cable etc.). I tried different parameters. However, I could not solve the problem.

There is no problem with a single panel. When I connect the second panel without making a chain in the code, the same image as the first panel is created on the second panel. In this case, there is no problem with the first panel, but there is a problem with the second panel.

[image]

Then, when I make a chain for the second panel in the code, the problem starts to occur on the first panel as well.

[image]

Endorfin35, 2025-01-21:
Is there a way to too much lower speed the panel for testing purposes?

board707, 2025-01-21:
Sorry, I can't help you here.

 I'm not an expert in ghostings , I only set up pixel mapping :)

mrcodetastic, 2025-01-21:
clkphase setting?

Endorfin35, 2025-01-21:
I solved the problem.

I chose DP3246_SM5368 as the driver, it gave me a proper line selection scan, but I don't have these chips on my boards. So I looked into the code a little deeper.

As a result, shiftDriver(m_cfg); in begin() is unnecessary. This does not solve the problem, but it is ineffective for SM16208.

To solve the problem, the lines in ESP32-HUB75-MatrixPanel-I2S-DMA.cpp should be changed as follows. several cycles continue latch signal creates ghosting.

     
```
// DP3246 needs the latch high for 3 clock cycles, so start 2 cycles earlier
      if (m_cfg.driver == HUB75_I2S_CFG::DP3246_SM5368) 
      {
        row[ESP32_TX_FIFO_POSITION_ADJUST(fb->rowBits[row_idx]->width - 3)] |= BIT_LAT;   //3 DP3246 needs 3 clock cycle latch 
        row[ESP32_TX_FIFO_POSITION_ADJUST(fb->rowBits[row_idx]->width - 2)] |= BIT_LAT;   //2 DP3246 needs 3 clock cycle latch 
      } // DP3246_SM5368
```

```
     if (m_cfg.driver == HUB75_I2S_CFG::DP3246_SM5368) 
      {
        row[ESP32_TX_FIFO_POSITION_ADJUST(fb->rowBits[row_idx]->width - 2)] |= BIT_LAT;   //3 DP3246 needs 3 clock cycle latch 
        row[ESP32_TX_FIFO_POSITION_ADJUST(fb->rowBits[row_idx]->width - 1)] |= BIT_LAT;   //2 DP3246 needs 3 clock cycle latch 
      } // DP3246_SM5368
```

[image]

mrcodetastic, 2025-01-21:
Good work. So the library should have an option for the SM16208, using your code?

board707, 2025-01-21:
> So the library should have an option for the SM16208,

I would say not for SM16208 led driver, but for 5158 line switcher.

Endorfin35, 2025-01-21:
I think there should be an option for this. Something other than SHIFTREG=0 can be added.

```
    enum shift_driver
    {
      SHIFTREG = 0,
      FM6124,
      FM6126A,
      ICN2038S,
      MBI5124,
      SM5266P,
      DP3246_SM5368
    };
```

Endorfin35, 2025-01-21:
Since the beginning of the tests, I have not been able to get any images with the double_buff = true setting. Is it necessary to do anything additional about this?

Endorfin35, 2025-01-24:
I'm still having issues. When I chain two panels vertically, I can't get an image from one of the panels.

[image]

```
    /*mx_width*/                PANEL_RES_X*2,              // DO NOT CHANGE THIS 
    /*mx_height*/               PANEL_RES_Y/2,              // DO NOT CHANGE THIS 
```

If I write only PANEL_RES_Y instead of PANEL_RES_Y/2, I get the image from the second panel, but of course the scan is distorted.

board707, 2025-01-24:
Please show your full code, including VirtualMatrix get_coords() method

Endorfin35, 2025-01-24:
I fixed it. But I don't know what I did. 

```
  HUB75_I2S_CFG mxconfig(
    /*mx_width*/                PANEL_RES_X*2,              // DO NOT CHANGE THIS  *2
    /*mx_height*/               PANEL_RES_Y,                // DO NOT CHANGE THIS   /2
    /*chain_length*/            NUM_ROWS*NUM_COLS,          // DO NOT CHANGE THIS
```

I removed the division from PANEL_RES_Y/2.

```
        uint8_t pxbase =32;   
         if ((coords.y & 8) == 0)//if ((virt_y & 8) == 0)
        {
             coords.x += ((coords.x / pxbase) + 1) * pxbase; 
        }
        else
        {
            coords.x += (coords.x / pxbase) * pxbase;
        }
        // http://cpp.sh/4ak5u
        // Real number of DMA y rows is half reality
        // coords.y = (y / 16)*8 + (y & 0b00000111);
        //coords.y = (virt_y >> 4) * 8 + (virt_y & 0b00000111); 
```

I disabled the coords.y line in ESP32-VirtualMatrixPanel-I2S-DMA.h.

it is working properly now.

[image]

board707, 2025-01-24:
Ok... so my help don't now

dobro486, 2025-02-06:
Hi, could you provide the full code? I can't get this panel to work properly.

dobro486, 2025-02-06:
> Since the beginning of the tests, I have not been able to get any images with the double_buff = true setting. Is it necessary to do anything additional about this?

Oh, it's very simple. First clear the buffer, then fill it. The update is instantaneous. No flickering.
// Buffer cleanup
  dma_display->clearScreen();
  
  dma_display->setCursor(0, 0);    // start at top left, with 8 pixel of spacing
  dma_display->setTextColor(dma_display->color444(255,0,0));
  dma_display->println("TEST");

// Updating the screen
  dma_display->flipDMABuffer();
  delay(30);

board707, 2025-02-06:
> I can't get this panel to work properly.

Are you sure that your panel is the same as the ones of @Endorfin35 ?

dobro486, 2025-02-06:
> Are you sure that your panel is the same as the ones of [@Endorfin35](https://github.com/Endorfin35) ?

I am absolutely sure that this is the same panel. Surprisingly. I reinstalled the library and it's almost working!

[image]

https://github.com/user-attachments/assets/e99acf5c-fb90-43db-8c43-8158a1c8cf85

I used this code

uint8_t pxbase =32;   // pixel base
        if ((virt_y & 8) == 0)
                {
                    coords.x += ((coords.x / pxbase) + 2) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer                    

                }
                else
                {
                    coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
                }

         coords.y = (virt_y >> 4) * 8 + (virt_y & 0b00000111);

board707, 2025-02-06:
> that this is the same panel.

I don't think so:

@Endorfin35 panel is Q5H19B8V1.1

but yours - Q5H25V8V1

The difference in index may indicate different pixel mapping on your panel.

dobro486, 2025-02-06:
> The difference in index may indicate different pixel mapping on your panel.

Oh, I'm obviously not as good at this as you are. I have little experience with LED panels. But it almost worked. There is a slight offset, but I don't think it's fatal, I'm not sure if I can fix it, but I'll try.

Endorfin35, 2025-02-06:
@dobro486, What kind of controller are you using? I did my tests here with the ESP32-S3 DevKit.

After adding a 74HC245 to the I/O lines, I encountered a very similar issue to yours.

I commented out these lines in ESP32-HUB75-MatrixPanel-I2S-DMA.cpp and changed the value -1 to -2 (or vice versa, I can't access my code right now) in the line just below.

```
     if (m_cfg.driver == HUB75_I2S_CFG::DP3246_SM5368) 
      {
        row[ESP32_TX_FIFO_POSITION_ADJUST(fb->rowBits[row_idx]->width - 2)] |= BIT_LAT;   //3 DP3246 needs 3 clock cycle latch 
        row[ESP32_TX_FIFO_POSITION_ADJUST(fb->rowBits[row_idx]->width - 1)] |= BIT_LAT;   //2 DP3246 needs 3 clock cycle latch 
      } // DP3246_SM5368
```

Endorfin35, 2025-02-21:
I was happy with the Qiangli panel, so I reordered it. But the panels I received turned out to have different ICs. sm16169SH is used instead of sm16208sj in new panels. I couldn't even get a low-quality image. (I verified that the panels work with Huidu).

Any idea what could be causing this problem?

[image]

board707, 2025-02-21:
> Any idea what could be causing this problem?

SM16169 is driver of the different type , it is incompatible with the current code.

Endorfin35, 2025-02-23:
The structure of the SM16169SH looks quite similar to the DP3246. I believe a special initialization function is required. If I manage to solve it, I will add it here.

Endorfin35, 2025-02-24:
> > The structure of the SM16169SH looks quite similar to the DP3246.
> 
> Where does this conclusion come from, since your panel doesn’t show anything?

I am writing a simple test code without using a library for the working panel. Both panels use RUL5154C for row selection. The only difference is the LED drivers. The SM16169 has a register for current adjustment. I suspect that this setting is 0 at startup.

Endorfin35, 2025-02-25:
@board707 , 

I am reviewing the working structure of the library. I have also ordered a logic analyzer. I have gotten some idea, but I am still continuing my research. I read some of your comments in this link: [https://github.com/hzeller/rpi-rgb-led-matrix/issues/466](https://github.com/hzeller/rpi-rgb-led-matrix/issues/466). Do you think this library can be edited for the SM16169, or is it impossible?

StephenArwine, 2025-02-25:
It seems like this might be a new trend on panels to use SM16169. I just found this post after recieving a shipment of panels out of china all with the SM16169 when then had previously been SM16208SJ

Endorfin35, 2025-02-26:
@StephenArwine , There is some information about SM16169 at the link below. Let me know if you can use SM16169 too! :)

[https://github.com/hzeller/rpi-rgb-led-matrix/issues/1504#issuecomment-1406583087](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1504#issuecomment-1406583087)

Endorfin35, 2025-02-27:
Here are a few updates:

Initializing SM16169 panels is not mandatory. When I connect the panel to the Huidu control board at any runtime, it works fine. Also, no special situation is observed during operation in the logic analyzer. Writing to the potential registers of the SM16169 may provide some advantages, but it seems unimportant at the moment.

Line Scan in the Library With the DP3246_SM5368 configuration, the panel's line scan works fine.when  I remove the data cable from the panel connected to Huidu and then connect it to the ESP panel displays the image. Since I did not turn off the power during this process, the data sent by Huidu is stored in the panel's registers. The ESP continues the line scan process and the panel displays the image.

The problem is that I have not yet been able to write data to the sm16169 outputs

board707, 2025-02-27:
> Initializing SM16169 panels is not mandatory. When I connect the panel to the Huidu control board at any runtime, it works fine.

Huidu sends initialization values at every time updating the image as the part of the data updating process. 

> Line Scan in the Library With the DP3246_SM5368 configuration, the panel's line scan works fine

The names of the Library scan options a non-informative. This option has nothing to do with DP3246 driver. So it would be more helpful if you provide the info about the exact chips on your panel.

Endorfin35, 2025-02-27:
The panel I tested is sm16169sh + rul5158c.

I realized that what I wrote in the previous message was not completely correct. The panel gives an image but it is not good. 

1. I send full frame white with huidu.
[image]

2. When I transfer the cable to esp, the image continues.

[image]

3. I repeat the test with a text.

[image]

4. When I transfer the cable to esp, the image continues. However, the scanning is not correct. Working with full frame white was a mistake for me.

[image]

As a result, the panel's line scanning works. I think there is a problem only in the writing part to sm16169.

Endorfin35, 2025-02-27:
> It is because Huidu sends initialization values at every time updating the image as the part of the data updating process.

You're right. It repeats in a pattern.

[image]

board707, 2025-02-27:
Of course I'm right, I've been doing this for six years now :)

Endorfin35, 2025-02-27:
[image]

I am sending 16 bits and doing lat but I don't understand why the writing is not successful...

board707, 2025-02-27:
> I am sending 16 bits and doing lat

Did you try to send a data or config? Although, in both cases, 16 bits is not enough to transmit

When you transmit the config - first you must send the data to all the panel drivers, and only then give the latch signal.

If you transmit data - then you must transmit 16 bits of grayscale data for each pixel of the panel.

Endorfin35, 2025-02-28:
I am not sending config. I am trying to run the panel with huidu and then write any data with esp. I am trying to understand the logic by writing simple codes myself. I am sending gclk independently with a task. After selecting a row once, I stop the row scan. During this time, I get an image on the row due to the data sent by huidu. Then I am trying to change the row data. Since the input register of a single sm16169 is 16 bits, I am trying to get an output by sending lat every 16 bits.

While sending 16 bit data, I am trying lat with the last clk signal. Here I tried shifting with long and short lat signals for a few clks. There were some attempts where I got output by chance. However, when I tried again in the same way, I could not get output. So I still do not understand the logic.

I do not fully understand the 8K memory usage of sm16169. I think it is 8kbit. Since the input register is 16 bits, I calculate 8192/16=512. So if I write the same 16 bit data continuously from the input, I will only get an output after 512 lat?  

[image]

> If you transmit data - then you must transmit 16 bits of grayscale data for each pixel of the panel.

Is each 16-bit data given from the input group by group together for the outputs?

Endorfin35, 2025-02-28:
my panel is 32x64 pixels. there are 8 sm16169 in a data line (for example R1). 16x8=128, so one data line is responsible for 2 lines. this calculation is compatible with the 1/8 scan feature of the panel. then i need to send data for 128 pixels. so i need to lat one bit for every 128 clk. since 16 bits are set per ic in each lat operation, 8192/16=512 times, then i need to receive the image.

[image]

i set the lat signal exactly synchronized with clk as you said. i still can't get a response from the panel i started with huidu.

i will keep trying :)

Endorfin35, 2025-03-05:
I managed to write the panel. I can get a stable image, but I still can't init the panel. During the init sequence I found the same values ​​for 5 registers. 

@etarti Murat bey isterseniz whatsapp üzerinden paslaşabiliriz. (beşyüzkırikibeşyüzonüçyetmişaltıçiftsıfır)

board707, 2025-03-05:
> I still can't init the panel

What config values do you used - the ones provided by  @wata-net or that you read from the LA sequence?

Endorfin35, 2025-03-05:
I'm constantly trying different things :) I experimented with the information shared by  @wata-net and, as you suggested, tried adjusting the number of rows. However, the startup sequence I captured with the logic analyzer from HUIDU does not match  @wata-net 's description. The initial 3clk and 14clk parts are the same, but after that, there are 5 register write operations with 0x00AA, 0x01AA, 0x1104, 0x0055, and 0x0155. These write operations happen with a 5clk latency. Additionally, the number of gclk doesn't match  @wata-net 's description either. Currently, 8 gclks are applied per row. I'm trying to mimic HUIDU, and I have been successful in getting a stable image, but there is an issue with the startup.

There is a sh at the end of the ic model. SM16169Sh I'm not sure if there is any other difference. I asked for a datasheet from both the IC manufacturer and the panel manufacturer but they didn't respond.

Endorfin35, 2025-03-05:
@e-tarti, after reading your comment, I also realized that the 3rd register data is variable. After starting the panel with Huidu, I can write to the panel as I wish. At this stage, I haven't encountered any issues.

My problem is related to starting the panel. I am using my own ESP-based control board. So far, I haven't been able to start the panel myself. Right now, I am trying to use a counter for the 3rd register, inspired by your findings. However, I haven't achieved any results with the values you identified.

etarti, 2025-03-05:
@Endorfin35 i will contact with you tomorrow. I was continuously sending a counter. I realized that on every cycle I used OR operation with 0x0100 to this value and the next cycle AND operation with 0xFE7F (looking to code). I tried so many times and so many thinks, finally I lost my control and let it like this :)  I also try to ifnd a pattern on this magic counter looking sequentially. But I have no good logic analyzer. Just 100Mhz osciloscope with 2channel :) Proper datasheet will solve everything. Otherwise we are about to solve this routin by reverse enginering. I still do not understand REG1- REG2 registers and where are they? 0x00AA followed 0x01AA may be a REG1 initilization?

Endorfin35, 2025-03-05:
I have detected the changing Reg3 values in each frame until I got tired. I found 26 different values. I tried starting the panel with these values, and the panel worked. I'm not sure if any of these values specifically make it work or if something else is happening. I will look into this again.

voleminhtripro, 2025-03-06:
Has anyone been able to control two P5 Outdoor LED panels arranged vertically?

YassineElHosni, 2025-04-30:
@Endorfin35 any update on this please 🙏🏻 ?

Endorfin35, 2025-05-02:
@YassineElHosni,
 I only just got a chance to reply. It seems that the SM16169 and SM16169SH series are different from each other. I wasn't successful using the documented methods. I obtained the datasheet from the manufacturer under a non-disclosure agreement, so unfortunately I can't share the details. You may also request the documentation directly from the manufacturer. Ultimately, it's not possible to use it with a library—you'll need to write your own code.

YassineElHosni, 2025-05-03:
Thank you for sharing @Endorfin35, i have decided to just buy different led matrixes.

Tobilang3241, 2025-08-25:
Hi @Endorfin35,
So did you manage to write to the LED Panel with the SM16169S with just using your esp after you got the datasheet?
Thanks in Advance :)

Tobilang3241, 2025-08-25:
@Endorfin35 Is there any way that you could share this code or does it fall under the nda?

closed 2025-08-25 by Endorfin35

Tobilang3241, 2025-08-26:
And to find out how to drive the SM16169S you needed the datasheet?

Tobilang3241, 2025-09-02:
Hi @Endorfin35,

is there any way that you could provide a example of how you drive the SM16169SH, or any kind of script that shows how you do the initialization?