mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #736: Looking for help: How to adapt code to asimetric display

PeperonyPizza, 2025-01-28:
### Overview

I would like to ask for help
I have custom triangular panels for the robot, and I would like to use this repository to stream information from the PC to these panels.
I have connected ESP to the panel, but when I try the PIO example I can only get the Red, Yellow and White to show up 

https://github.com/user-attachments/assets/f674bc86-5fd6-4be5-a8ba-1c416aac7539

after this, the panel stays white with some static streaks of different colors.

[image]
I also sometimes get just random colors. 

These are the components on the LED panel
SM16389N under it, smaller text AJBJSKAD1
SM5369NS  under it, smaller text  AQCIQDAE15

[2303031730_Shenzhen-Sunmoon-Micro-SM16306S_C2830324.zh-CN.en.pdf](https://github.com/user-attachments/files/18574035/2303031730_Shenzhen-Sunmoon-Micro-SM16306S_C2830324.zh-CN.en.pdf)
 Note: found these datasheets using Perplexity, so I´m not sure if they are correct

74HC245KAJ  under it, smaller text NHT323000

[2311161805_Shenzhen-Fuman-Elec-74HC245KAJ_C19192982.zh-CN.en.pdf](https://github.com/user-attachments/files/18574047/2311161805_Shenzhen-Fuman-Elec-74HC245KAJ_C19192982.zh-CN.en.pdf)

I´m not sure about the scan rate
The size of one panel is 37*156px

#

### Help needed

I want to try to adapt this repo so I can use it with these panels. Can somebody help me or point me where to look at?

board707, 2025-01-28:
> These are the components on the LED panel
> SM16389N 
> SM5369NS 

Are you sure in these numbers? If so, why do you read a datasheet for SM16306 driver?

PeperonyPizza, 2025-01-28:
@board707 Yes, I´m sure, And  because it was the closest I could find.

PeperonyPizza, 2025-01-28:
rear side of the panel
[image]

closeup of the chips
[image]

board707, 2025-01-28:
Very interesting panel.
What setting did you use when you run an example with single color?

PeperonyPizza, 2025-01-28:
I only changed `PANELS_NUMBER` To 1 and `NUM_ROWS, NUM_COLS` to 1
i also tried changing ` mxconfig.driver = HUB75_I2S_CFG::FM6124;` to all of the available ones but without success, on SM5266P it wasn´t working at all and DP3246_SM5368 was just white with some pattern

board707, 2025-01-28:
What's panel dimensions did you use?

PeperonyPizza, 2025-01-28:
I left it on 64x64, I also tried changing it before, But I would always get similar results, eighter I would get a colorfully panel like on top or some random streaks of different intensity, or if I put a small number, I think less than 10,  it would not work at all

PeperonyPizza, 2025-01-28:
I´m also looking at thru current software that I´m using for control of this LED panels via Controller and found this
Driver IC: SM16389SF and Scan IC: 5369 DECODING

Software is Autoled, controller is MC75E

board707, 2025-01-28:
The label near the HUB75 connector says that it has a  "37x156 s52" format.
Real panel width is 155 pixels on base and 38 pixels height in the highest point.
It means that the panel should use a very weird pixel pattern.

board707, 2025-01-28:
> Driver IC: SM16389SF

I am afraid that this driver is unsupported by the library.

PeperonyPizza, 2025-01-28:
> I am afraid that this driver is unsupported by the library.

What would I need if I wanted to make a driver for it? or is there a way to adapt the current ones?

board707, 2025-01-28:
> What would I need if I wanted to make a driver for it?

First of all - you need the datasheet of that specific driver. I didn't found it in the net.

PeperonyPizza, 2025-01-28:
The only one I can find is on this page https://www.linkage.cn/xianshiping/620.html
and this one should be similar https://www.linkage.cn/xianshiping/620.html

board707, 2025-01-28:
Leaving aside that the 16389 and 16380 drivers might be different - the datasheet of the last one, SM18380, is not complete.
Protocols of new Chinese drivers, the so-called PWM type, are a commercial secret and datasheets for them are not publicly available or are distributed in a truncated form without descriptions of the protocol and configuration.

PeperonyPizza, 2025-01-28:
> Protocols of new Chinese drivers, the so-called PWM type, are a commercial secret and datasheets for them are not publicly available 

So no luck for me I guess, but I will try to ask the supplier for the datasheet

board707, 2025-01-28:
Do you really need panels of this shape?
If not, it would be much easier to buy another panel than to try to run this one

Lukaswnd, 2025-02-04:
just by looking at the image of the backsite I can tell this works quite different. Normal panels have shift registers to fill the lines and addressing chips to convert the ABCD(E) line selektor to a specific line. So there is like no change this panel will work with this library

PeperonyPizza, 2025-02-12:
[SM16389SF_ZIGOQZZV1.0_EN_2025.2.7.pdf](https://github.com/user-attachments/files/18764863/SM16389SF_ZIGOQZZV1.0_EN_2025.2.7.pdf)
[SM5369PS_ZWGGSZAV1.0_EN_.pdf](https://github.com/user-attachments/files/18764862/SM5369PS_ZWGGSZAV1.0_EN_.pdf)

I got datasheets from manufacturer