mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #737: P5 64x32 8s Not working properly

selam2, 2025-01-28:
My dashboard works but the scanning is not smooth. I tried but I couldn't do it. Can you show me how to scan? Türkçede biliyorum 

[image]

board707, 2025-01-29:
Do you try a QuarterScan panel example?

selam2, 2025-01-29:
ad "20250129_212912.mp4" -->

selam2, 2025-01-29:
Not downloading videos

selam2, 2025-01-29:
[image]
 There's only trouble with the horizontal one

board707, 2025-01-29:
Your VirtualMatrixPanel class ( in the code above) is for 32x16 1/4 panel, while your panel is 64x32 s8.

Please replace a getCoords() method in your VirtualMatrixPanel class  with code below and show the resulting video:
```
// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
    coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }
    if ((coords.y & 8) == 0)
        {
      coords.x += ((coords.x / panelResX) + 1) * panelResX; 
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX;
        }

        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
    return coords;
}

```

selam2, 2025-01-29:
https://github.com/user-attachments/assets/fcb1b4ee-923a-4e8c-8201-43b3445539dc

board707, 2025-01-29:
Thank you
Now change the method to this:
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
    coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }
  uint8_t pxbase =32;
     if ((coords.y & 8) == 0)
       {
            coords.x += ((coords.x / pxbase) + 1) * pxbase; 
        }
        else
        {
            coords.x += (coords.x / pxbase) * pxbase; 
        }
    coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
    return coords;
}
```

selam2, 2025-01-29:
İt is okey. Thank you very much .Can you tell me what you set this for? I have a lot of panels like this.

board707, 2025-01-29:
> Can you tell me what you set this for?

See this discussion:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/discussions/622

selam2, 2025-01-29:
Thank you. I didn't quite understand, but I got it better