mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #738: 128x32 panels - screen tearing

PastorL69, 2025-01-29:
Hi all. I am a ZeDMD user, project can be found here: https://github.com/PPUC/ZeDMD
It uses this library very well, although we are looking to optimize it to our use case as best as possible. 

Onto the issue:
I am experiencing occasional screen tearing. I use an ESP32-S3-N16R8 and double buffering is enabled, so you would think that the frames being transmitted should all arrive and display in one piece.
A few example pictures here:

[image]
[image]
(ignore the invisible lines, they are visible in person)

I already read about a previous issue which seems to be similar to mine: #592. There the person is talking about increasing the i2s clock speed or setting a min refresh rate.

Is this something that could even be fixed in general? It's almost not noticeable, but it definitely affects the "smoothness" of viewing these quick frames. Maybe someone else already looked into this and is aware of something that could work. This happens with about 10-20% of the frames, most of the time there is no tearing at all. 

Thanks in advance.

PastorL69, 2025-01-29:
https://github.com/user-attachments/assets/8b901711-df77-41b0-b38a-1b0f8737b683

Here another video with a way higher min refresh rate that I set. Also experimented with the i2sspeed. Could this just be a panel lottery moment?

Lukaswnd, 2025-02-04:
Are you just facing the problem of screen tearing, this should be eleminated by using the double buffer. But you have to wait for the buffer change. like mentioned in the issue you marked, you can skip this wait by kommenting out a few lines or by using multi threading (Task A initials the switch of the screen buffers, Taks B does not wait for it to be finished and writes to what ever parts are active at the moment -> tearing).

I think you are complaining about a different issue aswell. Especially when updating lare chunks of the matrix buffer it takes some time. This is how the library works in the very core. Updating one pixel is not just 1 or 3 bytes it is up to hundrets of bytes and the right bit of each of these bytes. By reducing the color depth this is speed up a little. For example I run ~24k pixels at 4 bit color depth with an s3 and when updaing the whole screen continuously I can't reach 20 fps (just for the sceen updating, normally you do other stuff in the background aswell, like GFX calculations or io). So I think you are limited there as well at around 100fps (depends on your color depth).
I did a quick scan of the ZeDMD code and I think you are using 8 bit color depth (default value), try to reduce it (maybe 5).
Other than that the power of esp32s3 might be limiting you can use the void printRunningTasks(Print &printer); and optimise task and core usage

PastorL69, 2025-02-04:
I would have to ask @mkalkbrenner that since he's the maintainer of ZeDMD. I'm pretty sure he told me that he uses his own double buffer. 

At least from my perspective it either looks like my panels can't take it or it is indeed the color depth. Perhaps the time between writing the changed pixel data to the buffers is too short. But then again, when you look at the LAST video I sent, you can clearly see that a lot of times there is no tearing at all. It seems to happen a few times back to back.

Appreciate your comment a lot, I hope we can smash this "bug".

Lukaswnd, 2025-02-04:
If ZeDMD uses an double buffer internally it would not fix the matrix tearing. The issue you fix with a double buffer is not writing to the currently active output. So the double buffer has to be on the matrix site.

PastorL69, 2025-02-04:
I just tested with the color bit depth to 5, behaviour remains the same.
That I guess leaves only the double buffering or my panels are not great.

mkalkbrenner, 2025-02-04:
I think, some clarification is needed.
ZeDMD uses it's own "double buffering" on the S3. But this is different from double buffering in the lib.
In ZeDMD it is a buffering of raw RGB565 data, so 2 bytes per pixel.
Only pixels that differ between both buffers get updated via drawPixel().
The nature of a pinball DMD is that often only small parts of the frame change. So in my experiments, the double buffing of this lib slowed down the rendering in general.
But for sure, some animations that break the "only a few pixels change" rule will look less smooth.
But I didn't look at the lib's internal double buffering for a long time.
Does it provide a feature like copying the other buffer?

PastorL69, 2025-02-11:
So the only fix for this is if we were to use the double buffering in the lib itself? For sure the screen is updating too early with the new data. If only there was a way to slightly delay the data that is being inserted into the shift registers from being displayed. Will increasing the latch blanking above the max value of 4 have any effect? By visual observation it seems like it really could be fixed with just a tiny tiny delay

mrcodetastic, 2025-02-17:
I haven't tested this personally, but I think the only solution is to have three (yes 3) buffers.

Over the years I've tried to solve various incarnations of flicker with 'flipDMABuffer', but I've come to the conclusion that it's not really possible. 

- The DMA hardware loops continuously and asynchronously and does its own thing once established on begin()
- Whilst flipDMABuffer waits for an interrupt to be generated that 'one loop' of the buffer ring has completed, there's no guarantee on the timing of this. 

My suggestion would be to:

1. Always write to your own off-screen pixel buffer (not DMA).
2. Then when you've done painting, write this to the non-visible DMA buffer.
3. Then when you code is certain 2) has completed. Call flipDMABuffer.

mrcodetastic, 2025-02-17:
I haven't tested this personally, but I think the only solution is to have three (yes 3) buffers.

Over the years I've tried to solve various incarnations of flicker with 'flipDMABuffer', but I've come to the conclusion that it's not really possible. 

- The DMA hardware loops continuously and asynchronously and does its own thing once established on begin(). It doesn't care nor know what the sketch is doing.
- Whilst flipDMABuffer does wait for an interrupt to be generated by the DMA engine that 'a loop' of the buffer ring has completed before changing the pointer to the DMA memory block that future calls of drawPixel() will modify.... there's no guarantee on the timing of this for whatever reason, and as such, potentially stuff is still being written to the back-buffer whilst the 'flip' occurs.
- This is why potentially your 'tiny delay' idea works, but there's no science on what the delay should be.

My suggestion would be to:

1. Always write to your own off-screen pixel buffer (not DMA).
2. Then when you've done painting, write this to the non-visible DMA buffer.
3. Then when you code is certain 2) has completed. Call flipDMABuffer.