mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #741: I have to set a 63 pixels width for a 64 pixel wide panel.

roelvan, 2025-02-02:
Hi,

I have to set my panel's width to 63 to make it work with this library. Otherwise everything I draw at (0,0) is drawn at (-1, 0).
Is this the normal behaviour or am I doing something wrong?

My panel specs:
- Driving Chips: Chipone2037/DP5125/FM6124
- 16scan
- Resolution: 32(H)×64(W)

I am using these values for a adafruit matrix s3 portal board:

```
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_RES_X 63  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1   // Total number of panels chained one to another
#define R1_PIN 42
#define G1_PIN 41
#define B1_PIN 40
#define R2_PIN 38
#define G2_PIN 39
#define B2_PIN 37
#define A_PIN 45
#define B_PIN 36
#define C_PIN 48
#define D_PIN 35
#define E_PIN 21
#define LAT_PIN 47
#define OE_PIN 14
#define CLK_PIN 2
```

davemaster, 2025-02-02:
change this

#define PANEL_RES_X 63  // Number of pixels wide of each INDIVIDUAL panel module.

for this

#define PANEL_RES_X 64  // Number of pixels wide of each INDIVIDUAL panel module.

Before, RE-CHECK twice or more times the wires... continuity, one by one

Last, the power supply at least must have 6-8A

roelvan, 2025-02-03:
Thanks for your help David.
When I set it to `#define PANEL_RES_X 64`, there is a cutoff (which I do not have when I set it to 63). I also have differently colored pixels:

[image]

Cables should be ok since I am using a ready made hub75 adafruit matrix portal.

davemaster, 2025-02-03:
Questions:
1. Power supply? At least you MUST use one from 6-8A
2. Where is your command? Share it, and its result on picture or video
3. What RPi are you using? 2, 3, 4????

PD. those wrong colors can be, defective leds -hardware- or weak power supply...

roelvan, 2025-02-03:
1. I am using the same power supply as with the Protomatter library from adafruit, which draws correctly.
2. Relevant code (rectangles or other things are also cut off screen):
```
  dma_display->setCursor(0, 0);
  dma_display->setTextColor(leftScore.color);
  dma_display->println(leftScore.score);
```
3. adafruit matrix s3 portal, which is an ESP32-S3
4. Hardware is not defective, see point 1.

thanks for your help

Lukaswnd, 2025-02-04:
This library is compatible with ESP32-S3.
You should set your width to 64 and draw at (0,0) to see someting in top left corner. 
The Adafruit Library designed for the Adafruit Matrix Portal works very different, so just because you have enough power for that, does not mean you have enought power now. Try to reduce your brightness. Did you enable the FM6124 shift driver in your code (checkout examples)? Also try without it. If it still does not work try to play around with lat_blanking and clockphase.
You can also try to reduce your clock speed (i2s speed) on poor produces boards this might case Problems, but not like the ones you describe

roelvan, 2025-02-05:
Thanks for the info, good to know it should be compatible with esp32-s3. Will try again and make sure the power requirements are met and will try out your tips.

closed 2025-02-05 by roelvan