mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #744: Wf4

selam2, 2025-02-07:
I have Wf4 ESP32 S3 And this usb-a usb-a I connect it to my computer using cable And he sees the computer and then he goes I couldn't solve it in a way, please help me.

j-g00da, 2025-02-07:
Please explain exactly what's happening and steps to reproduce the problem. 
Can you provide links to devices you are using?

selam2, 2025-02-07:
[image]
 This card I use

j-g00da, 2025-02-07:
What are you trying to achieve? Flash a custom firmware?

selam2, 2025-02-07:
This card has an esp32 s3 processor on top of it n
Normally, when you plug the USB cable into the software installation, you should be able to install software, but it's like a lock And when I plug it in, it's seeing it and going back

j-g00da, 2025-02-08:
If you are using original firmware, then you are in the wrong place.

selam2, 2025-02-08:
The problem is I assign software to the device