mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #759: 80x40 P4 with A B C pin Config Problem

hfarcan, 2025-02-19:
I have an 80x40 P4 display. The LED driver, as far as I can read from the back, is ICN2038. My matrix pins are A, B, and C. I haven't been able to run the examples for a few days. The newly updated "FOUR_SCAN_40PX_HIGH" scan type in the VirtualMatrixPanel example today gave me hope. I think I'm making mistakes in the configuration.
Note: Among the driver options, the one that came closest to a proper display, interestingly, was the DP3246_SM5368 driver.
I only made the following changes on the config side in the code:

// Select example to compile!
 //#define EXAMPLE_NUMBER 1
 //#define EXAMPLE_NUMBER 2
 #define EXAMPLE_NUMBER 3

#define PANEL_RES_X      80     // Number of pixels wide of each INDIVIDUAL panel module. 
 #define PANEL_RES_Y     40     // Number of pixels tall of each INDIVIDUAL panel module.
 
 #define VDISP_NUM_ROWS      1 // Number of rows of individual LED panels 
 #define VDISP_NUM_COLS       1 // Number of individual LED panels per row

#define LAT  10  
#define CLK  11  
#define OE   12  

#define A    8  
#define B    7   
#define C    9   

#define R1   2 
#define G1   1  
#define B1   3  
#define R2   5  
#define G2   4 
#define B2   6  

 #define PANEL_CHAIN_TYPE CHAIN_NONE

mrcodetastic, 2025-02-21:
Not supported by this library I'm afraid. It's not to do with pixel mapping, but how the data signal is transmitted. 

If you have a lot of these panels, then I can try and acquire one to test with. Got a link to where you purchased it from?

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/121

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/448

hfarcan, 2025-02-21:
I bought one from a local supplier. If you want, I can find the link for you, but I think it would make more sense to switch to a more standard panel for efficiency and time-saving reasons. Both the scan rate and the LED driver seem like they would require a lot of effort. Thanks for your response.

board707, 2025-02-21:
@hfarcan 
What is the panel scan number and what is the index of other chips on the panel?
Has the rear of the panel a label with driver and scan identification?

hfarcan, 2025-02-21:
The tops of the integrated circuits are coated with varnish, so I can't read them. Since there are A, B, and C pins, I think it has a **"1/8"** scan.
I was able to barely read **"ICN2038"** from one of the ICs. However, when testing the drivers, I only saw a near-correct image with this configuration: "mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;"
Note: The serial number engraved on the panel's casing is: P4D8040S-A1-Q2.

[image]

board707, 2025-02-21:
> Since there are A, B, and C pins, I think it has a **"1/8"** scan.

1/8 is impossible, because it should be integer delimiter of half of height. So it can be 1/20, 1/10 or 1/5
Three pins here because the panel most likely has a shift register line driver.
Please show a clear photo near "Glux" label

mrcodetastic, 2025-02-22:
I forgot that in the code somebody contributed this for a ABC scan panel... I wonder if this works?

m_cfg.driver == HUB75_I2S_CFG::DP3246_SM5368)

mrcodetastic, 2025-02-22:
> I only saw a near-correct image with this configuration: "mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;"

That makes sense as this is a user contribution for ABC panels. 

Can you please try the panel without using the virtual matrix class?

mrcodetastic, 2025-02-22:
'''

void loop() {
  for (int i = 0; i < dma_display->height(); i++)
  {
    for (int j = 0; j < dma_display->width(); j++)
    {
      dma_display->drawPixel(j, i, dma_display->color565(255, 0, 0));
      delay(30);
    }
  }
  delay(2000);
  dma_display->clearScreen();
} // end loop

'''

hfarcan, 2025-02-22:
> At least please show a clear photo an area near "Glux" label
Hi. I read ICN2037BP in many places, but to avoid clutter in the photo, I took a picture of the one right above the Glux label. I know the photo is bad, but this was the best I could do because of the varnish.

[image]

board707, 2025-02-22:
Thank you
According to the number of the chips, your panel seems to has a 1/10 scan.

To be sure about the multiplexing I need index of the chips, highlighted in the picture:

[image]

hfarcan, 2025-02-22:
Hi @mrcodetastic . Here videos with different configs:

1. Video with default Config

| Feature | Val|
|--------|--------|
| PANEL_RES_X| 80|
| PANEL_RES_Y| 40|
| mxconfig.drive| Default |

https://github.com/user-attachments/assets/15566599-3ab2-4f9c-abce-f6bde6e616a4

2. Video with driver ICN2038S

| Feature | Val|
|--------|--------|
| PANEL_RES_X| 80|
| PANEL_RES_Y| 40|
| mxconfig.drive| ICN2038S|

https://github.com/user-attachments/assets/438c9a12-926b-4a98-b1eb-80eaee028718

3. Video with driver DP3246_SM5368

| Feature | Val|
|--------|--------|
| PANEL_RES_X| 80|
| PANEL_RES_Y| 40|
| mxconfig.drive| DP3246_SM5368|

https://github.com/user-attachments/assets/12c03670-66f1-4849-9020-d1c8f1600957

hfarcan, 2025-02-22:
> Thank you According to the number of the chips, your panel seems to has a 1/10 scan.
> 
> To be sure about the multiplexing I need index of the chips, highlighted in the picture:
> 

Thank you for interest @board707 
The text I could read in the highlighted part of the picture is ICND2018 and D57085ZA. 

[image]

board707, 2025-02-22:
Thank you
So your panel has a shift register line driver.

A full panel specification is as a follows:
P4 80x40 1/10 
Drivers: ICN2038 ICN2018

The most close library config is, indeed, the DP3246_SM5368 option. But it is not fully compatible with the ICN2018 chip and needs some work.
Unfortunately, testing your panel remotely is complicated by the fact that it is also a "Quarter_scan" - two effects overlapping each other. 
I am afraid that it will be difficult to configure it remotely...although we can try.

hfarcan, 2025-02-22:
> The most close library config is, indeed, the DP3246_SM5368 option. But it is not fully compatible with the ICN2018 chip and needs some work. Unfortunately, testing your panel remotely is [...]

I think it would be better to choose a panel that is more in line with the standards. I'm still at the beginning of the project and have the flexibility to do so. Thank you for all your interest and effort.

mrcodetastic, 2025-02-22:
Interesting, because it seems it is a 'four scan' panel, because 40/10 = 4

```
So I know you have tried this, but I'm surprised this doesn't work

HUB75_I2S_CFG mxconfig(
160, // DO NOT CHANGE THIS
20, // DO NOT CHANGE THIS
1,
_pins// A Single panel
);

mxconfig.clkphase = true; // can you try this with false as well?
mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;
```

Just for my own intellectual curiosity, would you mind running the following code?

```
void loop() {
  int height = dma_display->height();
  int quarter = height / 4; // Calculate 1/4th of the display height

  for (int i = 0; i < height; i++) {
    uint16_t color;

    // Change color based on which quarter 'i' is in
    if (i < quarter) {
      color = dma_display->color565(255, 0, 0); // Red
    } else if (i < 2 * quarter) {
      color = dma_display->color565(0, 255, 0); // Green
    } else if (i < 3 * quarter) {
      color = dma_display->color565(0, 0, 255); // Blue
    } else {
      color = dma_display->color565(255, 255, 0); // Yellow
    }

    for (int j = 0; j < dma_display->width(); j++) {
      dma_display->drawPixel(j, i, color);
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();
} // end loop

```

hfarcan, 2025-02-22:
> ```
> 
> Just for my own intellectual curiosity, would you mind running the following code?
> 
> ```

| Feature | Val|
|--------|--------|
| PANEL_RES_X| 160|
| PANEL_RES_Y| 20|
| mxconfig.drive| DP3246_SM5368|
|ClkPhase| True |

1. Video: 

https://github.com/user-attachments/assets/93ed43ad-6416-4806-8b4d-068625f11ea9

| Feature | Val|
|--------|--------|
| PANEL_RES_X| 160|
| PANEL_RES_Y| 20|
| mxconfig.drive| DP3246_SM5368|
|ClkPhase| False |

2. Video

https://github.com/user-attachments/assets/75a7b439-d1b9-4c13-a1b7-ac44e8495f91

mrcodetastic, 2025-02-23:
Well, that was fun.

Try the VirtualMatrixPanel_T class again and use the new FourScan option of 'FOUR_SCAN_40_80PX_HFARCAN'

mrcodetastic, 2025-02-23:
That is, use DP3246_SM5368 for the dma_display config. 
CHAIN_NONE
But then apply the FOUR_SCAN_40_80PX_HFARCAN for the Scan Type

mrcodetastic, 2025-02-23:
That is, use `DP3246_SM5368` for the dma_display config stuff. And then:

`#define PANEL_CHAIN_TYPE CHAIN_NONE`
 and
` #define PANEL_SCAN_TYPE  FOUR_SCAN_40_80PX_HFARCAN `

mrcodetastic, 2025-02-23:
Those two busted pixels on your panel are a loose soldering joint there, not a library issue. You'll need a hot air gun to fix etc.

mrcodetastic, 2025-02-23:
btw, those two busted pixels on the bottom right of your panel are a loose soldering joint, not a library issue. You'll need a hot air gun to fix etc.

hfarcan, 2025-02-23:
Thank you for all your effort. You are truly impressive! You managed to bring such a weird panel to this level even remotely. 
PS: As you will notice in the video, the LEDs in certain areas of the panel are blinking.

| Feature | Val|
|--------|--------|
| PANEL_RES_X| 80|
| PANEL_RES_Y| 40|
| mxconfig.drive| DP3246_SM5368|
|Panel Scan Type | FOUR_SCAN_40_80PX_HFARCAN|
|EXAMPLE_NUMBER| 2 |

1. Video: VirtualMatrixPanel Example 

https://github.com/user-attachments/assets/4f311b2e-4c58-404c-9e37-e8ca1327c917

 | Feature | Val|
|--------|--------|
| PANEL_RES_X| 80|
| PANEL_RES_Y| 40|
| mxconfig.drive| DP3246_SM5368|
|Panel Scan Type | FOUR_SCAN_40_80PX_HFARCAN|
|EXAMPLE_NUMBER| 2 |

2. Video Row Scanning: 
```

Loop Code: 
 void loop() {
    for (int i = 0; i < virtualDisp->height(); i++)
    {
      for (int j = 0; j < virtualDisp->width(); j++)
      {
        virtualDisp->drawPixel(j, i, virtualDisp->color565(255, 0, 0));
        delay(20);
      }
    }
    delay(2000);
    virtualDisp->clearScreen();
  } // end loop
```

https://github.com/user-attachments/assets/4bb112ca-068b-4dcc-be09-99fd6fd124e7

PS: As you will notice in the video, the LEDs in certain areas of the panel are blinking.

mrcodetastic, 2025-02-23:
Glad I could get it to work. That has to be one of the weirdest pixel mappings I've seen yet. But why generate more e-waste and spend money on a new panel when a code can solve! The videos helped immensely.

hfarcan, 2025-02-23:
Thank you so much for your help. 

>  But why generate more e-waste and spend money on a new panel when a code can solve!

That's a really good question. The unfortunate part is that local suppliers have no knowledge about the product, so the component documentation can't be trusted either. I’ve also started looking for a panel compatible with this library.

It could be completely my mistake, but when using the virtual screen, I didn’t get the desired result when I changed the font. Has an issue been created for this? I searched but couldn't find.

board707, 2025-02-23:
> That has to be one of the weirdest pixel mappings I've seen yet.

It is a one of the typical pixel layouts with a Pixelbase =16 :)

board707, 2025-02-23:
@mrcodetastic  @hfarcan 
It's should be the same as @mrcodetastic  FOUR_SCAN_40_80PX_HFARCAN pattern, but in more generic form - can be used with similar pattern with different pix_base
```
if constexpr (ScanType == FOUR_SCAN_40_80PX_HFARCAN) 
	{
	int panel_local_x = coords.x % 80; // compensate for chain of these panels 
	
if ( (((coords.y) / 10) % 2) ^  ((panel_local_x / panel_pixel_base) %2) )
		  {				
				coords.x += (((coords.x / panel_pixel_base) + 1) * panel_pixel_base);
		} else {
				coords.x += ((coords.x / panel_pixel_base) * panel_pixel_base);
		}
		
          coords.y =  = (coords.y % 10) + 10 * ((coords.y / 20) % 2); 
		
	}	
```

mrcodetastic, 2025-02-23:
This is the behaviour I observed from @hfarcan's video after it had sent out one single DMA line of (row 0) of 160px length.

Check if your code maps physical x and y to the right output coords.x and y ('electrical')

ROW 1 |   |   |   |   | ROW 2 |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | --
Electrical_Y | Electrical_X | Physical_Y | Physical_X |   | Electrical_Y | Electrical_X | Physical_Y | Physical_X
0 | 0 | 10 | 0 |   | 0 | 16 | 0 | 0
0 | 1 | 10 | 1 |   | 0 | 17 | 0 | 1
0 | 2 | 10 | 2 |   | 0 | 18 | 0 | 2
0 | 3 | 10 | 3 |   | 0 | 19 | 0 | 3
0 | 4 | 10 | 4 |   | 0 | 20 | 0 | 4
0 | 5 | 10 | 5 |   | 0 | 21 | 0 | 5
0 | 6 | 10 | 6 |   | 0 | 22 | 0 | 6
0 | 7 | 10 | 7 |   | 0 | 23 | 0 | 7
0 | 8 | 10 | 8 |   | 0 | 24 | 0 | 8
0 | 9 | 10 | 9 |   | 0 | 25 | 0 | 9
0 | 10 | 10 | 10 |   | 0 | 26 | 0 | 10
0 | 11 | 10 | 11 |   | 0 | 27 | 0 | 11
0 | 12 | 10 | 12 |   | 0 | 28 | 0 | 12
0 | 13 | 10 | 13 |   | 0 | 29 | 0 | 13
0 | 14 | 10 | 14 |   | 0 | 30 | 0 | 14
0 | 15 | 10 | 15 |   | 0 | 31 | 0 | 15
0 | 48 | 10 | 16 |   | 0 | 32 | 0 | 16
0 | 49 | 10 | 17 |   | 0 | 33 | 0 | 17
0 | 50 | 10 | 18 |   | 0 | 34 | 0 | 18
0 | 51 | 10 | 19 |   | 0 | 35 | 0 | 19
0 | 52 | 10 | 20 |   | 0 | 36 | 0 | 20
0 | 53 | 10 | 21 |   | 0 | 37 | 0 | 21
0 | 54 | 10 | 22 |   | 0 | 38 | 0 | 22
0 | 55 | 10 | 23 |   | 0 | 39 | 0 | 23
0 | 56 | 10 | 24 |   | 0 | 40 | 0 | 24
0 | 57 | 10 | 25 |   | 0 | 41 | 0 | 25
0 | 58 | 10 | 26 |   | 0 | 42 | 0 | 26
0 | 59 | 10 | 27 |   | 0 | 43 | 0 | 27
0 | 60 | 10 | 28 |   | 0 | 44 | 0 | 28
0 | 61 | 10 | 29 |   | 0 | 45 | 0 | 29
0 | 62 | 10 | 30 |   | 0 | 46 | 0 | 30
0 | 63 | 10 | 31 |   | 0 | 47 | 0 | 31
0 | 64 | 10 | 32 |   | 0 | 80 | 0 | 32
0 | 65 | 10 | 33 |   | 0 | 81 | 0 | 33
0 | 66 | 10 | 34 |   | 0 | 82 | 0 | 34
0 | 67 | 10 | 35 |   | 0 | 83 | 0 | 35
0 | 68 | 10 | 36 |   | 0 | 84 | 0 | 36
0 | 69 | 10 | 37 |   | 0 | 85 | 0 | 37
0 | 70 | 10 | 38 |   | 0 | 86 | 0 | 38
0 | 71 | 10 | 39 |   | 0 | 87 | 0 | 39
0 | 72 | 10 | 40 |   | 0 | 88 | 0 | 40
0 | 73 | 10 | 41 |   | 0 | 89 | 0 | 41
0 | 74 | 10 | 42 |   | 0 | 90 | 0 | 42
0 | 75 | 10 | 43 |   | 0 | 91 | 0 | 43
0 | 76 | 10 | 44 |   | 0 | 92 | 0 | 44
0 | 77 | 10 | 45 |   | 0 | 93 | 0 | 45
0 | 78 | 10 | 46 |   | 0 | 94 | 0 | 46
0 | 79 | 10 | 47 |   | 0 | 95 | 0 | 47
0 | 112 | 10 | 48 |   | 0 | 96 | 0 | 48
0 | 113 | 10 | 49 |   | 0 | 97 | 0 | 49
0 | 114 | 10 | 50 |   | 0 | 98 | 0 | 50
0 | 115 | 10 | 51 |   | 0 | 99 | 0 | 51
0 | 116 | 10 | 52 |   | 0 | 100 | 0 | 52
0 | 117 | 10 | 53 |   | 0 | 101 | 0 | 53
0 | 118 | 10 | 54 |   | 0 | 102 | 0 | 54
0 | 119 | 10 | 55 |   | 0 | 103 | 0 | 55
0 | 120 | 10 | 56 |   | 0 | 104 | 0 | 56
0 | 121 | 10 | 57 |   | 0 | 105 | 0 | 57
0 | 122 | 10 | 58 |   | 0 | 106 | 0 | 58
0 | 123 | 10 | 59 |   | 0 | 107 | 0 | 59
0 | 124 | 10 | 60 |   | 0 | 108 | 0 | 60
0 | 125 | 10 | 61 |   | 0 | 109 | 0 | 61
0 | 126 | 10 | 62 |   | 0 | 110 | 0 | 62
0 | 127 | 10 | 63 |   | 0 | 111 | 0 | 63
0 | 128 | 10 | 64 |   | 0 | 144 | 0 | 64
0 | 129 | 10 | 65 |   | 0 | 145 | 0 | 65
0 | 130 | 10 | 66 |   | 0 | 146 | 0 | 66
0 | 131 | 10 | 67 |   | 0 | 147 | 0 | 67
0 | 132 | 10 | 68 |   | 0 | 148 | 0 | 68
0 | 133 | 10 | 69 |   | 0 | 149 | 0 | 69
0 | 134 | 10 | 70 |   | 0 | 150 | 0 | 70
0 | 135 | 10 | 71 |   | 0 | 151 | 0 | 71
0 | 136 | 10 | 72 |   | 0 | 152 | 0 | 72
0 | 137 | 10 | 73 |   | 0 | 153 | 0 | 73
0 | 138 | 10 | 74 |   | 0 | 154 | 0 | 74
0 | 139 | 10 | 75 |   | 0 | 155 | 0 | 75
0 | 140 | 10 | 76 |   | 0 | 156 | 0 | 76
0 | 141 | 10 | 77 |   | 0 | 157 | 0 | 77
0 | 142 | 10 | 78 |   | 0 | 158 | 0 | 78
0 | 143 | 10 | 79 |   | 0 | 159 | 0 | 79

board707, 2025-02-24:
I usually don't use tables. I work from videos. The panel filling scheme is not too rare. I have already written pixel mapping for this scheme before. But I certainly don't rule out that I could have made a mistake. The final check always is to run it with a real panel.

hfarcan, 2025-02-24:
> [@mrcodetastic](https://github.com/mrcodetastic) [@hfarcan](https://github.com/hfarcan) It's should be the same as [@mrcodetastic](https://github.com/mrcodetastic) FOUR_SCAN_40_80PX_HFARCAN pattern, but in more generic form - can be used with similar patterns with different pix_base
> 
> ```
> if constexpr (ScanType == FOUR_SCAN_40_80PX_HFARCAN) 
> 	{
> 	int panel_local_x = coords.x % 80; // compensate for chain of these panels 
> 	
> if ( (((coords.y) / 10) % 2) ^  ((panel_local_x / panel_pixel_base) %2) )
> 		  {				
> 				coords.x += (((coords.x / panel_pixel_base) + 1) * panel_pixel_base);
> 		} else {
> 				coords.x += ((coords.x / panel_pixel_base) * panel_pixel_base);
> 		}
> 		
>           coords.y =  = (coords.y % 10) + 10 * ((coords.y / 20) % 2); 
> 		
> 	}	
> ```

Hi @board707 . Thank you for your effort. I'm testing the code here using Pixelbase 16. The desired results were not achieved with other Pixelbase values:

https://github.com/user-attachments/assets/082d29d5-de49-440a-880e-6ca704ce2907

board707, 2025-02-24:
Thank you for test.
As see, we need to swap an `if` and `else` branches:

```
if constexpr (ScanType == FOUR_SCAN_40_80PX_HFARCAN) 
	{
	int panel_local_x = coords.x % 80; // compensate for chain of these panels 
	
if ( (((coords.y) / 10) % 2) ^  ((panel_local_x / panel_pixel_base) %2) )
		  {				
				coords.x += ((coords.x / panel_pixel_base) * panel_pixel_base);
		} else {
				
                               coords.x += (((coords.x / panel_pixel_base) + 1) * panel_pixel_base);
		}
		
          coords.y =  = (coords.y % 10) + 10 * ((coords.y / 20) % 2); 
		
	}	
```

board707, 2025-02-24:
> The desired results were not achieved with other Pixelbase values:

Of course, it will run on your panel only with Pixelbase 16. When I talking about "different values",  I meant another panels with similar mapping scheme.

mrcodetastic, 2025-02-24:
```cpp

#include <iostream>
#include <iomanip> // For output formatting

struct Coords {
    int x;
    int y;
};

int main() {
    int panel_pixel_base = 16; // Updated pixel base
    Coords coords;

    // Header for easy Excel copying (Pipe-delimited format)
    std::cout << "Input X|Input Y|Output X|Output Y" << std::endl;

    // Loop for testing multiple inputs
    for (int y = 0; y < 40; y += 3) { // Updated y increment to 3
        for (int x = 0; x < 80; x += 10) {
            coords.x = x;
            coords.y = y;

            // Store original coordinates for display
            int input_x = coords.x;
            int input_y = coords.y;

            // Mapping logic
            int panel_local_x = coords.x % 80; // Compensate for chain of panels

            if ((((coords.y) / 10) % 2) ^ ((panel_local_x / panel_pixel_base) % 2)) {
                coords.x += ((coords.x / panel_pixel_base) * panel_pixel_base);
            } else {
                coords.x += (((coords.x / panel_pixel_base) + 1) * panel_pixel_base);
            }

            coords.y = (coords.y % 10) + 10 * ((coords.y / 20) % 2);

            // Output results in pipe-delimited format for easy Excel import
            std::cout << input_x << "|" << input_y << "|" << coords.x << "|" << coords.y << std::endl;
        }
    }

    return 0;
}```

Nothing like asking chatgpt to quickly turn this into a program you can run in cpp.sh to check the output matches my existing logic. Which it now does. 👌

mrcodetastic, 2025-02-24:
```cpp

#include <iostream>
#include <iomanip> // For output formatting

struct Coords {
    int x;
    int y;
};

int main() {
    int panel_pixel_base = 16; // Updated pixel base
    Coords coords;

    // Header for easy Excel copying (Pipe-delimited format)
    std::cout << "Input X|Input Y|Output X|Output Y" << std::endl;

    // Loop for testing multiple inputs
    for (int y = 0; y < 40; y += 3) { // Updated y increment to 3
        for (int x = 0; x < 80; x += 10) {
            coords.x = x;
            coords.y = y;

            // Store original coordinates for display
            int input_x = coords.x;
            int input_y = coords.y;

            // Mapping logic
            int panel_local_x = coords.x % 80; // Compensate for chain of panels

            if ((((coords.y) / 10) % 2) ^ ((panel_local_x / panel_pixel_base) % 2)) {
                coords.x += ((coords.x / panel_pixel_base) * panel_pixel_base);
            } else {
                coords.x += (((coords.x / panel_pixel_base) + 1) * panel_pixel_base);
            }

            coords.y = (coords.y % 10) + 10 * ((coords.y / 20) % 2);

            // Output results in pipe-delimited format for easy Excel import
            std::cout << input_x << "|" << input_y << "|" << coords.x << "|" << coords.y << std::endl;
        }
    }

    return 0;
}
```

Nothing like asking ChatGPT to quickly turn this into a program one can run at [cpp.sh](https://cpp.sh) quickly to test that the output matches my existing logic. Which it now does. 👌

hfarcan, 2025-02-24:
> Thank you for test. As see, we need to swap an `if` and `else` branches:

Thank you, it worked nicely now.
@mrcodetastic and @board707 You two are great men!