mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #767: New release for esp-idf projects without Arduino

IngmarStein, 2025-03-17:
I'm building a PlatformIO project with `framework = espidf`, i.e. not `framework = espidf, ardino` which uses a very old IDF version.

With that setup, I can't use the most recent release of this library with `mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display@^3.0.12` because its `library.json` uses `"frameworks": [ "arduino", "espidf" ]` which is then ignored by PlatformIO's library dependency checker because it thinks the library is not compatible with the project. This can be overridden with `lib_compat_mode = off` (the library then appears in the dependency graph), but compilation fails because https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/commit/c7cb4e30b1df48de5c01365ee9769df2a0aa313a is not part of the release.

Would it be possible to release a new version which includes this commit and optionally removes the arduino requirement?

Jason2866, 2025-03-17:
A quick solution is to use the Platformio fork [pioarduino](https://github.com/pioarduino/platform-espressif32) which provides actual Arduino core 3.1.3 which is based on IDF 5.3.2. The compile mode Arduino as a component from Arduino is supported too. In this case IDF 5.3.2 is used.

IngmarStein, 2025-03-17:
Sure, that would work, but I don't need the Arduino framework and building from HEAD or a pinned version works too: `lib_deps=https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA.git#15fd8671d99291a9865fb03fa4bbb5214c56b0e0`. I just can't use the most recent release (3.0.12 currently).

Jason2866, 2025-03-17:
Why not? Just did a PR to update/fix CI and there is branch master used and working.

IngmarStein, 2025-03-17:
I tried to explain in the initial description: with release 3.0.12, the library is skipped by library dependency finder _for projects not using the Arduino framework_ because library.json states that this framework is required.

Jason2866, 2025-03-17:
Removing is not a solution since this would remove compability for Arduino.
The entry does not mean it is a dependency, it declares only that the library is compatible to the listed entrys.

IngmarStein, 2025-03-17:
I don't see a platformio.ini file in https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/esp-idf/without-gfx?

My test project is https://github.com/tronbyt/firmware-http.

With
```
[env]
platform = espressif32 @ 6.10.0
framework = espidf

[env:tidbyt]
lib_deps =
	https://github.com/webmproject/libwebp.git#1d86819f49edc8237fa2b844543081bcb8ef8a92
	mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display@^3.0.12
```

I get:

```
Framework incompatible library /Users/ingmar/Developer/tronbyt-firmware-http/.pio/libdeps/tidbyt/ESP32 HUB75 LED MATRIX PANEL DMA Display
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
…
src/display.cpp:3:10: fatal error: ESP32-HUB75-MatrixPanel-I2S-DMA.h: No such file or directory
```

With

```
lib_deps =
	https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA.git#15fd8671d99291a9865fb03fa4bbb5214c56b0e0
```

The project compiles cleanly.

Jason2866, 2025-03-17:
`platform = espressif32 @ 6.10.0` uses IDF 5.4.0.  This lib is not tested against IDF 5.4.0

IngmarStein, 2025-03-17:
Ah, I think what pointed me in the Arduino direction was the fact that when I disable the dependency checker (`lib_compat_mode = off`), the compilation fails due to a missing Arduino header (due to the commit I referenced in the initial   description not being part of the release).

I couldn't spot any supported IDF versions in https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/library.json, do you know where this requirement for 5.3.2 is specified?

IngmarStein, 2025-03-17:
`pio pkg` shows `arduino` as the _only_ compatible framework. There's no `espidf`. Is that expected?

```
❯ pio pkg show "mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display"

mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display
Library • 3.0.12 • Public • Published on Sun Jul 28 22:40:32 2024

HUB75 LED Matrix Library for ESP32, ESP32-S2 and ESP32-S3. An Adafruit GFX compatible library for LED matrix modules which uses DMA for ultra-fast refresh rates and therefore very low CPU usage.

---------------------  ----------------------------------------------------------------------------------------------------------
Registry               https://registry.platformio.org/libraries/mrfaptastic/ESP32%20HUB75%20LED%20MATRIX%20PANEL%20DMA%20Display
Homepage               https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA
Repository             https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA.git
License                MIT
Popularity             255
Stars                  898
Examples               15
Installed Size         494.21KB
Compatible Platforms   espressif32
Compatible Frameworks  arduino
Keywords               display
---------------------  ----------------------------------------------------------------------------------------------------------

Version    Size      Published
---------  --------  -------------------
3.0.12     124.36KB  2024-07-28 22:40:32
```

Jason2866, 2025-03-17:
Now you found the reason :-)

IngmarStein, 2025-03-17:
I'm new to PlatformIO and its dependency management system and assumed that the Registry uses `library.json` as input, but it seems to be separate? Should it be updated then to list `espidf` as well?

Jason2866, 2025-03-17:
Such issues and changes in used libs, is the reason why i never use libs from sources outside my repo. Copy everything needed in repo libs folder. So it is safe to have a reproducible build and no weird issues when  libs changes or not available for whatever reason.

IngmarStein, 2025-03-17:
I used a pinned version in `lib_deps` for reproducible builds (doesn't guard against unavailability, of course).

IngmarStein, 2025-08-06:
Looks like the registry wasn't updated for 3.0.13. It still shows only `arduino` as compatible framework:

```
❯ pio pkg show "mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display"

mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display
Library • 3.0.13 • Public • Published on Sat Aug  2 13:22:56 2025

HUB75 LED Matrix Library for ESP32, ESP32-S2 and ESP32-S3. An Adafruit GFX compatible library for LED matrix modules which uses DMA for ultra-fast refresh rates and therefore very low CPU usage.

---------------------  ----------------------------------------------------------------------------------------------------------
Registry               https://registry.platformio.org/libraries/mrfaptastic/ESP32%20HUB75%20LED%20MATRIX%20PANEL%20DMA%20Display
Homepage               https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA
Repository             https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA.git
License                MIT
Popularity             261
Stars                  1231
Examples               15
Installed Size         515.88KB
Compatible Platforms   espressif32
Compatible Frameworks  arduino
Keywords               display
---------------------  ----------------------------------------------------------------------------------------------------------

Version    Size      Published
---------  --------  -------------------
3.0.13     132.63KB  2025-08-02 13:22:56
```

IngmarStein, 2025-08-06:
The registry entry is already there. I'm just asking for it to be updated to not require Ardunio after this library was changed to work without it.

Jason2866, 2025-08-06:
Yes, thats the pain with the Platformio Registry. Correcting or updating is so time consuming and complicated

board707, 2025-08-06:
I don't see any benefits to use Platformio at all.

Platformio had its advantages over the old Arduino IDE v1.x. After the release of version 2.x and integration with VS Code, there was no reason to choose Platformio now.

IngmarStein, 2025-08-06:
Maybe @mrcodetastic wants to chime in whether PlatformIO should be supports in the future or not and whether the registry entry could be updated.