mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #769: P4 40x80 1/20 indoor panel not working properly

denizcelik213, 2025-03-19:
I'm using P4 40x80 1/20S indoor panel and not working properly. Here is my code and display photo

```
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include <Adafruit_GFX.h>

// Panel dimensions
#define PANEL_RES_X 40 // Panel width
#define PANEL_RES_Y 80 // Panel height
#define PANEL_CHAIN 1  // Number of panels in chain

// Change these to whatever suits
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 19
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32 // required for 1/32 scan panels, like 64x64px. Any available pin would do, i.e. IO32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16

// Initialize DMA matrix panel
MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup()
{
  Serial.begin(115200);

  HUB75_I2S_CFG::i2s_pins _pins = {
      R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
      A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
      LAT_PIN, OE_PIN, CLK_PIN};

  HUB75_I2S_CFG mxconfig(
      PANEL_RES_X, // DO NOT CHANGE THIS
      PANEL_RES_Y, // DO NOT CHANGE THIS
      PANEL_CHAIN, // DO NOT CHANGE THIS
      _pins        // Uncomment to enable custom pins
  );

  // Initialize display
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(128); // Brightness (0-255)
  dma_display->clearScreen();

  // Configure text properties
  dma_display->setTextSize(1);
  dma_display->setTextColor(dma_display->color565(255, 255, 255)); // White text

  // Position text near center of panel
  dma_display->setCursor(2, 35);
  dma_display->print("Hello");

  Serial.println("Hello written on the panel");
}

void loop()
{
  // Add screen updates or animations here if needed
  delay(1000);
}
```

[image]

And I'm researching issues than find Virtual Matrix QuarterScan settings. Than tested it. Here is my code and display photo.

```
/*************************************************************************
   Description:

   The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
   supports output to HALF scan panels - which means outputting
   two lines at the same time, 16 or 32 rows apart if a 32px or 64px high panel
   respectively.
   This cannot be changed at the DMA layer as it would require a messy and complex
   rebuild of the library's internals.

   However, it is possible to connect QUARTER (i.e. FOUR lines updated in parallel)
   scan panels to this same library and
   'trick' the output to work correctly on these panels by way of adjusting the
   pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
   library.

 **************************************************************************/

/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
   correctly on a 32x16 1/4 Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
#include <Adafruit_GFX.h> // Include the library for the display functions

#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13

#define A_PIN 23
#define B_PIN 22
#define C_PIN 5
#define D_PIN 17
#define E_PIN 32

#define CLK_PIN 16
#define LAT_PIN 4
#define OE_PIN 15

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
public:
  using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

protected:
  VirtualCoords getCoords(int16_t x, int16_t y); // custom getCoords() method for specific pixel mapping
};

// define custom getCoords() method for specific pixel mapping
// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y)
{

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if (coords.x == -1 || coords.y == -1)
  { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

  uint8_t pxbase = 80; // pixel base
  if ((coords.y % 20) < 10)
  {
    coords.x += (coords.x / pxbase + 1) * pxbase;
  }
  else
  {
    coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
  }

  coords.y = (coords.y / 20) * 10 + (coords.y % 10);

  return coords;
}

/* ================================================== */
// Panel configuration
#define PANEL_RES_X 80 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 40 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println("");
  Serial.println("");
  Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/8 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

  // 62x32 1/8 Scan Panels don't have a D and E pin!

  HUB75_I2S_CFG::i2s_pins _pins = {
      R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
      A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
      LAT_PIN, OE_PIN, CLK_PIN};

  HUB75_I2S_CFG mxconfig(
      PANEL_RES_X * 2,    // DO NOT CHANGE THIS
      PANEL_RES_Y / 2,    // DO NOT CHANGE THIS
      NUM_ROWS * NUM_COLS // DO NOT CHANGE THIS

      //_pins // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
  mxconfig.driver = HUB75_I2S_CFG::ICN2038S;
  // mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40); // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if (not dma_display->begin())
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
  // **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
  // FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);

  dma_display->begin(); // Start the display

  dma_display->setTextColor(dma_display->color565(255, 255, 255)); // Set text color to white
  dma_display->setTextSize(1);                                     // Set text size. Adjust as needed for your specific panel size
  dma_display->setCursor(0, 0);                                    // Set cursor to top left corner of the display

  dma_display->println("hello"); // Print "hello" at the current cursor position
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop()
{

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 0, 0));
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```
https://youtu.be/eFhuHj8eZb0

How can I fix it ?

board707, 2025-03-19:
Your panel has a scan 20s - a half of 40pixels height. So it is a standard DoubleScan panel, not a QuarterScan one. 
You don't need any VirtualPanel methods to work with it.

denizcelik213, 2025-03-19:
FM6124D 

[image]

board707, 2025-03-19:
There should be at least one another type of chips.

denizcelik213, 2025-03-20:
[image]

[image]

Okey. I found these chips.

denizcelik213, 2025-03-23:
I still haven't solved the problem. The chips in my display are FM6124D and CNCC29. what should I do? @board707

board707, 2025-03-23:
> The chips in my display are FM6124D and CNCC29.

The first is conventional led driver, the second chip is "3-to-8" type line switcher TC7262 - the both chips are standard type that not needs any special setting to work.

As a conclusion, I don't see anything in your panel, that can prevents using it with the library.

board707, 2025-03-23:
> // Panel dimensions
> #define PANEL_RES_X 40 // Panel width
> #define PANEL_RES_Y 80 // Panel height

You mixed up the height and width of the panel.