mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #771: 64x64 (1/32 Scan) Panel only showing Half of the pixels

magicroomy, 2025-03-22:
I bought a Panel on AliExpress. 64x64 Pixel 1/32 Scan.
Chips on the panel read:
RUC7258D
ICN2037BP
MW245BC

My Problem is that instead of a 64x64 pic I get 4 stripes. The plasma animation seems to work tho (Visible in the Stripes).

[image]

I adapted the Plasma Example.
Config part looks like this:

//Attached to an ESP32

#define R1 25
#define G1 26
#define BL1 27
#define R2 14
#define G2 12
#define BL2 13
#define CH_A  23
#define CH_B  19
#define CH_C  5
#define CH_D  17
#define CH_E  22 
#define LAT 4
#define OE  15
#define CLK 16

HUB75_I2S_CFG mxconfig;  
    // those are the defaults
    mxconfig.mx_width = 64;                   // physical width of a single matrix panel module (in pixels, usually it is always 64 ;) )
    mxconfig.mx_height = 64;                  // physical height of a single matrix panel module (in pixels, usually almost always it is either 32 or 64)
    mxconfig.chain_length = 1;                // number of chained panels regardless of the topology, default 1 - a single matrix module
    mxconfig.gpio.r1 = R1;                    // pin mappings
    mxconfig.gpio.g1 = G1;
    mxconfig.gpio.b1 = BL1;
    mxconfig.gpio.r2 = R2;                    // pin mappings
    mxconfig.gpio.g2 = G2;
    mxconfig.gpio.b2 = BL2;
    mxconfig.gpio.a = CH_A;                   // pin mappings
    mxconfig.gpio.b = CH_B;
    mxconfig.gpio.c = CH_C;
    mxconfig.gpio.d = CH_D;
    mxconfig.gpio.e = CH_E;                   // pin mappings
    mxconfig.gpio.lat = LAT;                  // pin mappings
    mxconfig.gpio.oe = OE;                    // pin mappings
    mxconfig.gpio.clk = CLK;                  // pin mappings
                        // etc
    mxconfig.driver = HUB75_I2S_CFG::SHIFTREG;   // shift reg driver, default is plain shift register

board707, 2025-03-22:
Check the connections of the A B C D E lines, especially of D

closed 2025-03-24 by mrcodetastic

Smango07, 2025-06-05:
After 2 days of troubleshooting I stumbled on this solution. My panel has the D pin mislabeled also. Over a year later and still not corrected.