mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #772: 1/16 scan 64x32 256x128mm P4 FLICKERING/GHOSTING

namato18, 2025-03-23:
I have attempted using shorter wires for the io pins (as mentioned in similar issues) with the same result. I have also attempted ensuring a real solid connection to power and gnd for the esp32, while that can help a bit, the issue still remains. I understand that the io pins of an esp32 are outputting at 3.3v, but these boards generally are built for 5v signals. As you can see in the images, they come through relatively well, with flickering and ghosting occurring for certain displays. My best guess is this is some sort of noise/interference due to the 3.3v signal.

You'll notice in the first picture that I'm only using one power input and one ground input for the LED matrix from the power supply. The power line on the left of the JST-VH connector is being used to power the ESP32, while the gnd line on the right side of the JST-VH isn't connected to anything for the time being. I've replicated this setup with 2 power/gnd lines coming from the power supply as well (esp32 powered by power supply as well), same result.

The LED matrix in use can be found [here](https://www.aliexpress.us/item/2251832328529483.html?spm=a2g0o.order_list.order_list_main.11.570118023IUj8b&gatewayAdapt=glo2usa)

The power supply in use is 5v 4a. I've attempted with 5v 8a to see if that was an issue as well.

[image]

[image]

[image]

[image]

[image]

The last remaining solution that I have in mind right now would be to use a level shifter between the esp32 and the LED matrix to bump up the signals from 3.3v to 5v. I'd love to hear other ideas on this, or if anyone has run into this exact problem?

In any case, I really appreciate and thank everyone for any advice or help!

mrcodetastic, 2025-03-24:
The Cables are way too long and likely causing noise/capacitance. 

Make sure you're connection one of the GND lines on the HUB75 connector to a GND pin on the ESP32 as well.

namato18, 2025-03-26:
Unfortunately even 10cm wires from the esp32 to hub75 still cause the ghosting/flickering. HUB75 GND is connected to ESP32 GND as well. I've ordered some wires with (hopefully) better/more secure dupont connectors. I'll see if a more secure connection helps out!

mrcodetastic, 2025-03-26:
How about the mxconfig.clkphase setting, does changing that help?

mrcodetastic, 2025-03-26:
Try turning the clock frequency down, see if that helps.

If you're running an ESP32 (i.e. the original ESP32, not the ESP32-S3, S2 etc.), then try [hacking this line](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/aed04adfcda1838bf85c629a8c3b560919b3a327/src/platforms/esp32/esp32_i2s_parallel_dma.cpp#L300).

Set the clock divider to 10 or something (i.e. 4mhz). Sure, it'll flicker, but if there's no ghosting then it's something to do with these panels not being able to handle the frequency. Perhaps you do need a logic level converter.

I would suggest simply bring a WF1 or WF2 as they have all the stuff you need including the logic level converter, built in: https://github.com/mrcodetastic/HD-WF1-WF2-LED-MatrixPanel-DMA

namato18, 2025-03-26:
Thank you for the advice, unfortunately there is still some flicker/ghosting. for reference, the esp32 I'm using is [this one](https://www.amazon.com/Teyleten-Robot-ESP-WROOM-32-Development-Microcontroller/dp/B08246MCL5/ref=sr_1_5?crid=UKMEKTEDNADI&dib=eyJ2IjoiMSJ9.9FsF1_g0F_gENoTNxylVzhlE-f5AuPRamTr33nRxSXkHnJjRSu5AAlWSq4oPIlEm15KkLnJxbhyKlbG0JZjwziCS7QhJdUS-aWBg85odsILgLhq4IvMQcsrRJanSDHJbGzFBf4Enk-U7bQeOauv8tui7zi_1yxbprg3RQr1U3zS6PXAbIIVujkHveGiqNVVkfdsvhoHGCYzY32bBA0kI8RR-InWveAY3vDawTnsW15E.M5AB_O-LzJ4_ftmaDmQBL99ud527RWO8SseBL0eW2u8&dib_tag=se&keywords=esp32%2Bdevelopment%2Bboard&qid=1743020270&sprefix=esp32%2Bdevelopment%2Bboar%2Caps%2C117&sr=8-5&th=1) which I believe is just the original esp32?

I tried a modified example of the scrolling text example from this repo. I noticed that it seems to be only the left side of the panel that's having issues with flickering/ghosting, everything comes onto the screen just fine originally. I consulted the magical chatGPT and it seems to think it's probably a timing issue, as you mentioned, or a power supply issue. Perhaps I'll order a WF1 and see where that gets me.

[image]

[image]

mrcodetastic, 2025-03-27:
I think it's a defective panel to be honest. Get a refund.

namato18, 2025-03-27:
I got 5 of them, it looks like the same issue is persistent across them all. Could it be that the manufacturer just had a bad batch? Or if I buy from a different manufacturer, am I likely to get the same result?

board707, 2025-03-27:
@namato18 
To be sure, if you have a Raspberry Pico (RP2040) type board, you could test your panels with it.

namato18, 2025-04-13:
I was actually able to get these panels to work without flickering/ghosting using an esp8266 NodeMcu and the PxMatrix library. I'm curious as to why that process works? Does an esp8266 output slightly higher voltage than an esp32 via the io pins? Is it the multiplexing wiring setup that distributes the power slightly better? As a follow up to all those questions, is it possible to multiplex using this library (I prefer the syntax of this library)

mrcodetastic, 2025-04-21:
Those libraries bitbang the GPIO output using the CPU and as a result do so at a much slower frequency. 

Perhaps these panels aren't designed to be driven at a high frequency that this library does.

mrcodetastic, 2025-08-01:
> I had similar issues with using a bare ESP32 S3, issues remained the same when using one of the breakout boards and panel original wiring. Then I bought the HUIDU WF2, flashed it with this library and all issues disappeared using the same wiring and panel. I guess these boards use some additional chips to get a proper signal (LLC's and caps?).

These HUB75 LED panels are supposed to be driven by a 5V signal, hence why the Huidu HUB75 have a logic level converter between the ESP32 and the HUB75 connector increase the signal voltage from 3.3V to 5V.

Many panels seem to work fine with the raw 3.3volt GPIO output from the ESP32s, but that's just good luck.

yskhee, 2025-08-08:
> > I had similar issues with using a bare ESP32 S3, issues remained the same when using one of the breakout boards and panel original wiring. Then I bought the HUIDU WF2, flashed it with this library and all issues disappeared using the same wiring and panel. I guess these boards use some additional chips to get a proper signal (LLC's and caps?).
> 
> These HUB75 LED panels are supposed to be driven by a 5V signal, hence why the Huidu output is after logic level converter between the ESP32 and the HUB75 connector to increase the signal voltage from 3.3V to 5V.
> 
> Many panels seem to work fine with the raw 3.3volt GPIO output from the ESP32s, but that's just good luck.

I m using ESP32, P10 LED board, 1/8 scan, 32x16. I found that the flickering/ghosting issue is less apparent when i set dma_display->setTextSize(2); using the default font type. However, the display looks weird when i setTextSize(1). I try to create my own bitmap font to suit my need, for example 7x11 pixel, the ghosting/flickering come back again. 
There is no 74hct245 level up (3.3v to 5v) in between my LED and ESP32 because the supplier claimed that it is built in within the LED board.

board707, 2025-08-08:
> There is no 74hct245 level up (3.3v to 5v) in between my LED and ESP32 because the supplier claimed that it is built in within the LED board.

The buffer chips on the panel should be clearly seen

mrcodetastic, 2025-08-24:
> i guess should be something that i need to adjust in the coding

Without seeing your code I can't say. But if you're trying to implement some scrolling text, or doing the following:

a) writing text to the screen
b) clearing the screen (i.e. black)
(looping)

Then you'll get flicker unless you use double buffering.