mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #775: ESP32-S3 PIO_TestPatterns crash in gdma_lcd_parallel16.cpp (Platformio)

Tuet, 2025-04-02:
I am trying to run the PIO_TestPatterns example in Platformio on an ESP32-S3, but I am out of luck and it crashes somwhere in gdma_lcd_parallel16.cpp. I tried just to program several ESP32-S3 boards (board = esp32-s3-devkitc-1), run the code with high debug level, not connected to a display yet, just looking at the serial CDC output. I tried with a non -S3 board (only ESP32) and at least I don't see this crash.

I just opened Platformio and opened the PIO_TestPatterns folder and added the following env to platformio.ini:
`
[env:esp32s3]
framework = arduino
;board = wemos_d1_mini32
board = esp32-s3-devkitc-1
board_upload.flash_size = 4MB 
board_build.partitions = default.csv
build_flags = 
  -DCORE_DEBUG_LEVEL=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args 
  -DARDUINO_USB_MODE=1
  -DARDUINO_USB_CDC_ON_BOOT=1  
  -DBOARD_HAS_PSRAM
`
Any ideas, comments what I did wrong? Would like to use ESP32-S3 boards instead of old ESP32 boards for it.

Jason2866, 2025-04-03:
There is the `platform` entry missing. Try

```
[env:esp32s3]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip
framework = arduino
;board = wemos_d1_mini32
board = esp32-s3-devkitc-1
board_upload.flash_size = 4MB 
board_build.partitions = default.csv
build_flags = 
  -DCORE_DEBUG_LEVEL=5 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args 
  -DARDUINO_USB_MODE=1
  -DARDUINO_USB_CDC_ON_BOOT=1  
  -DBOARD_HAS_PSRAM
```

Tuet, 2025-04-27:
Thank you for trying to help me. I have now switched to pioarduino (latest release and changed from Platformio to pioduino in VSCode) AND started with 1_SimpleTestShapes. It works now on ESP32S3 - even with a much changed Pin-Mapping.