mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #776: What am i doing wrong, only some pixels light up?

fwarmuth, 2025-04-08:
Hey,
I checked all connections, but i guess i am doing things wrong.
The gif shows the default `PIO_TestPatterns` example.
What do i need to do?

cheers!

[image]

mrcodetastic, 2025-04-08:
What's the code?

fwarmuth, 2025-04-08:
As i wrote:
`https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/master/examples/PIO_TestPatterns/src/main.cpp`

out of the box `PIO_TestPatterns` example.

fwarmuth, 2025-04-08:
Hey,
as i wrote:
Panel/Matrix Information:
```
Dimensions: 160mm × 80mm
Pixel: 64×32=2048 DOTS
Pitch: 2.5mm
Scan: 1/16 scan
```
Here most chips have nothing written, the ones close to power input have:
```
RUC7258D
OAE41287
```

here some images:

[image]

[image]

board707, 2025-04-08:
According to the picture, you have a problem with ABCDE lines. Test the connections, especially of B pin.

fwarmuth, 2025-04-09:
Thanks,
I now soldered everything to avoid bad jumpercables. I also doubled checked all the connections, all are fine.
i connected the panel as follows:

[image]

Still same behavior.
Any additional ideas?

davemaster, 2025-04-09:
Greetings

If you have a 64x32 panel, why, WHY aren't changed these lines:

// Configure for your panel(s) as appropriate!
//#define PIN_E 32
#define PANEL_WIDTH 64
//#define PANEL_HEIGHT 64         // Panel height of 64 will required PIN_E to be defined.
**#define PANEL_HEIGHT 32**

**#ifdef VIRTUAL_PANE
 #define PANELS_NUMBER 1         // Number of chained panels, if just a single panel, obviously set to 1
#else
 #define PANELS_NUMBER 1         // Number of chained panels, if just a single panel, obviously set to 1
#endif**

TRY, and let us know, WITH THE CODE IN YOUR COMPUTER, not in github please....

fwarmuth, 2025-04-09:
ps. beside the change of panel layout, i  changed channel `B` pin to `22` as my connection diagram fits, and enabled the pin redefinition.

Here my `git diff`, so you don't need to look for the diff:
<details>
  <summary>main.cpp</summary>
```
--- a/examples/PIO_TestPatterns/src/main.cpp
+++ b/examples/PIO_TestPatterns/src/main.cpp
@@ -28,8 +28,8 @@
 // CLK| LAT
 // OE | GND

-/*  Default library pin configuration for the reference
-  you can redefine only ones you need later on object creation
+// /*  Default library pin configuration for the reference
+//   you can redefine only ones you need later on object creation
 #define R1 25
 #define G1 26
 #define BL1 27
@@ -37,24 +37,23 @@
 #define G2 12
 #define BL2 13
 #define CH_A 23
-#define CH_B 19
+#define CH_B 22
 #define CH_C 5
 #define CH_D 17
 #define CH_E -1 // assign to any available pin if using panels with 1/32 scan
 #define CLK 16
 #define LAT 4
 #define OE 15
-*/

 // Configure for your panel(s) as appropriate!
 #define PIN_E 32
 #define PANEL_WIDTH 64
-#define PANEL_HEIGHT 64         // Panel height of 64 will required PIN_E to be defined.
+#define PANEL_HEIGHT 32         // Panel height of 64 will required PIN_E to be defined.

 #ifdef VIRTUAL_PANE
  #define PANELS_NUMBER 4         // Number of chained panels, if just a single panel, obviously set to 1
 #else
- #define PANELS_NUMBER 2         // Number of chained panels, if just a single panel, obviously set to 1
+ #define PANELS_NUMBER 1         // Number of chained panels, if just a single panel, obviously set to 1
 #endif

 #define PANE_WIDTH PANEL_WIDTH * PANELS_NUMBER
```
  
</details>

board707, 2025-04-09:
> beside the change of panel layout, i changed channel `B` pin to `22`

As I already wrote, the matrix problems look like your pin B is not working. And if it was pin B that you changed, it is unlikely to be a coincidence.

fwarmuth, 2025-04-09:
Ok, right now i just try things out... and noticed the following:

When i change back so that `CH_B` is `19` and connect GPIO19 with my panel, it works!
But, when i configure `CH_B` to be `22`, and connect GPIO22 with the panel doesn't work!

Conclusion:
My GPIO22 is broken - seems strange, its a brand new ESP...
Or
the `CH_B` macro is not used correctly - also strange...

Its fine for me.

board707, 2025-04-09:
> When i change back so that `CH_B` is `19` and connect GPIO19 with my panel, it works!

So I guessed right :)

fwarmuth, 2025-04-11:
I will close this!
Thank you for very fast support guys, keep it on!

closed 2025-04-11 by fwarmuth