mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #777: How to install to esp-idf

airCoder2, 2025-04-11:
Hello, How would I use this library with esp-idf? Thanks!

abrender, 2025-04-11:
Hi @airCoder2  - there are currently [2 examples](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/esp-idf) for using `esp-idf` - one with the Adafruit GFX library, and one without the Adafruit GFX library.

I hope this helps :)

airCoder2, 2025-04-18:
> Hi [@airCoder2](https://github.com/airCoder2) - there are currently [2 examples](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/esp-idf) for using `esp-idf` - one with the Adafruit GFX library, and one without the Adafruit GFX library.
> 
> I hope this helps :)

I am not sure if the example works. When I tried to build, it output some issues with some of the used functions. When I asked AI, it suggested downgrading my ESP-IDF, which I did. But then Python also had to be downgraded. I don't know if I want to do that. Is there another way of doing it?
Thank you!