mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #779: Issues with `SPIRAM_FRAMEBUFFER`

ashthespy, 2025-04-17:
I don't seem to be able to get the framebuffer working with a S3. ([`ESP32-S3-DEV-KIT-N8R8`](https://www.waveshare.com/wiki/ESP32-S3-DEV-KIT-N8R8)). The board *should* have octal SPI RAM as per the datasheet. 

<details close>
<summary>Snippet I am testing with </summary>

```c++

#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#include <array>

#pragma message "Compiling for ESPS3 board pins"
#define R1 4
#define G1 6
#define BL1 5
#define R2 7
#define G2 16
#define BL2 15

#define CH_A 17
#define CH_B 18
#define CH_C 8
#define CH_D 3

#define LAT 9
#define OE 10
#define CLK 11
#define CH_E -1  // assign to pin 14 if using more than two panels

MatrixPanel_I2S_DMA *display = nullptr;

constexpr std::size_t color_num = 5;
using colour_arr_t = std::array<uint16_t, color_num>;

uint16_t myDARK, myWHITE, myRED, myGREEN, myBLUE;
colour_arr_t colours;

struct Square {
  float xpos, ypos;
  float velocityx;
  float velocityy;
  boolean xdir, ydir;
  uint16_t square_size;
  uint16_t colour;
};

const int numSquares = 25;
Square Squares[numSquares];

void setup() {
  // put your setup code here, to run once:
  delay(1000);
  Serial.begin(115200);
  delay(200);

  Serial.println("...Starting Display");
  // Initialize the panel.
  HUB75_I2S_CFG::i2s_pins pins = {R1, G1, BL1, R2, G2, BL2, CH_A,
                                  CH_B, CH_C, CH_D, CH_E, LAT, OE, CLK};
  HUB75_I2S_CFG mxconfig(64,    // width
                         32,    // height
                         1,     // chain length
                         pins,  // pin mapping
                         HUB75_I2S_CFG::SHIFTREG,
                         true,  // double-buffering
                         HUB75_I2S_CFG::HZ_10M,
                         1,    // latch blanking
                         true  // invert clock phase
  );
[... 137 lines cut ...]
------------------------------------------
  Total Size        :   385248 B ( 376.2 KB)
  Free Bytes        :   320244 B ( 312.7 KB)
  Allocated Bytes   :    59252 B (  57.9 KB)
  Minimum Free Bytes:   320156 B ( 312.7 KB)
  Largest Free Block:   253940 B ( 248.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8352808 B (8157.0 KB)
  Allocated Bytes   :    32912 B (  32.1 KB)
  Minimum Free Bytes:  8352808 B (8157.0 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------
    19 : USB_DM
    20 : USB_DP
    43 : UART_TX[0]
    44 : UART_RX[0]
============ After Setup End =============
```

ashthespy, 2025-04-17:
Okay the error is quite trivial in the end. 
 https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/pull/718 introduced new API that explicitly needed `transfer_config.access_ext_mem=true` 

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/4c7585debe783c05eee0b671b93dda021ec3e289/src/platforms/esp32s3/gdma_lcd_parallel16.cpp#L223-L230

But `SPIRAM_DMA_BUFFER` is only defined in `platform_detect.hpp`. Since the `CMakeLists.txt` automatically picks up these files, it's compiled already before platform_detect does it's thing. 

Even if this didn't happen we currently have `SPIRAM_DMA_BUFFER` after the `#include "esp32s3/gdma_lcd_parallel16.hpp"`

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/4c7585debe783c05eee0b671b93dda021ec3e289/src/platforms/platform_detect.hpp#L32-L44

How would you prefer this to be cleaned up? To be a bit more scalable, we could move the CMake platform picking stuff to `platform_detect.*`?

mrcodetastic, 2025-04-20:
Thanks for investigating this and finding the fix. I'll probably get rid of this compile time define in this part of the library. 

I'll implement a fix in the coming days. I hope you have a good Easter.

mrcodetastic, 2025-05-04:
I've now removed the 'SPIRAM_FRAMEBUFFER' define as it's pointless have a define that defines another define.

In your platformio.ini build_flags have `-DSPIRAM_DMA_BUFFER=1` instead

ashthespy, 2025-05-09:
I spent the afternoon and have hit a wall. 

My board ([ESP32-S3-DEV-KIT-N8R8](https://www.waveshare.com/wiki/ESP32-S3-DEV-KIT-N8R8)) seems to be working fine with the default clock divider divisor 12 / 13.33 MHz - but only with the Arduino framework.  On ESP_IDF, it fails unless I reduce this to 10Mhz. 

No I am not sure I am comparing apples to apples- so will close this issue :-)

closed 2025-05-09 by ashthespy