mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #781: P5 64x32 1/16 SM16208SJ, MW245BC HELP

tinicoweb, 2025-04-17:
Hi there!
I have panel and it fill as shown in the video:

https://github.com/user-attachments/assets/8c9318b4-a13c-46fc-ab98-3d8ce4d03c90

Help me with mapping, please

This is the code

`
/*************************************************************************
   Description:
   The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
   supports output to HALF scan panels - which means outputting
   two lines at the same time, 16 or 32 rows apart if a 32px or 64px high panel
   respectively.
   This cannot be changed at the DMA layer as it would require a messy and complex
   rebuild of the library's internals.
   However, it is possible to connect QUARTER (i.e. FOUR lines updated in parallel)
   scan panels to this same library and
   'trick' the output to work correctly on these panels by way of adjusting the
   pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
   library.
 **************************************************************************/
/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
   correctly on a Four Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)
  protected:
    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping
};
// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }
 if ((coords.y & 8) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
  return coords;
}
/* ================================================== */
// DEFINICION DE LOS PINES PERSONALIZADOS
/* ================================================== */
#define R1_PIN 23
#define G1_PIN 22
#define B1_PIN 1
#define R2_PIN 3
#define G2_PIN 21
#define B2_PIN 19
#define A_PIN  18
#define B_PIN  5
#define C_PIN  17
#define D_PIN  16
#define E_PIN  -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 32
#define OE_PIN  33
#define CLK_PIN 25
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS
// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN
// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;
// placeholder for the virtual display object
EightPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/8 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

/* ================================================== */
// ACTIVA LOS PINES PERSONALIZADOS
/* ================================================== */
  // 62x32 1/8 Scan Panels don't have a D and E pin!
  HUB75_I2S_CFG::i2s_pins _pins = {
  R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
  A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
  LAT_PIN, OE_PIN, CLK_PIN
  };
 
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // DO NOT CHANGE THIS
    PANEL_RES_Y / 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    ,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(100);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,    VIRTUAL_MATRIX_CHAIN_TYPE);

  // THE IMPORTANT BIT BELOW!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
 // FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(20);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop

`

dates of panel
[image]

board707, 2025-04-17:
Hi
You don't need a Virtual panel code for 64x32 1/16 panel.

tinicoweb, 2025-04-18:
try the example code

`
// Example sketch which shows how to display some patterns
// on a 64x32 LED matrix
//

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another

/* ================================================== */
// DEFINICION DE LOS PINES PERSONALIZADOS
/* ================================================== */
#define R1_PIN 23
#define G1_PIN 22
#define B1_PIN 1
#define R2_PIN 3
#define G2_PIN 21
#define B2_PIN 19
#define A_PIN  18
#define B_PIN  5
#define C_PIN  17
#define D_PIN  16
#define E_PIN  -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 32
#define OE_PIN  33
#define CLK_PIN 25

//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;

/* ================================================== */
// ACTIVA LOS PINES PERSONALIZADOS
/* ================================================== */
  HUB75_I2S_CFG::i2s_pins _pins = {
  R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
  A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
  LAT_PIN, OE_PIN, CLK_PIN
  };
 

uint16_t myBLACK, myWHITE, myRED, myGREEN, myBLUE;

// Input a value 0 to 255 to get a color value.
// The colours are a transition r - g - b - back to r.
// From: https://gist.github.com/davidegironi/3144efdc6d67e5df55438cc3cba613c8
uint16_t colorWheel(uint8_t pos) {
  if(pos < 85) {
    return dma_display->color565(pos * 3, 255 - pos * 3, 0);
  } else if(pos < 170) {
    pos -= 85;
    return dma_display->color565(255 - pos * 3, 0, pos * 3);
  } else {
    pos -= 170;
    return dma_display->color565(0, pos * 3, 255 - pos * 3);
  }
}

void drawText(int colorWheelOffset)
{
  
  // draw text with a rotating colour
  dma_display->setTextSize(1);     // size 1 == 8 pixels high
  dma_display->setTextWrap(false); // Don't wrap at end of line - will do ourselves

  dma_display->setCursor(5, 0);    // start at top left, with 8 pixel of spacing
  uint8_t w = 0;
  const char *str = "ESP32 DMA";
  for (w=0; w<strlen(str); w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print(str[w]);
  }

  dma_display->println();
  dma_display->print(" ");
  for (w=9; w<18; w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print("*");
  }
  
  dma_display->println();

  dma_display->setTextColor(dma_display->color444(15,15,15));
  dma_display->println("LED MATRIX!");

  // print each letter with a fixed rainbow color
  dma_display->setTextColor(dma_display->color444(0,8,15));
  dma_display->print('3');
  dma_display->setTextColor(dma_display->color444(15,4,0));
  dma_display->print('2');
  dma_display->setTextColor(dma_display->color444(15,15,0));
  dma_display->print('x');
  dma_display->setTextColor(dma_display->color444(8,15,0));
  dma_display->print('6');
  dma_display->setTextColor(dma_display->color444(8,0,15));
  dma_display->print('4');

  // Jump a half character
  dma_display->setCursor(34, 24);
  dma_display->setTextColor(dma_display->color444(0,15,15));
  dma_display->print("*");
  dma_display->setTextColor(dma_display->color444(15,0,0));
  dma_display->print('R');
  dma_display->setTextColor(dma_display->color444(0,15,0));
  dma_display->print('G');
  dma_display->setTextColor(dma_display->color444(0,0,15));
  dma_display->print("B");
  dma_display->setTextColor(dma_display->color444(15,0,8));
  dma_display->println("*");

}

void setup() {

  // Module configuration
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // module width
    PANEL_RES_Y,   // module height
    PANEL_CHAIN    // Chain length
    ,_pins            // Uncomment to enable custom pins
  );

  //mxconfig.gpio.e = 18;
  //mxconfig.clkphase = false;
  //mxconfig.driver = HUB75_I2S_CFG::FM6126A;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(90); //0-255
  dma_display->clearScreen();

  myBLACK = dma_display->color565(0, 0, 0);
  myWHITE = dma_display->color565(255, 255, 255);
  myRED = dma_display->color565(255, 0, 0);
  myGREEN = dma_display->color565(0, 255, 0);
  myBLUE = dma_display->color565(0, 0, 255);
  

  dma_display->fillScreen(myWHITE);
  
  // fix the screen with green
  dma_display->fillRect(0, 0, dma_display->width(), dma_display->height(), dma_display->color444(0, 15, 0));
  delay(500);

  // draw a box in yellow
  dma_display->drawRect(0, 0, dma_display->width(), dma_display->height(), dma_display->color444(15, 15, 0));
  delay(500);

  // draw an 'X' in red
  dma_display->drawLine(0, 0, dma_display->width()-1, dma_display->height()-1, dma_display->color444(15, 0, 0));
  dma_display->drawLine(dma_display->width()-1, 0, 0, dma_display->height()-1, dma_display->color444(15, 0, 0));
  delay(500);

  // draw a blue circle
  dma_display->drawCircle(10, 10, 10, dma_display->color444(0, 0, 15));
  delay(500);

  // fill a violet circle
  dma_display->fillCircle(40, 21, 10, dma_display->color444(15, 0, 15));
  delay(500);

  // fill the screen with 'black'
  dma_display->fillScreen(dma_display->color444(0, 0, 0));

  //drawText(0);

}

uint8_t wheelval = 0;
void loop() {
    // animate by going through the colour wheel for the first two lines
    drawText(wheelval);
    wheelval +=1;
    delay(20); 
/*
  drawText(0);
  delay(2000);
  dma_display->clearScreen();
  dma_display->fillScreen(myBLACK);
  delay(2000);
  dma_display->fillScreen(myBLUE);
  delay(2000);
  dma_display->fillScreen(myRED);
  delay(2000);
  dma_display->fillScreen(myGREEN);
  delay(2000);
  dma_display->fillScreen(myWHITE);
  dma_display->clearScreen();
  */
  
}
`

and this is the result

https://github.com/user-attachments/assets/a2707fb8-81ce-4007-bdde-80354af66aa1

tinicoweb, 2025-04-18:
> Hi You don't need a Virtual panel code for 64x32 1/16 panel.

I have added the line

 mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

and now the result is

https://github.com/user-attachments/assets/11a8e336-af80-4386-b2ef-d0a020baa09d

tinicoweb, 2025-04-18:
help please

board707, 2025-04-18:
Please show a clear photo of the panel rear side.
And please look up a chips on the panel - in addition to the two chips you mentioned (SM16208SJ, MW245BC) , there should be one more type.

tinicoweb, 2025-04-18:
[image]

 SM16208SJ, MW245BC and SM5368PS

Thanks

board707, 2025-04-18:
Thanks.
According to the text on the panel, it has a 1/8 scan rather than 1/16s.
So you have to use a Virtual panel class, but with addition with option:
`mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;`

tinicoweb, 2025-04-18:
Could you help me with the virtual panel?
I'm not sure how to adjust it so the lines go in their correct place.

tinicoweb, 2025-04-18:
/*************************************************************************
Description:
The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
supports output to HALF scan panels - which means outputting
two lines at the same time, 16 or 32 rows apart if a 32px or 64px high panel
respectively.
This cannot be changed at the DMA layer as it would require a messy and complex
rebuild of the library's internals.
However, it is possible to connect QUARTER (i.e. FOUR lines updated in parallel)
scan panels to this same library and
'trick' the output to work correctly on these panels by way of adjusting the
pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
library.
*************************************************************************/
/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
correctly on a Four Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
public:
using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)
protected:
VirtualCoords getCoords(int16_t x, int16_t y); // custom getCoords() method for specific pixel mapping
};

// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
return coords;
}
if ((coords.y & 8) == 0)
{
coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
return coords;
}
/* ================================================== */
// DEFINICION DE LOS PINES PERSONALIZADOS
/* ================================================== */
#define R1_PIN 23
#define G1_PIN 22
#define B1_PIN 1
#define R2_PIN 3
#define G2_PIN 21
#define B2_PIN 19
#define A_PIN 18
#define B_PIN 5
#define C_PIN 17
#define D_PIN 16
#define E_PIN -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 32
#define OE_PIN 33
#define CLK_PIN 25
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS
// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel *FourScanPanel = nullptr;

/******************************************************************************
Setup!
******************************************************************************/
void setup()
{
delay(250);

Serial.begin(115200);
Serial.println(""); Serial.println(""); Serial.println("");
Serial.println("");
Serial.println("* 1/8 Scan Panel Demonstration *");
Serial.println("");

/* ================================================== /
// ACTIVA LOS PINES PERSONALIZADOS
/* ================================================== */
HUB75_I2S_CFG::i2s_pins _pins = {
R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
LAT_PIN, OE_PIN, CLK_PIN
};

HUB75_I2S_CFG mxconfig(
PANEL_RES_X * 2, // DO NOT CHANGE THIS
PANEL_RES_Y / 2, // DO NOT CHANGE THIS
NUM_ROWS * NUM_COLS // DO NOT CHANGE THIS
,_pins // Uncomment to enable custom pins
);

mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

//mxconfig.driver = HUB75_I2S_CFG::FM6126A; // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

// OK, now we can create our matrix object
dma_display = new MatrixPanel_I2S_DMA(mxconfig);

// let's adjust default brightness to about 75%
dma_display->setBrightness8(100); // range is 0-255, 0 - 0%, 255 - 100%

// Allocate memory and start DMA display
if ( not dma_display->begin() )
Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

dma_display->clearScreen();
delay(500);

// create FourScanPanellay object based on our newly created dma_display object
FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

// THE IMPORTANT BIT BELOW!
// FOUR_SCAN_16PX_HIGH
// FOUR_SCAN_32PX_HIGH
// **** You don't need to set PhysicalPanelScanRate when use your own virtual panel class
// FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

for (int i = 0; i < FourScanPanel->height(); i++)
{
for (int j = 0; j < FourScanPanel->width(); j++)
{
FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
delay(10);
}
}

delay(2000);
dma_display->clearScreen();

} // end loop

tinicoweb, 2025-04-18:
https://github.com/user-attachments/assets/ed63f726-ba8a-4f8a-8479-3b4dc125cbe3

board707, 2025-04-18:
Sorry, the mapping on your video looks incorrect.
Try this test for a single panel - change this to one:

> #define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW

tinicoweb, 2025-04-18:
> Sorry, the mapping on your video looks incorrect. Try this test for a single panel - change this to one:
> 
> > #define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW

I've already done it
the result is the same

tinicoweb, 2025-04-18:
I have swapped these two rows and now the result is

[image]

https://github.com/user-attachments/assets/474b7b1d-0165-483b-a350-26e372fc3758

board707, 2025-04-18:
Hi
As I see, the main problem that the pixels appears at two lines simultaneously. 
I suspect the issue is that the option `HUB75_I2S_CFG::DP3246_SM5368` is not fully compatible with your panel.
Unfortunately, it is very late in my local time and I can hardly help just now

tinicoweb, 2025-04-18:
Thanks
I would appreciate it if you could help me sometime later. It's a project for a gift for someone special to me.

tinicoweb, 2025-04-18:
I'm from Spain and it's late here too.

board707, 2025-04-19:
Hi
Could you change your VirtualMatrixPanel getCoords() method this way:
```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
    if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
        return coords;
    }
    uint8_t pxbase =32;   // pixel base
	if ((virt_y & 8) == 0)	
	{
		coords.x += (coords.x / pxbase) * pxbase;
	}
	else
	{
		coords.x += ((coords.x / pxbase) + 1) * pxbase;
	}
      coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
     return coords;
}
```

tinicoweb, 2025-04-19:
error
C:\Users\valen\Downloads\Kettlebell Counter instalacion\PRUEBA4\prueba10\prueba10.ino: In member function 'virtual VirtualCoords EightPxBasePanel::getCoords(int16_t, int16_t)':
C:\Users\valen\Downloads\Kettlebell Counter instalacion\PRUEBA4\prueba10\prueba10.ino:37:14: error: 'virt_y' was not declared in this scope
   37 |         if ((virt_y & 8) == 0)
      |              ^~~~~~
exit status 1

Compilation error: 'virt_y' was not declared in this scope

tinicoweb, 2025-04-19:
have I done something wrong?

board707, 2025-04-19:
Sorry. Please replace virt_y to coords.y

tinicoweb, 2025-04-19:
> Sorry. Please replace virt_y to coords.y

https://github.com/user-attachments/assets/d9c6b012-4d67-405c-b7c3-b206bb09bcae

board707, 2025-04-19:
Thank you. I will analyse it lately. If it is not a trouble for you, please rescan a video after  increasing interpixel delay from 20 to 50 or even 80

tinicoweb, 2025-04-19:
I have followed the instructions of

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/730

Getting exactly the same results
except
with pixbase =1
In case 730 it was positive but in my case it was not.

```
uint8_t pxbase =1;   // pixel base
if ((coords.y & 8) == 0)
{
coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
```

in my case this is the result

https://github.com/user-attachments/assets/6ee4c799-c83b-46b2-87df-4e398acb3560

board707, 2025-04-19:
As about the video, it's looks like a panel pixel base is 4 pixels.

Try this:
```
uint8_t pxbase =4;   // pixel base
if (((coords.y & 8) == 0) ^ ((coords.x/pxbase) % 2) )
{
coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
```

tinicoweb, 2025-04-19:
thanks
Indeed the result is as you say

https://github.com/user-attachments/assets/91861412-3b3f-4e1c-8356-4430d320aaed

What should I exchange?
Or how should I do it?
Sorry, but I'm very new to these languages.

board707, 2025-04-20:
```
uint8_t pxbase =4;   // pixel base
if (((coords.y & 8) == 0) ^ ((coords.x/pxbase) % 2) )
{
coords.x += (coords.x / pxbase) * pxbase;// 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += ((coords.x / pxbase) + 1) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
```

tinicoweb, 2025-04-20:
> ```
> uint8_t pxbase =4;   // pixel base
> if (((coords.y & 8) == 0) ^ ((coords.x/pxbase) % 2) )
> {
> coords.x += (coords.x / pxbase) * pxbase;// 1st, 3rd 'block' of 8 rows of pixels, offset by panel width [...]

Oh, wow, this code has no effect. That is, no pixels on the screen light up.

board707, 2025-04-20:
> this code has no effect

Sorry, but it can not be so. This code is the same as previous, except a little.
Please test your code and connections again.

tinicoweb, 2025-04-20:
The first code doesn't fill the entire screen. 

[image]

If you swap if and else, no LEDs light up.
I've checked and changed it several times in case I copied it incorrectly.

tinicoweb, 2025-04-20:
I'll leave you my email in case it's easier to contact me that way.
valentin.parra@gmail.com

board707, 2025-04-21:
Hi
Thanks for your email, I have remember it, so you can remove it from the message.
But the problem that I have nothing to say more about the issue. I will try to reproduce the case on my panels, but I haven't exact same matrix as yours.

tinicoweb, 2025-04-21:
Thank you.
I promised my wife and I would make some stopwatches like the ones I've attached for her youth sports club.
The first one I made worked perfectly, but when I ordered panels to make the rest, the chips changed, and the problems I've been telling you about arose.
Now the day of the competition is here, and I haven't been able to finish them.
I'll keep investigating, and thank you for your help.

[image]

board707, 2025-04-21:
> Now the day of the competition is here, and I haven't been able to finish them.

I am very sorry that it happened this way. I tried to apply my experience to solving your problem, but the panel turned out to be different from what I had worked with before. So I am learning your panel together with you.

tinicoweb, 2025-04-21:
I don't understand the logic, but only the first and second blocks are filled with pixels.
The third and fourth aren't.
What part of the logic controls those two blocks that don't light up?

board707, 2025-04-21:
The upper half of panel affected by R0 G0 B0 pins, while a lower one - by R1 G1 B1. 
All control pins like a LAT OE CLK are works in parallel to the both parts.

tinicoweb, 2025-04-21:
I don't know what the point is or why.
I copied a piece of code from the examples in ESP32-HUB75-VirtualMatrixPanel_T.hpp, and now the first and third blocks of pixel files are highlighted.

```
	if ((coords.y & 8) != ((coords.y & 16) >> 1)) {
	coords.y = (((coords.y & 0b11000) ^ 0b11000) + (coords.y & 0b11100111));
```

Also, if I change

`if (((coords.y & 8) == 0) ^ ((coords.x/pxbase) % 2) )`

by

`if (((coords.y & 8) %2) ^ ((coords.x/pxbase) % 2))`

 the first block of 8 rows to the top of the screen.

This would be the code fragment

```
uint8_t pxbase =4;   // pixel base

	if ((coords.y & 8) != ((coords.y & 16) >> 1)) {
	coords.y = (((coords.y & 0b11000) ^ 0b11000) + (coords.y & 0b11100111));
	}

if (((coords.y & 8) %2) ^ ((coords.x/pxbase) % 2))
{
coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
```

the video

https://github.com/user-attachments/assets/47e33e06-c53e-493a-ab66-d06bf0a968e7

and this image is with the same code I have added text to identify the blocks

[image]

board707, 2025-04-21:
What will be on the panel if you use the code above, removing this three lines (below) ?
```
if ((coords.y & 8) != ((coords.y & 16) >> 1)) {
	coords.y = (((coords.y & 0b11000) ^ 0b11000) + (coords.y & 0b11100111));
	}
```

tinicoweb, 2025-04-21:
the 1st block at the top of the screen

and the 2nd block where the 3rd should be

3 and 4 block do not light up

[image]

board707, 2025-04-21:
Could you please show a full code for the last picture?

tinicoweb, 2025-04-21:
```
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};
// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
    if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
        return coords;
    }
uint8_t pxbase =4;   // pixel base

				//if ((coords.y & 8) != ((coords.y & 16) >> 1)) {
				//	coords.y = (((coords.y & 0b11000) ^ 0b11000) + (coords.y & 0b11100111));
				//}

if (((coords.y & 8) %2) ^ ((coords.x/pxbase) % 2))
{
coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
}
/* ================================================== */
// DEFINICION DE LOS PINES PERSONALIZADOS
/* ================================================== */
#define R1_PIN 23
#define G1_PIN 22
#define B1_PIN 1
#define R2_PIN 3
#define G2_PIN 21
#define B2_PIN 19
#define A_PIN 18
#define B_PIN 5
#define C_PIN 17
#define D_PIN 16
#define E_PIN -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 32
#define OE_PIN 33
#define CLK_PIN 25
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_TOP_RIGHT_DOWN

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
EightPxBasePanel *FourScanPanel = nullptr;

// placeholder for the virtual display object INTENTANDO TRASPONER A CODIGO DE VULTURE
EightPxBasePanel *matrix = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/

// elimino esto a modo de prueba
void setup()
{
  delay(250);

  Serial.begin(115200);
  Serial.println(""); Serial.println(""); Serial.println("");
  Serial.println("*****************************************************");
  Serial.println("*         1/8 Scan Panel Demonstration              *");
  Serial.println("*****************************************************");

//hasta aqui

/* ================================================== /
// ACTIVA LOS PINES PERSONALIZADOS
/* ================================================== */
HUB75_I2S_CFG::i2s_pins _pins = {
R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN,
A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
LAT_PIN, OE_PIN, CLK_PIN
};

  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // NO CAMBIES ESTO
    PANEL_RES_Y / 2,            // NO CAMBIES ESTO
    NUM_ROWS * NUM_COLS         // NO CAMBIES ESTO
    ,_pins            // Descomentar para habilitar pines personalizados
  );

  mxconfig.clkphase = false; // Cambie esto si ve que los píxeles aparecen desplazados incorrectamente una columna hacia la izquierda o la derecha.

  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;     // En caso de que usemos paneles basados ​​en el chip FM6126A, podemos configurarlo aquí antes de crear el objeto MatrixPanel_I2S_DMA
  mxconfig.driver   = HUB75_I2S_CFG::DP3246_SM5368;
  //mxconfig.driver   = HUB75_I2S_CFG::ICN2038S;
  //mxconfig.driver   = HUB75_I2S_CFG::FM6124;
  //mxconfig.driver   = HUB75_I2S_CFG::FM6126A;
  //mxconfig.driver   = HUB75_I2S_CFG::MBI5124;
  //mxconfig.driver   = HUB75_I2S_CFG::SHIFTREG;

  // Ok, ahora podemos crear nuestra matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // Ajustemos el brillo predeterminado a aproximadamente 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);
  matrix = new EightPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

  // ¡LA PARTE IMPORTANTE A CONTINUACIÓN!!
  // FOUR_SCAN_16PX_HIGH
  // FOUR_SCAN_32PX_HIGH
// **** No es necesario configurar PhysicalPanelScanRate cuando usa su propia virtual panel class
// FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_32PX_HIGH);
}

// Test de mapeo de píxeles: rellena el panel píxel por píxel

void loop() {

  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(40, 0, 0));
      delay(10);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```

board707, 2025-04-21:
Thanks, I will test it and return

board707, 2025-04-21:
Hi @tinicoweb 
I think I found a bug. In your code, in the function getCoords(), the last line is missing:
`return coords;`

So, the full method should be:
```

inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
    if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
        return coords;
    }
uint8_t pxbase =4;   // pixel base

if (((coords.y & 8) %2) ^ ((coords.x/pxbase) % 2))
{
coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);

return coords;
}
```

The line was in place in some of your first sketches... what made me think that line remains in the code.

tinicoweb, 2025-04-21:
I'll be home from work in a few minutes and check it out and try it out.
Thanks.

tinicoweb, 2025-04-21:
Okay, yes, now yes.
I had to swap if and else, but it worked perfectly.
I don't know when I lost the code. "returns coords;"
Thanks a million.

```
// define custom getCoords() method for specific pixel mapping

inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
    if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
        return coords;
    }
uint8_t pxbase =4;   // pixel base

if (((coords.y & 8) == 0) ^ ((coords.x/pxbase) % 2) )
{
coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);

return coords;
}
```

https://github.com/user-attachments/assets/ab8e71c0-0f2c-4398-95c9-cf41fd53bbc7

tinicoweb, 2025-04-21:
I'm very grateful. 
Could I follow you on social media? 
Could I mention you in the modified stopwatch code?

board707, 2025-04-21:
My congratulations.
I'm very glad that everything worked.

> Could I mention you in the modified stopwatch code?

Yes, of course, if you want.

tinicoweb, 2025-04-21:
How do I do it?
I don't know your name or your social media accounts. I only have board707???

board707, 2025-04-21:
I am sorry for this.
I have use a local social networks, that have no point to specify on an international site. My name is Dmitry Dmitriev, you can mention me by link to my github.  Although it might be easier to specify nothing :)

Better come with other interesting tasks, I enjoyed solving this problem.

tinicoweb, 2025-04-21:
One more question:
If you look at the video, some ghost pixels appear.
Why could this be?
Is there a solution?

board707, 2025-04-22:
I think these artifacts are related to the update speed - the panel can't keep up with the signals generated by the library. Mister @mrcodetastic  should know better

tinicoweb, 2025-04-24:
Do you have any way to contact @mrcodetastic because he hasn't read this issue?
On the other hand, I have another problem and maybe you can help me.
I used to use Neocat's ESP32-P3RGB64x32MatrixPanel to run the panels, but it's not compatible with these chips.
When I adapted my code to @mrcodetastic's ESP32-HUB75-MatrixPanel-DMA, everything works fine except for two functions that the previous library had, but apparently aren't.
They are begin and stop, which, as far as I understand, reset the panel.
I'm attaching the code for the previous one in case you have any ideas on how to fix it.
Thanks.

P3RGB64x32MatrixPanel.h
```
#ifndef _ESP32_P3_RGB_64_32_MATRIX_PANEL
#define _ESP32_P3_RGB_64_32_MATRIX_PANEL

#include <vector>
#include <array>
#include "Adafruit_GFX.h"

class P3RGB64x32MatrixPanel : public Adafruit_GFX {
  public:
    P3RGB64x32MatrixPanel(uint8_t _pinR1, uint8_t _pinG1, uint8_t _pinB1, uint8_t _pinR2, uint8_t _pinG2, uint8_t _pinB2, uint8_t _pinCLK, uint8_t _pinLAT, uint8_t _pinOE, uint8_t _pinA, uint8_t _pinB, uint8_t _pinC, uint8_t _pinD, bool _doubleBuffer = false)
      : Adafruit_GFX(64, 32), pinR1(_pinR1), pinG1(_pinG1), pinB1(_pinB1), pinR2(_pinR2), pinG2(_pinG2), pinB2(_pinB2), pinCLK(_pinCLK), pinLAT(_pinLAT), pinOE(_pinOE), pinA(_pinA), pinB(_pinB), pinC(_pinC), pinD(_pinD), doubleBuffer(_doubleBuffer) {
      initMatrixBuff();
    }
    P3RGB64x32MatrixPanel(bool _doubleBuffer = false)
      : Adafruit_GFX(64, 32), doubleBuffer(_doubleBuffer) {
      initMatrixBuff();
    }
    void begin(void);
    void stop(void);
    virtual void drawPixel(int16_t x, int16_t y, uint16_t color);

    uint16_t color444(uint8_t r, uint8_t g, uint8_t b) { return ((r & 0xf) << 1) | ((uint16_t)(g & 0xf) << 6) | ((uint16_t)(b & 0xf) << 11); }
    uint16_t color555(uint8_t r, uint8_t g, uint8_t b) { return (r&0x1f) | ((uint16_t)(g & 0x1f) << 5) | ((uint16_t)(b & 0x1f) << 10); }
    uint16_t colorHSV(long hue, uint8_t sat, uint8_t val);

    void swapBuffer() {
      matrixbuff = drawBuffer();
    }

    uint16_t* matrixbuff;
    std::vector<std::array<uint16_t, 64*32>> _matrixbuff;

    void copyBuffer() {
      if (!doubleBuffer) return;
      if (matrixbuff == _matrixbuff[0].data())
        _matrixbuff[0] = _matrixbuff[1];
      else
        _matrixbuff[1] = _matrixbuff[0];
    }

  private:
    void initMatrixBuff() {
      _matrixbuff.resize(doubleBuffer ? 2 : 1);
      matrixbuff = _matrixbuff[0].data();
    }
    static void IRAM_ATTR onTimer(void);
    void IRAM_ATTR draw();

    uint16_t* drawBuffer() {
      if (!doubleBuffer) return _matrixbuff[0].data();
      if (matrixbuff == _matrixbuff[0].data())
        return _matrixbuff[1].data();
      else
        return _matrixbuff[0].data();
    }

    hw_timer_t* timer;

    uint8_t pinR1 = 25;
    uint8_t pinG1 = 26;
[... 139 lines cut ...]

    REG_WRITE(GPIO_OUT_REG, out |
       ((uint32_t)r1 << pinR1) |
       ((uint32_t)g1 << pinG1) |
       ((uint32_t)b1 << pinB1) |
       ((uint32_t)r2 << pinR2) |
       ((uint32_t)g2 << pinG2) |
       ((uint32_t)b2 << pinB2));

    REG_WRITE(GPIO_OUT_W1TS_REG, (1 << pinCLK));
  }
  //REG_WRITE(GPIO_OUT_W1TC_REG, (1 << pinCLK));

  REG_WRITE(GPIO_OUT1_W1TS_REG, (1 << (pinOE - 32)) | (1 << (pinLAT - 32)));

  out = ((y & 1) << pinA) | ((y & 2) << (pinB - 1)) |
        ((y & 4) << (pinC - 2)) | ((y & 8) << (pinD - 3));
  REG_WRITE(GPIO_OUT_REG, out);

  for (int x = 0; x < 8; x++) NOP(); // to wait latch and row switch

  REG_WRITE(GPIO_OUT1_W1TC_REG, (1 << (pinOE - 32)) | (1 << (pinLAT - 32)));
}

```

tinicoweb, 2025-04-24:
My code is a stopwatch and has several options. 
The timer option works fine, but it has another option that counts time intervals, for example, several one-minute intervals, from 60 seconds to zero. It resets and starts again, several times. 
And for that, it uses begin and stop. 
But since it's not in this new library, my code doesn't work.
 Do you have any way to do this? Thanks.

tinicoweb, 2025-04-24:
This is part of my stopwatch code in case it helps you understand it.
I didn't create it myself; I'm just adapting it to the new library, so I'm still unclear on some things.
Could you help me?
If you need the full code, I can send it to you without any problem.
It's on GitHub, so it's free to share.

https://github.com/VultureWood/KettlebellCounter?fbclid=IwAR3fIuEmN_KTVOgsuc_02m0UB0AoIeVsCd1bo3O1Ao-FusEV5bfYD-BwuCA

```
#include <Adafruit_GFX.h>   // Core graphics library
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h" // RGB display driver
#include <SPI.h> // spi for radio use
#include <nRF24L01.h> // radio libraries
#include <RF24.h> // radio libraries  // 
//#include <P3RGB64x32MatrixPanel.h> // esta es la vieja
#include <vector>
#include <array>

//////////////////////////////////////////////////////////
// DEFINICION DE LOS PINES PERSONALIZADOS
//////////////////////////////////////////////////////////
#define R1_PIN 23
#define G1_PIN 22
#define B1_PIN 1
#define R2_PIN 3
#define G2_PIN 21
#define B2_PIN 19
#define A_PIN 18
#define B_PIN 5
#define C_PIN 17
#define D_PIN 16
#define E_PIN -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 32
#define OE_PIN 33
#define CLK_PIN 25

//////////////////////////////////////////////////////////
// Definir una clase personalizada derivada de VirtualMatrixPanel ORIGINAL
//////////////////////////////////////////////////////////

class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

//////////////////////////////////////////////////////////
// Define un método getCoords() personalizado para el mapeo de píxeles específico
//////////////////////////////////////////////////////////
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {
coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach
    if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
        return coords;
    }
uint8_t pxbase =4;   // pixel base

if (((coords.y & 8) == 0) ^ ((coords.x/pxbase) % 2) )
{
coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
else
{
coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
}
coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
[... 768 lines cut ...]
    }
    if(CDT_menu == 3){
   CDT_mm_1 = CDT_var_val;
    }
    if(CDT_menu == 4){
   CDT_ss_10 = CDT_var_val;
    }
    if(CDT_menu == 5){
   CDT_ss_1 = CDT_var_val;
    }
          }
   }
}} // end else{}

timeHolder1 = GatherCDT(CDT_hh_1, CDT_mm_10, CDT_mm_1,CDT_ss_10,CDT_ss_1);

time_fractionCountdown(timeHolder1,14); // 14 is the line it's written on

lastButtonState = reading;
lastButtonStateRepUP = readingRepUP;
lastButtonStateRepDOWN = readingRepDOWN;
lastbuttonStateONEmenuUP = buttonStateONEmenuUP;
  } // end of while(setCountdownTimer)

```

tinicoweb, 2025-04-24:
P3RGB64x32MatrixPanel.h (by NeoCat) contains the following note:

Note that this library heavily uses the timer interrupt to drive LED matrix to enable software-based 5-bit PWM for each line. This may conflict with WiFi setup, so you should start WiFi before matrix.begin(), which sets up timer interrupts.

Have you tried simply commenting out these lines?
The timer has several options: a normal timer and an interval timer, that is, several time intervals in a row, for example, 2 min, 2 min, 2 min.
With the lines commented out, the normal timer works because it is not affected by begin and stop.
But the interval timer, when the first interval ends, does not "reset" for the next interval. I think this is because it cannot use the stop and begin methods.

These two methods are the only ones that differ between the two libraries, so that is the only reason I can find for the stopwatch code not working correctly.

board707, 2025-04-24:
> the stopwatch code not working correctly

Could you explain this " not working" in more detail?.

tinicoweb, 2025-04-25:
Good morning, I think the error was somewhere else in my code and has nothing to do with the begin and stop functions I had in the old library, which aren't there.

On the other hand, I can't get rid of the ghost pixels.
What could this be due to?
look at the second digit 0

[image]

mrcodetastic, 2025-04-29:
Looks to me like there's something wrong with the drawing code, are you sure you're "clearing" the framebuffer?