mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #782: Does anyone know if the LS9930 driver chip works with this library? (16-BIT CONSTANT CURRENT LED DRIVER WITH MEMORY)

ogul, 2025-04-22:
I couldn't add it to the library. Thanks to those who can help me.

ogul, 2025-04-22:
https://drive.google.com/file/d/1GU3zpRV5gegh9w06UDVSvxFbCutWbTjo/view?usp=sharing

board707, 2025-04-22:
> Does anyone know if the LS9930 driver chip works with this library? 

Definitely not.

> LS9930 pdf file is in this link.

The file is not a full datasheet. There are some important parts missing in the documents, such as a initialization protocol and configuration registers description.

ogul, 2025-04-22:
Thank you for your interest.
I will try to find the full file

mrcodetastic, 2025-04-22:
This library won't support PWM chips as they're totally different. 

I did write two different implementations for such a panel type in another repo of mine, but this isn't a "library" but a proof of concept.

board707, 2025-04-22:
@ogul 
Finding a datasheet is only a first step on the long route. The way the new drivers work is very different from the algorithm used in this library, so the code will need to be significantly rewritten. Even with a full datasheet, this is a task for an experienced programmer.

mrcodetastic, 2025-04-22:
Yep. Easier to just buy shift register based panel that work with the library. 

This library is for shift register based panels only.

ogul, 2025-04-22:
Thank you for your warning.
Then I will return this panel.

closed 2025-04-22 by ogul