mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #783: ESP32-S3-devkitc-1-n16r8v crashes on begin() call.

nate0202, 2025-04-23:
I have a panel (RGB-Matrix-P2.5-96x48-F) that has worked before with in a larger application that ran on the older "esp32dev" board. After switching to the ESP32-S3 to take advantage of larger memory I have not been able to get it to work. I created a minimal test to remove other possible issues. When this minimal program hits the begin() call it crashes and restarts.

I rewired the board to match the S3 default pins.

Any help you can provide would be appreciated. The test program and the console debug output are below. 

**Development Environment:**  PlatformIO

**platformio.ini**
```
;PlatformIO Project Configuration File
;
[env:esp32-s3-devkitc-1-n16r8v]
platform = espressif32
board = esp32-s3-devkitc-1-n16r8v
framework = arduino
monitor_speed = 115200
lib_deps = 
	mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display@^3.0.11
	adafruit/Adafruit GFX Library@^1.11.7
build_flags =
    -DCORE_DEBUG_LEVEL=5
    -DNO_GFX=1
```

**Panel type:** https://www.waveshare.com/wiki/RGB-Matrix-P2.5-96x48-F

**Test Program:** 
```
#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
// default S3 pins used 
#define PANEL_RES_X 96  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 48  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1   // Total number of panels chained one to another

MatrixPanel_I2S_DMA* dma_display = nullptr;

void setup() {
    Serial.begin(115200);
    delay(1000);
    Serial.println(">>>>> Starting up display test...");
    // HUB75 panel setup
    // Module configuration
    HUB75_I2S_CFG mxconfig(
        PANEL_RES_X,  // module width
        PANEL_RES_Y,  // module height
        PANEL_CHAIN  // Chain length
    ); 

    mxconfig.gpio.e = 32;
    mxconfig.clkphase = false;
    mxconfig.driver = HUB75_I2S_CFG::FM6124;

    // Display Setup
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
    Serial.println(">>>>> display created..., begin next");
    delay(1000);
    dma_display->begin();  // Crash here <<<<<<<<<<<<<<<<<<
    Serial.println(">>>>> begin complete...");
    delay(1000);
    dma_display->setBrightness8(50);  // 0-255
    dma_display->clearScreen();
}

void loop() {
    // put your main code here, to run repeatedly:
    dma_display->fillScreenRGB888(120, 0, 0);
    delay(2000);
    dma_display->fillScreenRGB888(0, 120, 0);
    delay(2000);
    dma_display->fillScreenRGB888(0, 0, 120);
    delay(2000);
    dma_display->fillScreenRGB888(0, 0, 0);
    delay(2000);
}

```

**Debug console output:** 

```
--- Terminal on COM10 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[    88][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[  2110][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:423] begin(): [begin()] Using GPIO 4 for R1_PIN
[  2118][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:424] begin(): [begin()] Using GPIO 5 for G1_PIN
[  2127][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:425] begin(): [begin()] Using GPIO 6 for B1_PIN
[  2135][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:426] begin(): [begin()] Using GPIO 7 for R2_PIN
[  2143][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:427] begin(): [begin()] Using GPIO 15 for G2_PIN
[  2152][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:428] begin(): [begin()] Using GPIO 16 for B2_PIN       
[  2160][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:429] begin(): [begin()] Using GPIO 18 for A_PIN
[  2168][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:430] begin(): [begin()] Using GPIO 8 for B_PIN
[  2177][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:431] begin(): [begin()] Using GPIO 3 for C_PIN
[  2185][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:432] begin(): [begin()] Using GPIO 42 for D_PIN        
[  2193][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:433] begin(): [begin()] Using GPIO 32 for E_PIN        
[  2202][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:434] begin(): [begin()] Using GPIO 40 for LAT_PIN
[  2210][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:435] begin(): [begin()] Using GPIO 2 for OE_PIN
[  2218][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:436] begin(): [begin()] Using GPIO 41 for CLK_PIN      
[  2227][I][ESP32-HUB75-MatrixPanel-leddrivers.cpp:50] fm6124init(): [LEDdrivers] MatrixPanel_I2S_DMA - initializing FM6124 driver...
[  2240][D][gdma_lcd_parallel16.cpp:317] allocate_dma_desc_memory(): [S3] Allocating 4608 bytes memory for DMA descriptors.
[  2255][I][gdma_lcd_parallel16.cpp:152] init(): [S3] Clock divider is 16
[  2261][D][gdma_lcd_parallel16.cpp:153] init(): [S3] Resulting output clock frequency: 10000000 Mhz
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)

```

Jason2866, 2025-04-23:
This lib is made for Arduino core 3.x. So use platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip
since official Platformio does not support Arduino core 3.x
```
;PlatformIO Project Configuration File
;
[env:esp32-s3-devkitc-1-n16r8v]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip
board = esp32-s3-devkitc-1-n16r8v
framework = arduino
monitor_speed = 115200
lib_deps = 
	mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display@^3.0.11
	adafruit/Adafruit GFX Library@^1.11.7
build_flags =
    -DCORE_DEBUG_LEVEL=5
    -DNO_GFX=1
```

srinn, 2025-04-24:
> This lib is made for Arduino core 3.x. So use platform = https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip since official Platformio does not [...]

In my case, a specific pin of esp32 was stuck at high. so I changed that pin to another, and it worked.
Check your esp32 pins.

nate0202, 2025-04-24:
I have updated my platform to the newer release and the problem still remains, crashing and restarting after the terminal debug message "_Resulting output clock frequency: 10000000 Mhz_".  I then checked the pins I was using, that was also suggested. I found that pin 2 was high, and moved the "OE_PIN" to 38 that wasn't high. This made no difference.

The platform update provided a lot of addition debug information about the chip, memory, board, software version, etc. so I am including the latest terminal log in case that is helpful...

I appreciate the time you have spent to help me resolve this. 

```
--- Terminal on COM11 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Dein>>>>> Starting up display test...
>>>>> display created..., begin next
[  5552][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:423] begin(): [begin()] Using GPIO 4 for R1_PIN
[  5552][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:424] begin(): [begin()] Using GPIO 5 for G1_PIN
[  5557][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:425] begin(): [begin()] Using GPIO 6 for B1_PIN
[  5565][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:426] begin(): [begin()] Using GPIO 7 for R2_PIN
[  5574][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:427] begin(): [begin()] Using GPIO 15 for G2_PIN
[  5582][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:428] begin(): [begin()] Using GPIO 16 for B2_PIN
[  5590][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:429] begin(): [begin()] Using GPIO 18 for A_PIN
[  5598][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:430] begin(): [begin()] Using GPIO 8 for B_PIN
[  5607][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:431] begin(): [begin()] Using GPIO 3 for C_PIN
[  5615][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:432] begin(): [begin()] Using GPIO 42 for D_PIN
[  5623][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:433] begin(): [begin()] Using GPIO 32 for E_PIN
[  5631][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:434] begin(): [begin()] Using GPIO 40 for LAT_PIN
[  5640][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:435] begin(): [begin()] Using GPIO 38 for OE_PIN
[  5648][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:436] begin(): [begin()] Using GPIO 41 for CLK_PIN
[  5656][I][ESP32-HUB75-MatrixPanel-leddrivers.cpp:50] fm6124init(): [LEDdrivers] MatrixPanel_I2S_DMA - initializing FM6124 driver...
[  5669][D][gdma_lcd_parallel16.cpp:317] allocate_dma_desc_memory(): [S3] Allocating 4608 bytes memory for DMA descriptors.
[  5682][I][gdma_lcd_parallel16.cpp:152] init(): [S3] Clock divider is 16
[  5686][D][gdma_lcd_parallel16.cpp:153] init(): [S3] Resulting output clock frequency: 10000000 Mhz
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403743c0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
[     0][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42006aa4
[    12][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42006a70
[    23][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42006a3c
[    34][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006a08
[    46][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42006aa4
[    57][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42006a70
[    68][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42006a3c
[    80][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006a08
[    91][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x42006aa4
[   102][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x42006a70
[   114][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x42006a3c
[   125][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x42006a08
[   136][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DM (45) successfully set to 0x42004048
[   148][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type USB_DP (46) successfully set to 0x42004048
[   173][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3fc9551c
[   184][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc9551c
[   194][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 19 successfully set to type USB_DM (45) with bus 0x3fc98c64
[   205][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 20 successfully set to type USB_DP (46) with bus 0x3fc98c64
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
[... cut ...]
[  3656][I][ESP32-HUB75-MatrixPanel-leddrivers.cpp:50] fm6124init(): [LEDdrivers] MatrixPanel_I2S_DMA - initializing FM6124 driver...
[  3669][D][gdma_lcd_parallel16.cpp:317] allocate_dma_desc_memory(): [S3] Allocating 4608 bytes memory for DMA descriptors.
[  3682][I][gdma_lcd_parallel16.cpp:152] init(): [S3] Clock divider is 16
[  3686][D][gdma_lcd_parallel16.cpp:153] init(): [S3] Resulting output clock frequency: 10000000 Mhz
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
```

nate0202, 2025-04-25:
The crashing was due to the 3 lines that I had copy/pasted from some other source into my simple test:
```
      mxconfig.gpio.e = 32;
      mxconfig.clkphase = false;
      mxconfig.driver = HUB75_I2S_CFG::FM6124;
```

The **"mxconfig.gpio.e = 32;"** was a pin assignment for the E pin and should have been -1. I have switched to defining the pins explicitly and then passing them in the 4th argument to mxconfig definition. 

With those lines removed the sample program no longer crashes, but the screen remains blank. So a new and different problem.

closed 2025-04-25 by nate0202