mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #788: Font Change Issue in Scroll Text Example

hfarcan, 2025-04-30:
Hello, thank you again for providing these wonderful libraries. 
I am experiencing an issue with changing the font in the Scroll Text example. I am using the FOUR_SCAN_40_80PX_HFARCAN from the Panel Scan Types of the Virtual Matrix. 
When I set a new font using gfx_layer_fg.setFont(...), I am unable to display anything on the screen. When the font is null, the text shows up fine.
Could it be that changing the font is currently not supported, or am I missing something?

mrcodetastic, 2025-04-30:
Have you read the AdafruitGFX guidance on how to use fonts? You need to load the font as well.

mrcodetastic, 2025-04-30:
Thank for your kind words. Have you read the AdafruitGFX guidance on how to use fonts? You need to load the font as well.

hfarcan, 2025-04-30:
> You need to load the font as well.

Yes, I can use my custom fonts in my normal application. When I set the font with a GFX_Lite layers, I cannot get an image with custom font.

mrcodetastic, 2025-05-02:
Could you please share example code?