mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #790: Only half of JPY-32X64P4B/P4-63x32-32S panel works

CheekiBreeki95, 2025-05-04:
Only half of my panel works, came with standard Chinese documentation (none) but the markings on the panel casing indicates its 1/32 scan, based off of the 32S. Attached is a video of the screen going through tests, a video of the screen displaying one of the test sample codes, and a picture of the back. I've checked over my wiring and tested continuity between the ESP and all the pins on the HUB75E connector with them all lining up as they should. I dont think this is a problem with the library but I can't think of anywhere else to ask, thanks in advance!

[Back of board](https://streamable.com/d8swq3)
[My code](https://streamable.com/fgqb91)
[Test code](https://streamable.com/lgna2s)

```
#include <WiFi.h>
#include <time.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <ESPAsyncWebServer.h>
#include <AsyncTCP.h>
#include <ArduinoJson.h>
#include <SPIFFS.h>

// WiFi credentials
const char* ssid = "SKYYX6V7";
const char* password = "Hzit7bNLscRy";

// NTP servers
const char* ntpServers[] = {
  "192.168.0.225",    // Your local NTP server
  "pool.ntp.org",
  "time.nist.gov",
  "time.google.com"
};

// Timezone offsets for BST (GMT+1)
const long gmtOffset_sec = 3600;
const int daylightOffset_sec = 0;

// LED Matrix Panel config
#define PANEL_WIDTH 64
#define PANEL_HEIGHT 32
#define PANEL_CHAIN_LENGTH 1

// Pin configuration for 1/32 scan panel
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 23
#define B_PIN 19
#define C_PIN 5
#define D_PIN 17
#define E_PIN 18
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 16

HUB75_I2S_CFG::i2s_pins _pins = {
  R1_PIN, G1_PIN, B1_PIN,
  R2_PIN, G2_PIN, B2_PIN,
  A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
  LAT_PIN, OE_PIN, CLK_PIN
};

HUB75_I2S_CFG mxconfig(
  PANEL_WIDTH, 
  PANEL_HEIGHT, 
  PANEL_CHAIN_LENGTH, 
  _pins
);

MatrixPanel_I2S_DMA* dma_display = nullptr;

// Web server
AsyncWebServer server(80);
[... 279 lines cut ...]
  // Display date
  dma_display->setCursor(2, 20);
  dma_display->print(dateStr);

  // Display scrolling text if any
  if (scrollingText.length() > 0) {
    if (millis() - lastScrollUpdate > 100) {
      dma_display->setCursor(scrollPosition, 30);
      dma_display->print(scrollingText);
      
      scrollPosition--;
      if (scrollPosition < -scrollingText.length() * 6) {
        scrollPosition = 

> PANEL_WIDTH;

      }
      
      lastScrollUpdate = millis();
    }
  }

  delay(100);
} 
```

srinn, 2025-05-05:
First, The problem is 64x32 panel with 1/32 scan rate itself. if the scan rate is same as pixel height, the panel isn't use RGB2 lines. it's only use RGB1 lines.
Normally, hub75 uses R1, G1, B1 lines for upper half lines of panel height, and R2, G2, B2 lines for lower half lines.

You can avoid this problem by
- set the height to 64, not 32
- or change 67th line of ESP32-HUB75-MatrixPanel-I2S-DMA.h to #define MATRIX_ROWS_IN_PARALLEL 1

CheekiBreeki95, 2025-05-05:
setting the height to 64 worked perfectly!!! thank you so much!

closed 2025-05-05 by CheekiBreeki95

board707, 2025-05-05:
See also #540 and #727

srinn, 2025-05-05:
> > the panel isn't use RGB2 lines. it's only use RGB1 lines.
> 
> Perhaps It is not a quite correct. Such panels uses a RGB1 lines for leds on the current panel, and a RGB2 for a chaining.

I've just tested it. RGB2 lines are disconnected, and it works fine. plus, chaining works fine too.

[image]
[image]

board707, 2025-05-05:
> chaining works fine too.

It's hard to tell from your picture whether the chaining is working or not :)
However, if everything suits you, I won't interfere.

srinn, 2025-05-05:
[image]

board707, 2025-05-05:
Great! 
You can offer your option to @mjmokhtar, I was unable to solve his problem in #540.

srinn, 2025-05-05:
> Great! You can offer your option to [@mjmokhtar](https://github.com/mjmokhtar), I was unable to solve his problem in [#540](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/540).

According to this post I think his screen is already working:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/540#issuecomment-2176233302

CheekiBreeki95, 2025-05-05:
> > the panel isn't use RGB2 lines. it's only use RGB1 lines.
> 
> Perhaps It is not a quite correct. Such panels uses a RGB1 lines for leds on the current panel, and a RGB2 for a chaining.
> 
> > setting the height to 64 worked perfectly
> 
> Do you have a multiple panels? Chain two or three of them and test.

I only have the one panel, I'm fairly sure disconnecting RGB2 had no effect for me either