mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #798: Adding ESP32-HUB75-MatrixPanel-DMA to ESP-IDF registry

patkepa, 2025-06-02:
Hello,

Are there any plans for including this library, directly into IDF registry? Currently it's kinda hacky to add it (but works!), since you need to download entire arduino-esp32 repo, which is huge, and makes it difficult to introduce when doing automated builds.

If there's interest, but no hands to do that, let me know!

abrender, 2025-09-25:
For some context, a simple `git clone` of the arduino-esp32 library involves downloading *2GB* of data. That 2GB is basically all in the `.git` folder, containing the commit histories and such. That is painful for automation and annoying for humans.  If only the most recent branch is cloned, then it's just 96MB.

```$ git clone git@github.com:espressif/arduino-esp32.git
Cloning into 'arduino-esp32'...
remote: Enumerating objects: 71746, done.
remote: Counting objects: 100% (581/581), done.
remote: Compressing objects: 100% (247/247), done.
remote: Total 71746 (delta 497), reused 334 (delta 334), pack-reused 71165 (from 5)
Receiving objects: 100% (71746/71746), 2.01 GiB | 15.99 MiB/s, done.
Resolving deltas: 100% (44992/44992), done.

$ du -s -h arduino-esp32/
2.1G	arduino-esp32/
$ du -s -h arduino-esp32/.git
2.1G	arduino-esp32/.git
```

vs

```
$ git clone --depth 1 git@github.com:espressif/arduino-esp32.git
Cloning into 'arduino-esp32'...
remote: Enumerating objects: 2664, done.
remote: Counting objects: 100% (2664/2664), done.
remote: Compressing objects: 100% (1799/1799), done.
remote: Total 2664 (delta 693), reused 1706 (delta 498), pack-reused 0 (from 0)
Receiving objects: 100% (2664/2664), 24.80 MiB | 8.59 MiB/s, done.
Resolving deltas: 100% (693/693), done.

$ du -s -h arduino-esp32/
96M	arduino-esp32/
$ du -s -h arduino-esp32/.git
26M	arduino-esp32/.git
```
I'll send a PR to update  [examples/esp-idf/with-gfx/README.md](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/bb1e6146f8c197d211b48946798e83a073f9afc1/examples/esp-idf/with-gfx/README.md) to use `git clone --depth 1` to help people avoid this pain.

aslak3, 2025-09-25:
FWIW I've found the ESP32S3 fast enough to bitbang a 64x32 x 12bpp display on its second core. No flickering. All the network activity is on the first core. I appreciate this wont work on all ESP32s which have a DMA engine, but it's fine for my ESP32S3 project running on an Adafruit MatrixPortal S3.

Arduino. I try to avoid it like the plague, being honest.

abrender, 2025-09-25:
Doing a little research on this, it seems like it would be great to have this library available in the [ESP Component Registry](https://components.espressif.com/).

Some resources:
* The registry: https://components.espressif.com/
* GitHub action to upload ESP-IDF component to the registry: https://github.com/espressif/upload-components-ci-action
* Documentation: https://docs.espressif.com/projects/idf-component-manager/en/latest/

mrcodetastic, 2025-10-25:
What changes would be required to the actual library to support this? With the NO_GFX option it should be bare metal in most cases to use as a library with the esp32-idf. If there are hidden dependencies with Adafruit / Arduino then I can look at how these can be unwound.

Jason2866, 2025-10-25:
to use the lib with espressif component manager it is not a must to upload to espressif registry. The espressif component manager can use a github repo branch (or commit or tag) as source. Since a library.json is already there the repo is prepared to be used with Platformio / pioarduino and espressif component  manager
For espressif component manager it would be nice to have a idf_component.yml in the repo. https://docs.espressif.com/projects/idf-component-manager/en/latest/reference/config_file.html

Jason2866, 2025-10-25:
> If there are hidden dependencies with Adafruit / Arduino then I can look at how these can be unwound.

For use with IDF there are no dependencies to Arduino libs allowed/possible.