mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #801: 32x16 LED panel : troubles to drawPixel at 0,0

PhilBauch, 2025-06-11:


PhilBauch, 2025-06-11:
The modified code is the following:
`
/*************************************************************************
   Description:

   The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
   supports output to HALF scan panels - which means outputting
   two lines at the same time, 16 or 32 rows apart if a 32px or 64px high panel
   respectively.
   This cannot be changed at the DMA layer as it would require a messy and complex
   rebuild of the library's internals.

   However, it is possible to connect QUARTER (i.e. FOUR lines updated in parallel)
   scan panels to this same library and
   'trick' the output to work correctly on these panels by way of adjusting the
   pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
   library.

 **************************************************************************/

/* Use a custom Virtual Display class to re-map co-ordinates such that they draw
   correctly on a 32x16 1/4 Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

/* ================================================== */
// Define custom class derived from VirtualMatrixPanel
class EightPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// define custom getCoords() method for specific pixel mapping
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

  uint8_t pxbase = 8;  // pixel base
  if ((coords.y & 4) == 0)
  {
    coords.x = (coords.x / pxbase) * 2 * pxbase  + pxbase  + 7 - (coords.x & 0x7); // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
  }
  else
  {
    coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
  }

  coords.y = (coords.y >> 3) * 4 + (coords.y & 0b00000011);

  return coords;
}
/* ================================================== */
// Panel configuration
#define PANEL_RES_X 32 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 16 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// ^^^ NOTE: DEFAULT EXAMPLE SETUP IS FOR A CHAIN OF TWO x 1/8 SCAN PANELS
[... 78 lines cut ...]
  //FourScanPanel->setPhysicalPanelScanRate(FOUR_SCAN_16PX_HIGH);
}

// Test the pixel mapping - fill the panel pixel by pixel
void loop() {

  int i = 0;
  int j = 0;
  //for (int i = 0; i < FourScanPanel->height(); i++)
  {
    //for (int j = 0; j < FourScanPanel->width(); j++)
    {
        Serial.printf("j = %i , i = %i , GPIO.CLK = %i , GPIO.D = %i , GPIO.E = %i \n", j, i
            , dma_display->getCfg().gpio.clk, dma_display->getCfg().gpio.d, dma_display->getCfg().gpio.e);

      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 255, 0));
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
`

tolik2025, 2025-07-02:
I found in the ICND2018 datasheet that they use the pins SDIN, DCK, and RCK instead of A, B, C.
When I swapped the definitions of pins A and C, the LED panel went completely dark.

[image]

[ICND2018_datasheet_EN_2021_V1.6.pdf](https://github.com/user-attachments/files/21009771/ICND2018_datasheet_EN_2021_V1.6.pdf)
[ICND2018_Programming Instruction Manual_CN_2017_V1.1 (1).pdf](https://github.com/user-attachments/files/21009770/ICND2018_Programming.Instruction.Manual_CN_2017_V1.1.1.pdf)

mrcodetastic, 2025-07-03:
> > I'd be interested to look at the code if so.
> 
> Hi @mrcodetastic 
> The code is rather simple. 
> Let's take a 4S matrix as an example.
> In this case, each of the four lines is controlled by its own pin from ABCD set, so you give HIGH to the pin you want to turn on and LOW to all the others.
> Sometimes you come across panels that are controlled by a LOW level, otherwise the principle is the same.

Right. So I just need to modify the library to invert the address lines?

board707, 2025-07-03:
I may be wrong, but you don't have such a method of switching lines yet.

mrcodetastic, 2025-07-05:
@PhilBauch - Let me look into this one for you. Should be relatively easy to implement.

PhilBauch, 2025-07-07:
@mrcodetastic - Thank you in advance for at least trying to implement the solution. 
This difficulty in controlling drawing a single pixel was not a good sign for me, as the final goal for my project is to display characters using relevant fonts for the panel size.
I will read your feedback with great interest.