mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #803: error: 'gpio_hal_iomux_func_sel'

jhohannfabian, 2025-06-18:
Hi, Im going to test this library, but I have this  error: 'gpio_hal_iomux_func_sel' was not declared in this scope; did you mean 'gpio_hal_func_sel'
Im compile Simpletestshape.ino, the example of the library.
I use ESP32 S3, ESP32 V3.3.0 .
Thank you

Bitimage, 2025-07-25:
Same error for me.
I have only compiled UART without errors.

Windows 11
arduino IDE V2.2.1
Elecro 5" display
ESP Display-5.0 inch
(Advanced series)
SKU:DIS02050A

Bitimage, 2025-07-25:
Sorry , now trying to compile Draw in tutorial 2
using ESP32S3 Dev Module 
Huge APP
PSRAM OPI

mrcodetastic, 2025-08-01:
Should be fixed in the git repository version. Let me know if you have issues.

closed 2025-08-01 by mrcodetastic

jhohannfabian, 2025-08-01:
Hi, Im compile: 
El Sketch usa 340251 bytes (25%) del espacio de almacenamiento de programa. El máximo es 1310720 bytes.
Las variables Globales usan 22480 bytes (6%) de la memoria dinámica, dejando 305200 bytes para las variables locales. El máximo es 327680 bytes. 
No problem, thats good. Im goint to testing in a matrix, and I going to tell you. Thank You.

commit 36edef55 (2025-08-01): Update gdma_lcd_parallel16.cpp

Fix #803