mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #804: Stretched display on the DP32020A and DP5135

yasirr33, 2025-06-22:
I got a Qiangli P3 module, 64x64px with DP5135 and DP32020A, It is a similar panel to issue #645 . The panel works when I set the following set it up like this :

```
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1  // Total number of panels chained one to another

HUB75_I2S_CFG mxconfig(
                PANEL_RES_X,              // DO NOT CHANGE THIS
                PANEL_RES_Y,              // DO NOT CHANGE THIS
                NUM_ROWS*NUM_COLS           // DO NOT CHANGE THIS
                //,_pins            // Uncomment to enable custom pins
    );
    mxconfig.clkphase = false;     
    mxconfig.driver   = HUB75_I2S_CFG::DP3246_SM5368; 
```

BUT the output comes out stretched along the X-axis. 

This is what it displays

[image]

and this is what the original font is supposed to look like 

[image]

These are the Pictures of the Panel

[image]

[image]

[image]

[image]

It basically regards every 2 pixels as 1 along the x-axis only. I want to chain 3 such panels at max, it used to work flawlessly with ICN2037 based panels but the vendor doesn't sell those anymore.

It's really important for my project, please help @board707

board707, 2025-06-23:
Hello
It's looks more like as programming error rather than incorrect hardware configuration. The number of LED driver chips on the panel corresponds to 64pix wide, not the 32pix. The number of line drivers DP32020 also consistent to defined (64x64 32S) panel mode.

yasirr33, 2025-06-23:
I have another P3 panel with ICN2012 and ICN2037BP chips. And the output of the same code after changing just this

` mxconfig.driver   = HUB75_I2S_CFG::ICN2038S;  `

and the cursor a bit

The output is

[image]

I'm pretty sure it's not related to the code for displaying stuff and the fonts etc. I have two of these panels with DP32020A, and both show the same behavior.

board707, 2025-06-23:
@yasirr33 
Could you prepare a test code that contains nothing but the output of the "45" on the panel and make sure that it prints the stretched text?
Then post this code here in full.

yasirr33, 2025-06-26:
Sorry for the delay, rough patch!

I've prepared the minimal code required to display the 45 on the panel. Check the output, I'll also send the output of the same code for the ICN based panel for comparison. 

 `#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Picopixel2.h>

#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1  // Total number of panels chained one to another

// MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup()
{

  // Module configuration
  HUB75_I2S_CFG mxconfig(
      PANEL_RES_X, // module width
      PANEL_RES_Y, // module height
      PANEL_CHAIN  // Chain length
  );

  // Uncomment these for ICN Panels
  // mxconfig.gpio.e = 1;
  // mxconfig.driver = HUB75_I2S_CFG::ICN2038S;

  mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(90);
  // dma_display->clearScreen();
  dma_display->setFont(&Picopixel2);
  dma_display->setTextSize(1);
  dma_display->setTextColor(dma_display->color444(255, 255, 255));

  dma_display->setCursor(32, 10);
  dma_display->print(45);
}

void loop()
{
}`

Output on DP32020A and DP5135 based panel , Cursor (32,10)

[image]

now uncommenting the lines and switching the panels

`  // mxconfig.gpio.e = 1;
  // mxconfig.driver = HUB75_I2S_CFG::ICN2038S;`

Output on ICN based panel; Cursor (32,10)

[image]}

board707, 2025-06-26:
Thank you
I have nothing to say, sorry.

...It is interesting that cursor position on the first matrix looks like (0,10) rather than 32,10...