mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #806: P4 64x32 problem

kalyan-m25, 2025-07-01:
[image]
Help

kalyan-m25, 2025-07-01:
A set of chips

[image]

kalyan-m25, 2025-07-02:
Thank you for your response! Of course, I checked the connection several times before asking for help. The power supply is 5V 6A. I don't understand what could be causing the issue. The image appears both at the top and bottom of the screen.

board707, 2025-07-02:
> I checked the connection several times

I just noticed that your photo only has five wires. 

[image]

Where did you get that wiring diagram from?

kalyan-m25, 2025-07-02:
The connection looks like this. The lines are not used

[image]

board707, 2025-07-02:
> I have another P4 80x40 matrix

What is the scan factor?

> with different chips.

You script is not so clear as your photos :) What is the first chip name - TC7559 ?

kalyan-m25, 2025-07-02:
I've put the P4 64x32 panel aside for now, and I think the problem is with the panel. I'm trying to connect another P4 80x40 panel. I've changed the size and driver type in the code, and here's what I see.

[image]

kalyan-m25, 2025-07-02:
Could it be related to the scanning mode?

kalyan-m25, 2025-07-03:
Yes, of course, this is a short example from the library.

`#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32    // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another
 
//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t myBLACK, myWHITE, myRED, myGREEN, myBLUE;
uint16_t colorWheel(uint8_t pos) {
  if(pos < 85) {
    return dma_display->color565(pos * 3, 255 - pos * 3, 0);
  } else if(pos < 170) {
    pos -= 85;
    return dma_display->color565(255 - pos * 3, 0, pos * 3);
  } else {
    pos -= 170;
    return dma_display->color565(0, pos * 3, 255 - pos * 3);
  }
}

void drawText(int colorWheelOffset)
{
  
  // draw text with a rotating colour
  dma_display->setTextSize(1);     // size 1 == 8 pixels high
  dma_display->setTextWrap(false); // Don't wrap at end of line - will do ourselves
  dma_display->setCursor(5, 0);    // start at top left, with 8 pixel of spacing
  uint8_t w = 0;
  const char *str = "ESP32 DMA";
  for (w=0; w<strlen(str); w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print(str[w]);
  }
 dma_display->println();
  dma_display->print(" ");
  for (w=9; w<18; w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print("*");
  }
  
  dma_display->println();
  dma_display->setTextColor(dma_display->color444(15,15,15));
  dma_display->println("LED MATRIX!");
  // print each letter with a fixed rainbow color
  dma_display->setTextColor(dma_display->color444(0,8,15));
  dma_display->print('3');
  dma_display->setTextColor(dma_display->color444(15,4,0));
  dma_display->print('2');
  dma_display->setTextColor(dma_display->color444(15,15,0));
  dma_display->print('x');
  dma_display->setTextColor(dma_display->color444(8,15,0));
  dma_display->print('6');
  dma_display->setTextColor(dma_display->color444(8,0,15));
  dma_display->print('4');
  // Jump a half character
  dma_display->setCursor(34, 24);
  dma_display->setTextColor(dma_display->color444(0,15,15));
  dma_display->print("*");
  dma_display->setTextColor(dma_display->color444(15,0,0));
  dma_display->print('R');
  dma_display->setTextColor(dma_display->color444(0,15,0));
  dma_display->print('G');
  dma_display->setTextColor(dma_display->color444(0,0,15));
  dma_display->print("B");
  dma_display->setTextColor(dma_display->color444(15,0,8));
  dma_display->println("*");
}
void setup() {
   // Module configuration
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // module width
    PANEL_RES_Y,   // module height
    PANEL_CHAIN    // Chain length
  );
  //mxconfig.gpio.e = 18;
  //mxconfig.clkphase = false;
  mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(90); //0-255
  dma_display->clearScreen();
  myBLACK = dma_display->color565(0, 0, 0);
  myWHITE = dma_display->color565(255, 255, 255);
  myRED = dma_display->color565(255, 0, 0);
  myGREEN = dma_display->color565(0, 255, 0);
  myBLUE = dma_display->color565(0, 0, 255);
  
  drawText(0);
}

void loop() {

}`

board707, 2025-07-03:
> #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
> #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

???
Isn't this a 80x40 panel?

kalyan-m25, 2025-07-03:
it looks like my problem is the horizontal size. If I can solve this issue with your help, everything will be fine.

board707, 2025-07-03:
You must use real panel size for test.

Did you try just to print a single word?
Or fill the panel with pixel by pixel?
For example:

```
void loop() {

 for (int i = 0; i < dma_display->height(); i++)
  {
    for (int j = 0; j < dma_display->width(); j++)
    {
      dma_display->drawPixel(j, i, dma_display->color565(255, 0, 0));
      delay(30);
    }
  }

  delay(2000);
  dma_display->clearScreen();

} // end loop
```

kalyan-m25, 2025-07-04:
Please look at my result
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
class CustomPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)
   protected:
  VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping
};
inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y) {
coords = VirtualMatrixPanel::getCoords(x, y); // call base class method to update coords for chaining approach
if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }
uint8_t pxbase = panelResX; 

  if (panelResY == 32) {//???????????
    if ((coords.y & 8) == 0){ coords.x += ((coords.x / pxbase) + 1) * pxbase;}
    else {  coords.x += (coords.x / pxbase) * pxbase;    }
  coords.y = (coords.y >> 4) * 8 + (coords.y & 0b00000111);
  }

  else if (panelResY == 16) {//??????????????????
    if ((coords.y & 4) == 0){coords.x += ((coords.x / pxbase) + 1) * pxbase; }
    else {coords.x += (coords.x / pxbase) * pxbase;}
 coords.y = (coords.y >> 3) * 4 + (coords.y & 0b00000011);
  }
  return coords;
}

#define PANEL_RES_X 80 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 40 // Number of pixels tall of each INDIVIDUAL panel module.
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_RIGHT_UP
MatrixPanel_I2S_DMA *dma_display = nullptr;
CustomPxBasePanel   *FourScanPanel = nullptr;
void setup()
{
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // DO NOT CHANGE THIS
    PANEL_RES_Y / 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    );
 mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
 dma_display = new MatrixPanel_I2S_DMA(mxconfig);
 dma_display->setBrightness8(2);    // range is 0-255, 0 - 0%, 255 - 100%
 mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;
 if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
 dma_display->clearScreen();
  delay(500);
 FourScanPanel = new CustomPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,  VIRTUAL_MATRIX_CHAIN_TYPE);

}
void loop() {
  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 0, 0));
      delay(10);
    }
  }
  delay(1000);
  dma_display->clearScreen();
}

kalyan-m25, 2025-07-04:
Video

kalyan-m25, 2025-07-04:
https://github.com/user-attachments/assets/75ca4459-76fd-413b-a073-dac185822d2d

kalyan-m25, 2025-07-05:
it looks like the D pin on the panel is not being used. It looks like it's an ABC panel

kalyan-m25, 2025-07-28:
Can we continue setting up this panel?

board707, 2025-07-28:
Please test with code below:

```
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
class CustomPxBasePanel : public VirtualMatrixPanel
{
public:
using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)
protected:
VirtualCoords getCoords(int16_t x, int16_t y); // custom getCoords() method for specific pixel mapping
};

inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y) {
coords = VirtualMatrixPanel::getCoords(x, y); // call base class method to update coords for chaining approach
if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
return coords;
}
if (((coords.y / 10)%2) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

       coords.y = (coords.y / 20) * 10 + (coords.y % 10);

return coords;
}

#define PANEL_RES_X 80 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 40 // Number of pixels tall of each INDIVIDUAL panel module.
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_RIGHT_UP
MatrixPanel_I2S_DMA dma_display = nullptr;
CustomPxBasePanel FourScanPanel = nullptr;
void setup()
{
HUB75_I2S_CFG mxconfig(
PANEL_RES_X * 2, // DO NOT CHANGE THIS
PANEL_RES_Y / 2, // DO NOT CHANGE THIS
NUM_ROWS * NUM_COLS // DO NOT CHANGE THIS
);
mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.
dma_display = new MatrixPanel_I2S_DMA(mxconfig);
dma_display->setBrightness8(80); // range is 0-255, 0 - 0%, 255 - 100%
mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;
if ( not dma_display->begin() )
Serial.println("**** !KABOOM! I2S memory allocation failed ***********");
dma_display->clearScreen();
delay(500);
FourScanPanel = new CustomPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, VIRTUAL_MATRIX_CHAIN_TYPE);

}
void loop() {
for (int i = 0; i < FourScanPanel->height(); i++)
{
for (int j = 0; j < FourScanPanel->width(); j++)
{
FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 0, 0));
delay(10);
}
}
delay(1000);
dma_display->clearScreen();
}
```

board707, 2025-07-28:
Could you increase a inter-pixel delay up to 50-70 ms?

kalyan-m25, 2025-07-28:
https://github.com/user-attachments/assets/ffea7ee0-24ab-4a1d-8857-5fd82ab5018a

kalyan-m25, 2025-07-28:
> > Each line is filled in two passes, 1 pass 50% brightness, 2 pass +50% brightness.
> 
> I shouldn't be this way. It means, probably, that we have something incorrect in multiplexing settings...

Can I have your email?