mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #816: ESP32 _div_num calculation causes red artifacts on some panels.

RichardD012, 2025-08-06:
I have 2 Tidbyt Gen2 Boards that I've loaded with tronbyt: https://github.com/tronbyt/firmware-esp32. This technically is a fork of the official tidbyt open source firmware. Tronbyt is using a newer version of this firmware

The tidbyt OS firmware uses a very old version of this repo: https://github.com/tidbyt/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/8f284afe7ba4ff369a4427121338e1673026320e

On one of the Gen2 Tidbyt (ESP32-D0WD-V3) (revision v3.1) the display is "fine". Animations look correct and there's no issue. Colors seem a tad off but nothing I'd seriously notice. 

On the second Gen2 Tidbyt, running the same firmware/configurtion/evertyhing, animations have red pixel artifacts. They aren't random, the animation will consistently have the same red pixel artifact in the same place (that moves with the animation). Obviously this points to some tolerance issue in the HW assuming the same boards but different manufacturing

I loaded the older OS firmware and the red pixels were gone. I'm not a HW guy by any means so I dug in to figure out what could be causing the issue. One of the first things I saw was this line: 

https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/blob/e83243b974f9a3be22923b5e29fa142cdd9cb3cc/src/platforms/esp32/esp32_i2s_parallel_dma.cpp#L240

versus 

https://github.com/tidbyt/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/8f284afe7ba4ff369a4427121338e1673026320e/esp32_i2s_parallel_dma.c#L163

As I said, I'm not a HW guy but messing with the _div_num values was clearly the cause of these weird red pixels. At first I tried `unsigned int _div_num = 80000000L / freq / 2;` and it worked but really that is just the value of 5. 

I've attached a picture of what I mean by red artifacts. These pixels should be white. With `_div_num = 5` they're white. I'm trying to understand what the actual "problem" here is and what is "correct". This may not be the best place for it but the only difference in my tests has been the underlying firmware (current, very old, my custom _div_num calculation).  

[image]

board707, 2025-08-06:
> I tried `unsigned int _div_num = 80000000L / freq / 2;` and it worked but really that is just the value of 5.

What is your `freq` value that the expression above give you a value of 5?

RichardD012, 2025-08-06:
8000000 (so 8Mhz)

RichardD012, 2025-08-08:
Since I don't know the extent of what this would/could change would you be opposed to a new flag or config setup for this specific case? Currently in my platformio project I have a pre-build step that patches this line but I'd imagine others could be having the same sort of issue if they're coming from Tidbyt (I could literally be the only person this applies to so I'm open to any idea)

mrcodetastic, 2025-08-24:
I'll add being able to set an 8Mhz clock frequency to the list.

In your situation I would say it's an issue with the LED panel is used in that it can't handle a frequency above > 8Mhz and does some weird red LED thing.