mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #821: Need help integrating new panels

TopoAS, 2025-08-11:
i have HRLOP3‑16S‑1415‑A1.0 //// OP3-16S-16269S-5166-937H-B3, these types of modules use SM16269S chips trying the library and i am not able to light the panels.
any help is apreciated.
i manage to get this pdf from the manufacturer
https://drive.google.com/file/d/1EIwC7BpSluJWeKNGaSM3Tl3NUPS4_MYU/view?usp=sharing
thank you

aslak3, 2025-08-26:
> So what do i need to ask from the manufacturer to be able to run this?

I am not associated with this project in any way nor have I used this software but I have messed with “conventional” HUB75 panels in my own projects using Pi Picos and FPGAs. 

Basically this panel you have bought has a “front end” interface on it consisting of an SPI slave bus and some SRAM. It does not present a standard HUB75 port. Presumably (I can’t read Chinese) it receives a bitmap in a particular format from the host and drives the LEDs from the memory content, including PWM. The interesting bit is this is roughly what this project does in software : it drives, using DMA and the LCD hardware in the ESP32 to produce a realtime hopefully flicker free image, the HUB75 port using a high bit depth bitmap held in the EPS32s memory.

In short if you wanted to make use of a panel like the one you’ve bought you’d have to rip out that part of this project and instead clock out, over SPI, the framebuffer data. To be honest you’d be as well not to use this project at all and investigate how to use a graphics library with it’s own framebuffer store and figure out how to drive the ESP32’s SPI bus with data, formatting that data in a way the panel expects. That  sounds like a fun project and if you can point me in the direction of where you got such a panel I might be able to source one and look at that myself. If you look at my GitHub you will see the kind of projects I’m busy with. Being upfront, the panel you have is very likely better and simpler to use then a typical HUB75 panel. And the fact you can interface with it using plain SPI is a massive plus point. Assuming you are not crippled in some other way, like bit depth. A translation of that datasheet would be awesome.

board707, 2025-08-26:
> To be honest you’d be as well not to use this project at all and investigate how to use a graphics library with it’s own framebuffer store and figure out how to drive the ESP32’s SPI bus with data, formatting that data in a way the panel expects.

Sorry, but to be honest it seems that you wrote about the things that you don't know. The protocol pf PWM type drivers has very little similarity with SPI.

TopoAS, 2025-08-26:
I ended up changing the pannel. The seller has provide me with a genuine controler from the manufacturer of the led and even that coud not light the panel. It require a startup sequence that the Chinese manufacturer is not providing. But instead he is willing to sell another controlers specifically programed for this module.
As i sayed i ended changing the modules with another versions that are not pwm and finished my project. Thank you all for the suport. I will continue to monitor this type of pannels because seems the chinese manufacturers are using this new chips.

aslak3, 2025-08-26:
> Sorry, but to be honest it seems that you wrote about the things that you don't know. The protocol pf PWM type drivers has very little similarity with SPI.

Fair dos. Won’t try reading between the lines of Chinese PDFs again. I did wonder how a panel with HUB75 clearly on the silkscreen would speak something else on its header. Will do some more reading. I got excited thinking someone had made an integrated “intelligent” controller but this is clearly not it.

board707, 2025-08-26:
> I did wonder how a panel with HUB75 clearly on the silkscreen would speak something else on its header

The HUB75 is type of connector, nothing more :) And the protocol may varied.
Even among panels with traditional drivers there is a lot of diversity.

In addition, despite the fact that the conventional protocol is quite close to SPI, this similarity gives little benefit, since most microcontrollers do not support SPI with 6 parallel data lines :)

board707, 2025-08-26:
@aslak3 
Since you write for FPGA for hub75 panels, it would be interesting to talk to you. I have figured out how to work with some of PWM-type drivers, but I know very little about FPGA...
I think that with the advent of new types of drivers, such as SM16269 in this thread - the use of FPGA is becoming more and more in demand.

aslak3, 2025-08-26:
> [@aslak3](https://github.com/aslak3) Since you write for FPGA for hub75 panels, it would be interesting to talk to you. I have figured out how to work with some of PWM-type drivers, but I know very [...]

Sounds very interesting.

My controller: https://github.com/aslak3/hub75-controller

It is very much targeted at my particular application in that it only supports 64x32 panels. Though the image fed over SPI is 64x32@32bpp it is is truncated to 16bpp (obviously only 12bpp in practice) due to size constraints in the FPGAs SRAM (Block RAM). This is dual ported; a single memory that can be accessed concurrently: written by the SPI side and read by the HUB75 side. PWM is achieved in the familiar way, by stretching the /OE pulse. Naturally there are two image buffers, one for the frame coming in and one for the frame being displayed. The controller flips the buffers by toggling a signal. The nice thing, of course, is the processor can feed the image in at whatever rate it wants. It can even crash and the image continues being shown. The timing was loosely based on the Pico’s PIO HUB75 code, though I simplified it somewhat to reduce the code in the FSM. 

Just to warn you, I’m basically self taught. Just a hacker. My interests, beside MCU type projects is in processor design. Well you can see in my GitHub.

Happy to help in any way I can. Would appreciate any guidance you’ve got on my little porting project? The code for it is here BTW: https://github.com/aslak3/matrix-display

board707, 2025-08-26:
Thank you for the interesting conversation. I visited your projects already a few days ago - I came across a link to your GitHub on one of the forums. I am very interested in your project on connecting Pi Pico and FPGA, because this is approximately what I would like to do.  
My main goal is to expand the number of supported panel types. Newest PWM drivers require rather complex pulse sequences, which are difficult to implement on microcontrollers - that's why I became interested in FPGA.

I now have a library for HUB75 panels based on STM32 and Pi Pico (RP2040) controllers. The code for Pico is a complex of DMA and PIO machines. It can work with conventional panels almost of any size and scan type and allows you to chain more than 10 panels 64x32 with 12bpp color. By the way, on Pico I also tried parallel output to two Hub75s at once, which will increase the number of panels by 2 times.

> I came to this project because I want to have a go at porting my Pico project (without FPGA) to the ESP32

To be honest, I know almost nothing about ESP32 internals, sorry :) so I can hardly be useful in porting the code. 
But I have accumulated some experience in working with different types of HUB75 panels, so if you have any questions, please contact me.
My project is https://github.com/board707/DMD_STM32

board707, 2025-08-26:
Continuing the conversation about different types of HAB75 drivers..
Following the link below you can see a table of chips used on DMD RGB panels with notes on compatibility with existing libraries:
https://github.com/board707/DMD_STM32/wiki/Led_drivers

board707, 2025-08-26:
> I didn't realise your library supported the RP2040.

It is another library, not the one on github we are on now :)

mrcodetastic, 2025-08-27:
> > I didn't realise your library supported the RP2040. That's great. I bought one of these for a few bucks off aliexpress the other day and was keen to experiment. The PIO capability seems incredibly powerful.
> 
> Yeah the PIO is great. Folks have done all kinds of cool things with it.
> 
>  Nothing against this project here, but it's massive and I think I will struggle picking it apart to get at the DMA engine, which is all I want.

Here's an example of using the DMA engine on a S2, which is applicable to the ESP32 as well (except you can't do 24 bit parallel with the original ESP32 as there aren't enough GPIOs, even though the silicon supports it lol)

https://github.com/mrcodetastic/esp32s2-24bit-i2s-lcd-parallel

Easy to change this for 16 bit output.

This code doesn't have any of the HUB75 BCM stuff which is what the majority of this library is.

psyirius, 2025-10-23:
s

closed 2025-10-25 by mrcodetastic