mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #824: ICN1065S SUPPORT

BubbleCaro, 2025-08-13:
Hello, I have some questions regarding the integrated circuit (IC) supported by this library. According to the documentation, it seems that there is no support for the IC shift register PWM with buffers, at least based on my understanding.

I currently have a panel with ICND1065s. While I know it utilizes PWM, I compared the information from the ChiponeIC website and the details in ICND2055, which indicate that it uses SRAM for storage. However, it appears that the ICND1065s does not have this feature. Does this mean that the library should support the ICND1065s? Additionally, I am confused about whether this library also supports 8s panels.

board707, 2025-08-13:
> Should I make any additional configurations?

See the discussion #819

board707, 2025-08-14:
> My module activates 4 rows at a time; it is supposed to be 1/16, right?

Could you show a clear photo of the panel rear side?

BubbleCaro, 2025-08-15:
I'm sorry, but I haven't been able to find a suitable library. It’s possible that I'm configuring it incorrectly.

[image]

this is using a nova card MRV208-1 
[image]

board707, 2025-08-15:
Sorry, your picture is blurry, take another picture of the area around the connector, so that all the labels are readable.

But in general, I'm afraid you won't be able to run this panel because new panels with only one connector are usually incompatible with the traditional hub75 protocol

BubbleCaro, 2025-08-15:
Thank you very much. I’ll leave you with a photo. By any chance, do you know the name of the protocol they use that I can search for?

[image]

board707, 2025-08-15:
> do you know the name of the protocol they use that I can search for?

Information about this should be found in the ICN1065 driver datasheet.

BubbleCaro, 2025-08-15:
I found the datasheet and I can see that information. I've seen it is similar to other ICN datasheets I've read.

[ICND1065_datasheet_CN_2023_V1.1(2)(1).pdf](https://github.com/user-attachments/files/21801602/ICND1065_datasheet_CN_2023_V1.1.2.1.pdf)

board707, 2025-08-17:
> I found the datasheet and I can see that information. I've seen it is similar to other ICN datasheets I've read.

The driver logic diagram on the datasheet page 4 clearly stated that this is PWM driver with SRAM storage. Your panel will not work with this library.