mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #828: DP5135+RUL5158C support ?

A97N, 2025-08-26:
Dear All,
    Following is the P4 panel

[image]

A reference to the RUL5158c chipset  is in this issue:  [#164](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/164)

But there is no output on the display using an SM5266P chipset for the driver.
This chipset setting:
mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

does give double horizontal pixel output as in video below:

https://github.com/user-attachments/assets/a2350915-1cf1-4ac6-b412-a6cc546f7640

The file being used is 1_SimpleTestShapes with the changes listed as follows:

#define PANEL_RES_X 80    // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 40     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another
//mxconfig.gpio.e = 18;
//mxconfig.clkphase = false;
mxconfig.driver = HUB75_I2S_CFG::DP3246_SM5368;

Any pointers to get the correct output? 
Thanks in advance.

A97N, 2025-08-27:
The video of the changes:

https://github.com/user-attachments/assets/8f9982ef-3d15-475f-a70b-60e2fc90177b

Next ? The letters still are garbled when printed.
TIA

board707, 2025-08-27:
Thanks
Unfortunately,  the scanning of dots in your video is too quick.  At such a speed it is impossible to understand the order of filling the screen. Put a delay of 30, or better 50 ms where you inserted delay(5).

Addition: You don't have to record the entire panel on video, the first quarter is enough.

A97N, 2025-08-27:
Yes sorry about that.
Hopefully this video is good enough

https://github.com/user-attachments/assets/039f3ee6-15e2-469f-a963-db23dd346651

The get_coords needs to provide a way for the physical pixels to be linearly mapped by calculation.

A97N, 2025-08-27:
https://github.com/user-attachments/assets/f7180e6c-c70c-46fc-8320-074896b67167

Pixel color change every 4 px, easier to spot changes. 100ms per pixel write.

board707, 2025-08-27:
`inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y)
{
  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

uint8_t pxbase = 4;   // pixel base
if  (((coords.y / 10)%2) == 0) 
    {
   coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
   }
else
  {
   coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
   }

       coords.y = (coords.y / 20) * 10 + (coords.y % 10);
  return coords;
}`

closed 2025-09-28 by A97N