mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #834: void myTest(String str){} -> drawing problem

Sascha8, 2025-09-13:
My 2x 64x32 panels chained to a 128x32 working really great on an ESP32. Thank you for the great lib! 

But when I declare an procedure with an arduino String parameter and call it, the display each time draws just one single line at row0 and row 15. 

`void myTest(String str) {}`

Can somebody please explain whats the reason?  chatGPT and Gemini both talking about heap fragmentation and DMA timing issues.  This makes me wonder, as I can use wifi, mqtt, OTA and other stuff without any drawing probs.  getFreeHeap() returns 289748 at the program below .

```
void setup() {
  HUB75_I2S_CFG mxconfig(PANEL_RES_X,PANEL_RES_Y,PANEL_CHAIN);
  mxconfig.clkphase = false;
  mxconfig.driver = HUB75_I2S_CFG::FM6124;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(20); //0-255
  dma_display->clearScreen();
  delay(200);
  dma_display->setTextSize(1, 1);
  dma_display->setCursor(0, 0); dma_display->println("Hello");
  myTest("");  //doesn't mater if I call it after or before panel init. (Using const char* str works fine)
}

void loop() {
  delay(20);
}

void myTest(String str) {
// do nothing
}

```
Debug log:
MinFreeHeap:326744
FreeHeap:332416
[   541][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:448] begin(): [begin()] Using GPIO 25 for R1_PIN
[   552][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:449] begin(): [begin()] Using GPIO 26 for G1_PIN
[   561][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:450] begin(): [begin()] Using GPIO 27 for B1_PIN
[   569][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:451] begin(): [begin()] Using GPIO 14 for R2_PIN
[   577][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:452] begin(): [begin()] Using GPIO 12 for G2_PIN
[   586][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:453] begin(): [begin()] Using GPIO 13 for B2_PIN
[   594][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:454] begin(): [begin()] Using GPIO 23 for A_PIN
[   602][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:455] begin(): [begin()] Using GPIO 19 for B_PIN
[   611][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:456] begin(): [begin()] Using GPIO 5 for C_PIN
[   619][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:457] begin(): [begin()] Using GPIO 17 for D_PIN
[   627][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:458] begin(): [begin()] Using GPIO -1 for E_PIN
[   635][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:459] begin(): [begin()] Using GPIO 4 for LAT_PIN
[   644][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:460] begin(): [begin()] Using GPIO 15 for OE_PIN
[   652][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:461] begin(): [begin()] Using GPIO 16 for CLK_PIN
[   661][I][ESP32-HUB75-MatrixPanel-leddrivers.cpp:50] fm6124init(): [LEDdrivers] MatrixPanel_I2S_DMA - initializing FM6124 driver...
[   673][I][ESP32-HUB75-MatrixPanel-I2S-DMA.h:472] begin(): [begin()] HUB75 effective display resolution of width: 128px height: 32px.
[   685][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:43] setupDMA(): [I2S-DMA] Free heap: 332204
[   693][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:44] setupDMA(): [I2S-DMA] Free SPIRAM: 0
[   701][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:71] setupDMA(): [I2S-DMA] Allocating 32768 bytes memory for DMA BCM framebuffer(s).
[   713][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:85] setupDMA(): [I2S-DMA] Minimum visual refresh rate (scan rate from panel top to bottom) requested: 60 Hz
[   727][W][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:105] setupDMA(): [I2S-DMA] lsbMsbTransitionBit of 0 gives 28 Hz refresh rate.
[   738][W][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:105] setupDMA(): [I2S-DMA] lsbMsbTransitionBit of 1 gives 55 Hz refresh rate.
[   749][W][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:105] setupDMA(): [I2S-DMA] lsbMsbTransitionBit of 2 gives 100 Hz refresh rate.
[   760][W][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:118] setupDMA(): [I2S-DMA] lsbMsbTransitionBit of 2 used to achieve refresh rate of 60 Hz. Percieved colour depth to the eye may be reduced.
[   776][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:121] setupDMA(): [I2S-DMA] DMA frame buffer color depths: 8
[   786][I][esp32_i2s_parallel_dma.cpp:456] allocate_dma_desc_memory(): [ESP32/S2] Allocating memory for 512 DMA descriptors.
[   797][W][esp32_i2s_parallel_dma.cpp:544] create_dma_desc_link(): [ESP32/S2] Creating final DMA descriptor and linking back to 0.
[   809][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:234] setupDMA(): [I2S-DMA] Created 512 DMA descriptors for buffer 0.
[   819][I][esp32_i2s_parallel_dma.cpp:126] config(): [ESP32/S2] Performing config for ESP32 or ESP32-S2
[   828][I][ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:301] setupDMA(): [I2S-DMA] DMA setup completed
[   840][I][esp32_i2s_parallel_dma.cpp:133] init(): [ESP32/S2] Performing DMA bus init() for ESP32 or ESP32-S2
MinFreeHeap:288616
FreeHeap:289748

mrcodetastic, 2025-09-21:
ChatGPT and Gemini are just hallucinating some radom reason. I can't see anything wrong in your code.

The first thing you should probably do though, is run this example to check that each pixel is being addressed correctly. If it is, then it's not an issue with this code and something funky with the length of the text you're trying to print.

I can't see from the code you're provided what is being printed in any case. 

Ihttps://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/Pixel_Mapping_Test

mrcodetastic, 2025-09-21:
ChatGPT and Gemini are just hallucinating some random reason.

The first thing you should probably do though, is run this example to check that each pixel is being addressed correctly. If it is, then it's not an issue with this code and something funky with the length of the text you're trying to print.

I can't see from the code you're provided what is being printed in any case to explain your issue.

Ihttps://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/tree/master/examples/Pixel_Mapping_Test

board707, 2025-09-22:
> dp->Print(str);

The C++ is a case-sensitive. The `print()` and `Print()` are two different methods. May be calling the `print()` will got you a better results?