mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #836: Dear Sir, i am also tring but getting success,May you help me out.

Rajkumar821108, 2025-09-15:
> I solved the problem of matrix panel with p8 20x40 pixels 1/5 scan rate. I managed to get it working with library https://github.com/2dom/PxMatrix#shiftreg_abc_bin_de. now i have problem with p10 rgb matrix panel which has 1/4 scan rate. there are only rows A and B. Again the same library ran, but one line works in reverse.> 
> > 
> @oldcity83 > 
> I  have same problem with P8 20*40 1/5, can you help me to solve this problem. 

 _Originally posted by @saka2080 in [#204](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/204#issuecomment-2011850953)_

Rajkumar821108, 2025-09-15:
Hi @board707 !
you check it, 

[image]

juber23567, 2025-09-16:
Please Explain in Details

Rajkumar821108, 2025-09-16:
Dear Sir, the panel designed with ICs DP5125D-30piece,RUC7258D-4piece,74HC245KAJ-3piece.
[image]

Rajkumar821108, 2025-09-16:
Dear Sir, i am using the code :/*************************************************************************
   Description:

   The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
   supports output to HALF scan panels - which means outputting
   two lines at the same time, 16 or 32 rows apart if a 32px or 64px high panel
   respectively.
   This cannot be changed at the DMA layer as it would require a messy and complex
   rebuild of the library's internals.

   However, it is possible to connect QUARTER (i.e. FOUR lines updated in parallel)
   scan panels to this same library and
   'trick' the output to work correctly on these panels by way of adjusting the
   pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
   library.

 **************************************************************************/

/* Use the Virtual Display class to re-map co-ordinates such that they draw
   correctly on a 32x16 1/4 or 64x32 1/8 Scan panel (or chain of such panels).
*/
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"

// Define custom class derived from VirtualMatrixPanel
class CustomPxBasePanel : public VirtualMatrixPanel
{
  public:
    using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

  protected:

    VirtualCoords getCoords(int16_t x, int16_t y);  // custom getCoords() method for specific pixel mapping

};

// custom getCoords() method for specific pixel mapping
inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

if (((coords.y / 5)%2) == 0)
        {
            coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }
        else
        {
            coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
        }

       coords.y = (coords.y / 10) * 5 + (coords.y % 5);
  return coords;
} 

// Panel configuration
#define PANEL_RES_X 40 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 20 // Number of pixels tall of each INDIVIDUAL panel module.

// Use a single panel for tests
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

// Chain settings, do not cnahge
#define SERPENT true
#define TOPDOWN false
#define VIRTUAL_MATRIX_CHAIN_TYPE CHAIN_BOTTOM_RIGHT_UP

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
CustomPxBasePanel   *FourScanPanel = nullptr;

/******************************************************************************
   Setup!
 ******************************************************************************/
void setup()
{
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X * 2,            // DO NOT CHANGE THIS
    PANEL_RES_Y / 2,            // DO NOT CHANGE THIS
    NUM_ROWS * NUM_COLS         // DO NOT CHANGE THIS
    //,_pins            // Uncomment to enable custom pins
  );

  mxconfig.clkphase = false; // Change this if you see pixels showing up shifted wrongly by one column the left or right.

  // OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(40);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not dma_display->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  dma_display->clearScreen();
  delay(500);

  // create FourScanPanellay object based on our newly created dma_display object
  FourScanPanel = new CustomPxBasePanel ((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,  VIRTUAL_MATRIX_CHAIN_TYPE);

}

void loop() {
  for (int i = 0; i < FourScanPanel->height(); i++)
  {
    for (int j = 0; j < FourScanPanel->width(); j++)
    {
      FourScanPanel->drawPixel(j, i, FourScanPanel->color565(255, 0, 0));
      delay(30);
    }
  }
  delay(2000);
  dma_display->clearScreen();
} // end loop

https://github.com/user-attachments/assets/4bbf563e-2935-4961-8df3-64a05ce0b601

board707, 2025-09-16:
Thanks for your video.
I'm at work now and will answer you later.

board707, 2025-09-16:
Please change the code of the getCoords() method this way and show the video:
```
inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y)
{
  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

uint8_t pxbase =4;   // pixel base
if  ((((coords.y / 5)%2) == 0) ^ ((coords.x/pxbase) % 2))
    {
   coords.x += (coords.x / pxbase) * pxbase; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
   }
else
  {
   coords.x += ((coords.x / pxbase) + 1) * pxbase; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
   }

       coords.y = (coords.y / 10) * 5 + (coords.y % 5);
  return coords;
}
```

Rajkumar821108, 2025-09-17:
Dear Sir,HAVE A WONDERFUL DAY!, May you guide for ScrollingTextLayer example on two panel(20x80pixel).

https://github.com/user-attachments/assets/c531de00-9fee-4885-a4a5-7ea120368fd5

Rajkumar821108, 2025-09-17:
> What is the problem?

Dear Sir,i am using Example "ScrollingTextLayer" with ESP by Espressif
System version 2.0.17, i am attaching output, where i have to change for
pixel setting?

script is
// Modified from:
https://github.com/wilson3682/My-HUB75-ESP32-Practice-Files/tree/main
// by MrCodetastic

/***************************************************************************************

   - This sketch does a couple of things, it uses GFX_Lite's GFX_Layer to
   independently
   - draw the background and the text onto separate layers. (memory used
   for each)
   - Then these are stacked on top of each other with some opacity, and
   drawn directly
   - to the dma_display via a callback (layer_draw_callback).
   -
   - These layers are offscreen pixel buffers that use memory of their own
   - (approx 3 bytes for each pixel).
   -
   - By using this approach, we don't really need to use DMA double
   buffering though.
   -

***************************************************************************************/

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

// You need to have: https://github.com/mrcodetastic/GFX_Lite
#include <GFX_Layer.hpp>

//------------------------------------------------------------------------------------------------------------------

// Configure for your panel(s) as appropriate!
#define PANEL_WIDTH 40
#define PANEL_HEIGHT 20 // Panel height of 64 will required PIN_E to be
defined.
#define CHAIN_LENGTH 1 // Number of chained panels, if just a single panel,
obviously set to 1

//------------------------------------------------------------------------------------------------------------------

#define RL1 25
#define GL1 26
#define BL1 27
#define RL2 14
#define GL2 12
#define BL2 13
#define CH_A 23
#define CH_B 19
#define CH_C 5
#define CH_D 17
#define CH_E -1 // assign to any available pin if using two panels or 64x64
panels with 1/32 scan
#define CLK 16
#define LAT 4
#define OE 15

//------------------------------------------------------------------------------------------------------------------

MatrixPanel_I2S_DMA *dma_display = nullptr;

//------------------------------------------------------------------------------------------------------------------

//====================== Variables For scrolling
Text=====================================================
unsigned long isAnimationDue;
int delayBetweeenAnimations = 18; // Smaller == faster
[... 204 lines cut ...]
}

 gfx_layer_bg.dim(192); // darken it a little

}

uint8_t wheelval = 0;
void loop() {

updateBackground();

scrollText(wheelval, "HAVE A WONDERFUL DAY!"); //Prints Scrolling text with
a rainbow color
printTextRainbowCentered(wheelval, "ENJOY IT", 24); //Prints text
X-Centered to chosen Y position with rainbow color

//gfx_layer_fg.display();
//gfx_layer_bg.display();

gfx_compositor.Blend(gfx_layer_bg, gfx_layer_fg); // blend and immediately
display

wheelval += 1;

}

board707, 2025-09-17:
To work with your panel you have to add a VirtualMatrixPanel class to each your code. See the code that we use to test the pixel mapping and try to edit your code similar way.

board707, 2025-09-19:
> please fix the problem as you have done previous

so start from the initial test code as we done in [post](https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/836#issuecomment-3296329476)

board707, 2025-09-19:
next one:

```
inline VirtualCoords EightPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

const uint8_t pixBase =8;
if (((coords.y / 5)%2) == 0)
          {
            coords.x = (coords.x / pxbase)*2*pxbase   + 7 - (coords.x & 0x7);
        }
        else
        {
           coords.x += ((coords.x / pxbase) + 1) * pxbase; 
        }

       coords.y = (coords.y / 10) * 5 + (coords.y % 5);
  return coords;
}

...

Rajkumar821108, 2025-09-19:
Thankyou sir, but i have come from workspace for room, tomorrow i will
check

Rajkumar821108, 2025-09-20:
Money is earned from rich people, only blessings can be obtained from freshers!  may God help you for your kindness!                                                                                      Dear Sir, next_one code giving error due to function name(EightPxBasePanel) and variable name(pxbase).                                           so i replaced these name as previous:                                                                                                                                                                                inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y) {

  coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

const uint8_t pixBase =8;
if (((coords.y / 5)%2) == 0)
          {
            coords.x = (coords.x / panelResX)*2*panelResX   + 7 - (coords.x & 0x7);
        }
        else
        {
           coords.x += ((coords.x / panelResX) + 1) * panelResX; 
        }

       coords.y = (coords.y / 10) * 5 + (coords.y % 5);
  return coords;
}

https://github.com/user-attachments/assets/cbbba7ee-c3f1-4e6d-82aa-cbea849015e2

board707, 2025-09-20:
Sorry for the errors.
Correct code
```
inline VirtualCoords CustomPxBasePanel ::getCoords(int16_t x, int16_t y)  { 

coords = VirtualMatrixPanel::getCoords(x, y); // first call base class

if ( coords.x == -1 || coords.y == -1 ) { 
 return coords; } 

const uint8_t pxbase =8; 
if (((coords.y / 5)%2) == 0) { 
  coords.x = (coords.x / pxbase)*2*pxbase + 7 - (coords.x & 0x7);
 } 
else { 
  coords.x += ((coords.x / pxbase) + 1) * pxbase; 
} 
 coords.y = (coords.y / 10) * 5 + (coords.y % 5);
 return coords; 
}
```

Rajkumar821108, 2025-11-19:
Dear Sir,
Is there an issue with this library? When the brightness is increased,
the ghosting LEDs start to glow. If the voltage is around 6 volts,it will be fixed,why?
 i am using the code:#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
#include <EEPROM.h>
#include "font5x7.h"
#include "font8x8.h"
#include "font10x16.h"
#include <pgmspace.h>

#define EEPROM_SIZE      512
#define ADDR_UP_TEXT     0
#define ADDR_LOW_TEXT    100
#define ADDR_UP_COLOR    200
#define ADDR_UP_MODE     201
#define ADDR_LOW_COLOR   202
#define ADDR_LOW_MODE    203
#define ADDR_UP_FONT     204
#define ADDR_LOW_FONT    205
#define ADDR_UP_SPEED    206
#define ADDR_LOW_SPEED   207

// NEW addresses for position + scroll state (two bytes for scroll stored as int16 scaled)
#define ADDR_UP_X        208
#define ADDR_UP_Y        209
#define ADDR_LOW_X       210
#define ADDR_LOW_Y       211
#define ADDR_SCROLLX_UP  212 // int16 -> uses 212 & 213
#define ADDR_SCROLLX_LOW 214 // int16 -> uses 214 & 215
#define ADDR_RELAY_STATE 216
#define ADDR_BRIGHTNESS 217

#define RELAY_PIN 5

#define R1_PIN 15
#define G1_PIN 2
#define B1_PIN 13
#define R2_PIN 12
#define G2_PIN 4
#define B2_PIN 14
#define A_PIN 27
#define B_PIN 23
#define C_PIN 26
#define D_PIN -1
#define E_PIN -1
#define LAT_PIN 32
#define OE_PIN 33
#define CLK_PIN 25

HUB75_I2S_CFG::i2s_pins _pins = {
  R1_PIN, G1_PIN, B1_PIN,
  R2_PIN, G2_PIN, B2_PIN,
  A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
  LAT_PIN, OE_PIN, CLK_PIN
};

class CustomPxBasePanel : public VirtualMatrixPanel {
public:
  using VirtualMatrixPanel::VirtualMatrixPanel;
protected:
  VirtualCoords getCoords(int16_t x, int16_t y);
};

inline VirtualCoords CustomPxBasePanel::getCoords(int16_t x, int16_t y)
{
  coords = VirtualMatrixPanel::getCoords(x, y);

  if (coords.x == -1 || coords.y == -1)
    return coords;

[... 568 lines cut ...]
    if (lowerText.length() > 0) {
      if (lowerMode == 2) {  // scrolling
        int textHeight = (lowerFont == 2) ? 16 : (lowerFont == 1 ? 8 : 7);

        // clear previous area only for scrolling text
        clearScrollArea((int)fScrollXlower, lowerY,
                        (int)fScrollXlower + lowerWidth - 1,
                        lowerY + textHeight - 1);

        fScrollXlower -= fScrollSpeedLower;
        if (fScrollXlower < -lowerWidth) fScrollXlower = FourScanPanel->width();

        drawCustomText(FourScanPanel, (int)fScrollXlower, lowerY,
                       lowerText, lowerColor, lowerFont);
      } else {  // static
        drawCustomText(FourScanPanel, lowerX, lowerY,
                       lowerText, lowerColor, lowerFont);
      }
    }
    // do NOT delay here, keep ISR/loop responsive
  }

  // allow background housekeeping; short yield keeps watchdog happy
  yield();
}

Rajkumar821108, 2025-11-20:
Good day Sir,
I am using below code, I am not sure whether I did it right or something wrong, because scrolling is not smooth, can you suggest better?
#include "ESP32-VirtualMatrixPanel-I2S-DMA.h"
#include <EEPROM.h>
#include "font5x7.h"
#include "font8x8.h"
#include "font10x16.h"
#include <pgmspace.h>

#define EEPROM_SIZE      512
#define ADDR_UP_TEXT     0
#define ADDR_LOW_TEXT    100
#define ADDR_UP_COLOR    200
#define ADDR_UP_MODE     201
#define ADDR_LOW_COLOR   202
#define ADDR_LOW_MODE    203
#define ADDR_UP_FONT     204
#define ADDR_LOW_FONT    205
#define ADDR_UP_SPEED    206
#define ADDR_LOW_SPEED   207

// NEW addresses for position + scroll state (two bytes for scroll stored as int16 scaled)
#define ADDR_UP_X        208
#define ADDR_UP_Y        209
#define ADDR_LOW_X       210
#define ADDR_LOW_Y       211
#define ADDR_SCROLLX_UP  212 // int16 -> uses 212 & 213
#define ADDR_SCROLLX_LOW 214 // int16 -> uses 214 & 215
#define ADDR_RELAY_STATE 216
#define ADDR_BRIGHTNESS 217

#define RELAY_PIN 5

#define R1_PIN 15
#define G1_PIN 2
#define B1_PIN 13
#define R2_PIN 12
#define G2_PIN 4
#define B2_PIN 14
#define A_PIN 27
#define B_PIN 23
#define C_PIN 26
#define D_PIN -1
#define E_PIN -1
#define LAT_PIN 32
#define OE_PIN 33
#define CLK_PIN 25

HUB75_I2S_CFG::i2s_pins _pins = {
  R1_PIN, G1_PIN, B1_PIN,
  R2_PIN, G2_PIN, B2_PIN,
  A_PIN, B_PIN, C_PIN, D_PIN, E_PIN,
  LAT_PIN, OE_PIN, CLK_PIN
};

class CustomPxBasePanel : public VirtualMatrixPanel {
public:
  using VirtualMatrixPanel::VirtualMatrixPanel;
protected:
  VirtualCoords getCoords(int16_t x, int16_t y);
};

inline VirtualCoords CustomPxBasePanel::getCoords(int16_t x, int16_t y)
{
  coords = VirtualMatrixPanel::getCoords(x, y);

  if (coords.x == -1 || coords.y == -1)
    return coords;

  uint8_t pxbase = 4;
[... 567 lines cut ...]
    if (lowerText.length() > 0) {
      if (lowerMode == 2) {  // scrolling
        int textHeight = (lowerFont == 2) ? 16 : (lowerFont == 1 ? 8 : 7);

        // clear previous area only for scrolling text
        clearScrollArea((int)fScrollXlower, lowerY,
                        (int)fScrollXlower + lowerWidth - 1,
                        lowerY + textHeight - 1);

        fScrollXlower -= fScrollSpeedLower;
        if (fScrollXlower < -lowerWidth) fScrollXlower = FourScanPanel->width();

        drawCustomText(FourScanPanel, (int)fScrollXlower, lowerY,
                       lowerText, lowerColor, lowerFont);
      } else {  // static
        drawCustomText(FourScanPanel, lowerX, lowerY,
                       lowerText, lowerColor, lowerFont);
      }
    }
    // do NOT delay here, keep ISR/loop responsive
  }

  // allow background housekeeping; short yield keeps watchdog happy
  yield();
}

softhack007, 2025-11-23:
@board707 did you recognize that his questions never end 😆

Everytime you help with the solution for one panel, our friend seems to just buy a different one that raises a new question ;-)

@Rajkumar821108 if you have the solution for one panel type, why not just keep buying the panel that works, instead of purchasing a different one?

Rajkumar821108, 2025-11-23:
@softhack007 I apologize for the trouble caused to you

jrousis, 2025-11-24:
> > Νow the new two sample modules that I have in my hands do not show anything in the test code, not even a LED light pu, obviously they may have different - invert signals?
> 
> I have a bad news for you - the panels has an incompatible (PWM-type) driver LS9936. I increased the contrast on your picture to make the label easier to read:
> 
> <img alt="Image" width="704" height="468" src="https://private-user-images.githubusercontent.com/50185434/518200934-8334b8e4-6057-4103-bb03-332915a35bb2.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQwMDEyMzAsIm5iZiI6MTc2NDAwMDkzMCwicGF0aCI6Ii81MDE4NTQzNC81MTgyMDA5MzQtODMzNGI4ZTQtNjA1Ny00MTAzLWJiMDMtMzMyOTE1YTM1YmIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTI0VDE2MTUzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA0MzkyNjg4OWJmOWM5NGQ3MjI4YmU2NWZlNmYxY2JkMTFkMWE5ZTUwOTFjYWViMmVhMTgyNmQ2NjQ2MDk0MTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.D3quV4KXcXV5jAkJZvg9Wu8Sq020BDnJkopsg1vQDQU">
> At the moment I don't know of any DIY library that would work with this driver.

Ohh I see!
Thanks a lota again, 
I will search for an other module of P8 40X20 then.

jrousis, 2025-11-29:
Finaly, I took some of these modules with the CHIPSET : DP5135 +DP32020, P6.67mm, for testing and just I run the Pixel_Mapping_Test.ino
with with your sugesting code: 
`inline VirtualCoords CustomPxBasePanel::getCoords(int16_t x, int16_t y) {

    coords = VirtualMatrixPanel::getCoords(x, y); // first call base class method to update coords for chaining approach

    if (coords.x == -1 || coords.y == -1) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
        return coords;
    }

    if (((coords.y / 5) % 2) == 0)
    {
        coords.x += ((coords.x / panelResX) + 1) * panelResX; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
    }
    else
    {
        coords.x += (coords.x / panelResX) * panelResX; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
    }

    coords.y = (coords.y / 10) * 5 + (coords.y % 5);
    return coords;
}`

although I think that this is for scan 1:5, but I she a strange behavior on the scaning...
some line arw scanned two times!

https://github.com/user-attachments/assets/5f4cb0c7-d7fd-4398-8832-9f51795c583f

[image]

board707, 2025-11-30:
You are welcome :)

jrousis, 2025-11-30:
The module with text
[image]

Rajkumar821108, 2026-01-13:
May i go with the library on RP2040 pico? or suggest better solution, because for big size one chain is not enough.
#include <Adafruit_Protomatter.h>

// -------- HUB75 PIN SETUP --------

// RGB pins
uint8_t rgbPins[] = {
  2, 3, 4,   // R1, G1, B1
  5, 6, 7    // R2, G2, B2
};

// A, B, C address lines only
uint8_t addrPins[] = { 8, 9, 10 };

// Control pins
uint8_t clockPin = 11;
uint8_t latchPin = 12;
uint8_t oePin    = 13;

// -------- MATRIX CONFIG --------
Adafruit_Protomatter matrix(
  32, 32,           // ONE P6 module
  rgbPins,
  3, addrPins,      // A, B, C only
  clockPin,
  latchPin,
  oePin,
  true              // double buffer
);

void setup() {
  matrix.begin();
  matrix.setBrightness(255);   // MAX brightness
  matrix.fillScreen(0);
  matrix.drawPixel(15, 15, matrix.color565(255, 0, 0));
  matrix.show();
}

void loop() {}

board707, 2026-05-17:
Hi @jrousis 
Is this the same module that we setup on December?