mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #837: Screen flickering issue in ESP-IDF 5.3

xixxi, 2025-09-17:
The example for ESP-IDF works on ESP32, but the screen flickers noticeably! The flickering only disappears with a 3-bit color depth, but appears with 8-bit. Regardless of double buffering and other settings. Interestingly, the example for Arduino IDE works perfectly! There is no flickering at all.

xixxi, 2025-09-21:
I used an example from the ESP32-HUB75-MatrixPanel-DMA\examples\esp-idf\without-gfx folder and compiled it in ESP-IDF 5.4 (not 5.3, but 5.4), and during operation, the screen flickers heavily. Meanwhile, the 2_PatternPlasma example, which is written for ArduinoIDE, works perfectly in ArduinoIDE 2.3.2.

mrcodetastic, 2025-09-21:
I'll take a look. Could be something specific to how that example was built.