mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #838: esp32-s3 Rows 16 to 23 overwrite rows 0 to 7 (96x48 panel)

nate0202, 2025-09-17:
I'm using a esp32-s3-devkitc-1-n16r8 board and the standard S3 pin assignments (esp32s3-default-pins.hpp). The panel is a single 96x48, no chaining.  

The problem I'm having is that the content of row 16 to 23 overwrite rows 0 to 7 (additively not replacement). This repeats on the bottom half of the board rows, 40 to 47 overwriting 24 to 31.

This panel works fine when paired with an older "esp32dev" esp board.

I'm using platformIO in Visual studio. This is my platformio.ini file:
```
; Flash: 16MB OT, PSRAM: 8MB OT
[env:esp32s3]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
board_build.arduino.partitions = default_16MB.csv
board_build.flash_mode = qio
board_build.f_flash = 80000000L
board_build.psram_mode = octal
board_build.psram_size = 8MB
lib_deps = 
	mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display@^3.0.13
	adafruit/Adafruit GFX Library@^1.12.1
```
I've tried a number of variations on different pin assignments, variations on board definitions, with no luck. I am glad to provide any additional information, pictures, code, etc.  Thanks.

nate0202, 2025-09-17:
here is the full back image, the chip marking are faint so let me know if  you need me to zoom in on any of them.

[image]

board707, 2025-09-17:
Thanks.
According to the image, the panel is 96x48 24S with ICN2037 drivers and SM5166 line switch - a good variant, nothing special.

Then please show your code. 
Didn't you forget to define a real value to E pin?

nate0202, 2025-09-17:
This is my sample code...
```

#include <Arduino.h>
//>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>> 01_Showing_Text
//----------------------------------------Including the libraries.
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <Adafruit_GFX.h>
//----------------------------------------

// default S3 pins used 
#define R1_PIN 4 //
#define G1_PIN 5 //
#define B1_PIN 6 // 
#define R2_PIN 7 // 
#define G2_PIN 15 // 
#define B2_PIN 16 //
#define A_PIN 18 // GRAY
#define B_PIN 8 // BLACK
#define C_PIN 3 // BROWN
#define D_PIN 42 // RED
#define E_PIN  -1 // GRAY
#define LAT_PIN  40 // YELLOW
#define OE_PIN 2 // GREEN
#define CLK_PIN 41 // ORANGE

HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, 
    A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};

#define PANEL_RES_X 96  // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 48  // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1   // Total number of panels chained one to another

MatrixPanel_I2S_DMA* dma_display = nullptr;

void setup() {
    Serial.begin(115200);
    delay(2000);
    Serial.println(">>>>> Starting up display test...");
    // HUB75 panel setup
    // Module configuration
    HUB75_I2S_CFG mxconfig(
        PANEL_RES_X,  // module width
        PANEL_RES_Y,  // module height
        PANEL_CHAIN  // Chain length
        ,_pins         // GPIO pin mapping
    ); 
    //mxconfig.latch_blanking = 4;
    //mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;
    //mxconfig.clkphase = false;
    //mxconfig.driver = HUB75_I2S_CFG::FM6124;

    // Display Setup
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);
    Serial.println(">>>>> display created..., begin next");
    delay(1000);
    dma_display->begin();  
    Serial.println(">>>>> begin complete...");
    delay(1000);
    dma_display->clearScreen();
}

void loop() {
    Serial.println(">>>>> loop...");
    dma_display->fillScreenRGB888(50, 50, 50);
    delay(2000);
    dma_display->fillRect(12, 12, 12, 16, dma_display->color565(180, 180, 0));
    delay(10000);
    dma_display->fillScreenRGB888(0, 50, 0);
    delay(2000);
    dma_display->fillScreenRGB888(0, 0, 50);
    delay(2000);
}

```

board707, 2025-09-17:
The error:
> #define E_PIN  -1 // GRAY

You have to define real pin number and connect it to panel E line.

nate0202, 2025-09-17:
Thank you that fixed it, based on the defaults in the esp32s3-default-pins.hpp I thought that was not used. 

I assigned and connected to pin 21 and it works now. 

Thank you so much.